minosworld/minos #1: Cannot move in some room

xcyan, 2017-12-13:
http://dovahkiin.stanford.edu/sim/simulator.html?scene[fullId]=p5dScene.0004d52d1aeeb8ae6de39d6bd993e992/&collisionDetection[mode]=navgrid&navmap[refineGrid]=true&agent[radialClearance]=0.2

msavva, 2017-12-14:
Thank you for trying out MINOS.

The scene id in the URL is not in the set of prepackaged scenes that we provide with the repository.  The set of prepackaged scenes is listed in [data/scenes.multiroom.csv](https://github.com/minosworld/minos/blob/master/data/scenes.multiroom.csv).

If you would like to experiment with other scenes in SUNCG, please run the [server/export_navgrid.js](https://github.com/minosworld/minos/blob/master/server/export_navgrid.js) script for the given scene ids, and add the output files to the navgrid directory under the extracted SUNCG data package.

We are working on expanding the documentation to include instructions on generating these scene metadata files, and on how to sample episodes for creating new benchmark train/val/test splits.  Please let us know if you have any other issues in the meantime.

closed 2017-12-14 by msavva

angelxuanchang, 2018-01-04:
We have precomputed navgrids for almost all of the SUNCG scenes now. 

You can try out the demo for scene 0004d52d1aeeb8ae6de39d6bd993e992 at: http://dovahkiin.stanford.edu/sim/simulator.html?scene[fullId]=p5dScene.0004d52d1aeeb8ae6de39d6bd993e992&collisionDetection[mode]=navgrid&navmap[refineGrid]=true&agent[radialClearance]=0.2

Note that the navgrid is precomputed for efficiency and having a well defined map of where the agent can go. It is also possible to use a collision mode that does not require the use of the navgrid: http://dovahkiin.stanford.edu/sim/simulator.html?scene[fullId]=p5dScene.0004d52d1aeeb8ae6de39d6bd993e992

xcyan, 2018-01-04:
@msavva @angelxuanchang Thank you Angela and Manolis!