minosworld/minos #3: Problem with running the pygame_client

nzer0, 2017-12-27:
**Environment**
OS: Ubuntu 16.04 LTS
Python: 3.5.2
node: v8.9.3
only SUNCG data is installed in the $HOME/work directory.

---

Hi, I am trying to run the code but it gives following messages. Though they are not "ERROR" messages, nothing happens. Could you help me out with this error?

```
2017-12-27 22:49:10,513 INFO {'stk_git_hash': '03855da', 'sim_id': 'sim00', 'sim_git_hash': '03855da', 'machine': 'mymachine'}
Starting simulator...
2017-12-27 22:49:10,524 INFO sim00:Starting sim server at /home/me/Projects/minos/server/server.js with port 17226
2017-12-27 22:49:11,533 INFO Starting new HTTP connection (1): localhost
2017-12-27 22:49:12,350 INFO sim00:connect
2017-12-27 22:49:12,447 WARNING localhost:17226/socket.io [connection error] recv disconnected ([Errno 104] Connection reset by peer)
2017-12-27 22:49:13,448 WARNING localhost:17226/socket.io [connection error] recv disconnected ([Errno 104] Connection reset by peer)
2017-12-27 22:49:13,448 INFO sim00:disconnect
2017-12-27 22:49:13,449 INFO Resetting dropped connection: localhost
2017-12-27 22:49:13,450 WARNING localhost:17226/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=17226): Max retries exceeded with url: /socket.io/?EIO=3&t=1514382553449-0&transport=polling (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x7fdae3e040f0>: Failed to establish a new connection: [Errno 111] Connection refused',))
2017-12-27 22:49:14,451 INFO Starting new HTTP connection (2): localhost
2017-12-27 22:49:14,451 WARNING localhost:17226/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=17226): Max retries exceeded with url: /socket.io/?EIO=3&t=1514382554450-0&transport=polling (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x7fdae3e04ef0>: Failed to establish a new connection: [Errno 111] Connection refused',))
2017-12-27 22:49:15,453 INFO Starting new HTTP connection (3): localhost
2017-12-27 22:49:15,453 WARNING localhost:17226/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=17226): Max retries exceeded with url: /socket.io/?EIO=3&t=1514382555452-0&transport=polling (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x7fdae3e05080>: Failed to establish a new connection: [Errno 111] Connection refused',))
...
...
```

Actually, it seems to be a problem while connecting the socket, so I tried to manually run the node server in separate terminal by typing in `node server.js -p 17226` inside the server directory. Then the python gives following message and stops to throw warnings. But it still does not pop up any windows or give anything.

```
2017-12-27 22:57:11,571 INFO Starting new HTTP connection (14): localhost
2017-12-27 22:57:12,089 INFO sim00:reconnect
```

msavva, 2017-12-28:
Thanks for trying out MINOS!

These log messages indicate that the client cannot successfully connect to the server.  Within the working directory from which you start the client, a set of log files will be generated in `logs/<timestamp>/simserver.log` and `logs/<timestamp>/simulator.log`.  Can you please inspect both and quote here any errors that you see in these log files (in particular `simserver.log`)?

Ideally, can you provide a link to the entire files?  That will help us to debug any issues you may be having on your system.

nzer0, 2017-12-28:
Hi thanks for the reply. Here are the complete logs.
From the logs, it seems to be the `gl` module inside the `sstk` module is causing the problem.
So I googled it and found this possible solution, https://github.com/stackgl/headless-gl/issues/65#issuecomment-252742795 but trying it still gives the same error log.

**simserver.log**
```
2017-12-28 16:13:39,504 INFO b'Configuration { fileCache: { size: 50 },' 
2017-12-28 16:13:39,504 INFO b'  imageCache: { size: 200 },' 
2017-12-28 16:13:39,505 INFO b'  imageQueue: { concurrency: 4 },' 
2017-12-28 16:13:39,505 INFO b"  base_url: '/home/me/work/',"
2017-12-28 16:13:39,505 INFO b"  assets_url: 'file:///home/me/Projects/minos/server/node_modules/sstk/ssc/../server/static' }"
2017-12-28 16:13:39,768 INFO b'Waiting for client connection on port 48566'
2017-12-28 16:13:39,781 ERROR b'Cannot register unknown asset 2d3ds'
2017-12-28 16:13:39,782 INFO b'register ../server/static/data/matterport/matterport3d.json'
2017-12-28 16:13:39,783 INFO b'Registered asset group: mp3d'
2017-12-28 16:13:39,786 INFO b'Got 90 assets'
2017-12-28 16:13:39,827 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2017-12-28 16:13:39,828 INFO b'Replace asset group p5d'
2017-12-28 16:13:39,828 INFO b'Registered asset group: p5d'
2017-12-28 16:13:39,868 INFO b'Got 2623 assets'
2017-12-28 16:13:40,516 INFO b'Loaded lightSpecs for 202 models.'
2017-12-28 16:13:40,517 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2017-12-28 16:13:40,518 INFO b'Replace asset group p5dTexture'
2017-12-28 16:13:40,519 INFO b'Registered asset group: p5dTexture'
2017-12-28 16:13:40,646 INFO b'Got 67072 assets'
2017-12-28 16:13:40,765 INFO b'Client EqxBFQL8PD8yJyMvAAAA connected on port 48566'
2017-12-28 16:13:40,769 INFO b'Loaded LightSpecs for 202 models'
2017-12-28 16:13:40,832 ERROR b'node: symbol lookup error: /home/me/Projects/minos/server/node_modules/sstk/ssc/node_modules/gl/build/Release/webgl.node: undefined symbol: _Z15XextFindDisplayP15_XExtensionInfoP9_XDisplay'
2017-12-28 16:13:40,846 INFO Finished simserver stdout
2017-12-28 16:13:40,846 INFO Finished simserver stderr
```

**simulator.log**
```
2017-12-28 16:13:39,069 INFO {'sim_git_hash': '03855da', 'stk_git_hash': '03855da', 'machine': 'mymachine', 'sim_id': 'sim00'}
2017-12-28 16:13:39,080 INFO sim00:Starting sim server at /home/me/Projects/minos/server/server.js with port 48566
2017-12-28 16:13:40,774 INFO sim00:connect
2017-12-28 16:13:41,848 INFO sim00:disconnect
2017-12-28 16:13:45,502 INFO sim00:Stopping the simulator
2017-12-28 16:13:45,502 INFO Counter()
```

msavva, 2017-12-28:
Hi,

The first error message regarding 2d3ds can be ignored -- it is just a message indicating that an additional dataset is not present.

The second error message is indeed an issue we have observed before on linux machines with nvidia gpus.  Can you confirm that you have indeed successfully rebuilt the webgl.node following the instructions in the linked issue thread, and then copied it over the original webgl.node file in `./server/node_modules/sstk/node_modules/gl/build/Release/webgl.node`?

nzer0, 2017-12-29:
I am indeed using nvidia gtx 970 GPU.
And yes, I've compiled webgl.node and copied it to that directory, but gives the same error log.

msavva, 2017-12-29:
Since the issue is actually with the `sstk` library, can you try cloning that repository, running `npm install` in the cloned directory, and then running the `ssc/render.js` script?  This is a more minimal test of whether the headless-gl module is working.  If you see a similar error message, overwrite `sstk/node_modules/gl/build/Release/webgl.node` with the file rebuilt on your system (copied from `headless-gl/build/Release/webgl.node` -- please also verify that this file is indeed rebuilt from source on your system, i.e. that it has a new timestamp).  It's also good to check that there are no other copies of the `webgl.node` file in your directory tree that have not been updated (under either `minos` or `sstk` depending on which one you are trying out)

XL2013, 2018-01-04:
I met the same problem, but the log's info is different. my environment is the same as **nzero**.

**simserver.log**
```
2018-01-04 14:34:12,646 INFO b'Configuration { fileCache: { size: 50 },'
2018-01-04 14:34:12,647 INFO b'  imageCache: { size: 200 },'
2018-01-04 14:34:12,647 INFO b'  imageQueue: { concurrency: 4 },'
2018-01-04 14:34:12,647 INFO b"  base_url: '/home/user2/work/',"
2018-01-04 14:34:12,647 INFO b"  assets_url: 'file:///home/user2/minos/server/node_modules/sstk/ssc/../server/static' }"
2018-01-04 14:34:12,953 INFO b'Waiting for client connection on port 19849'
2018-01-04 14:34:12,969 ERROR b'Cannot register unknown asset 2d3ds'
2018-01-04 14:34:12,969 INFO b'register ../server/static/data/matterport/matterport3d.json'
2018-01-04 14:34:12,971 INFO b'Registered asset group: mp3d'
2018-01-04 14:34:12,975 INFO b'Got 90 assets'
2018-01-04 14:34:13,021 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2018-01-04 14:34:13,022 INFO b'Replace asset group p5d'
2018-01-04 14:34:13,022 INFO b'Registered asset group: p5d'
2018-01-04 14:34:13,074 INFO b'Got 2623 assets'
2018-01-04 14:34:13,831 INFO b'Loaded lightSpecs for 202 models.'
2018-01-04 14:34:13,831 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2018-01-04 14:34:13,834 INFO b'Replace asset group p5dTexture'
2018-01-04 14:34:13,834 INFO b'Registered asset group: p5dTexture'
2018-01-04 14:34:14,236 INFO b'Got 67072 assets'
2018-01-04 14:34:14,476 INFO b'Client amKq_x3_GzfPEB2-AAAA connected on port 19849'
2018-01-04 14:34:14,491 INFO b'Loaded LightSpecs for 202 models'
2018-01-04 14:34:15,650 INFO b'THREE.WebGLRenderer 84'
2018-01-04 14:34:15,655 ERROR b'THREE.WebGLRenderer: Error creating WebGL context.'
2018-01-04 14:34:15,690 ERROR b'/home/user2/minos/server/node_modules/sstk/client/js/vendor/three/three.js:19603'
2018-01-04 14:34:15,690 ERROR b"\t\t\t\t\t\textension = gl.getExtension( 'WEBGL_depth_texture' ) || gl.getExtension( 'MOZ_WEBGL_depth_texture' ) || gl.getExtension( 'WEBKIT_WEBGL_depth_texture' );"
2018-01-04 14:34:15,690 ERROR b'\t\t\t\t\t\t               ^'
2018-01-04 14:34:15,690 ERROR b''
2018-01-04 14:34:15,690 ERROR b"TypeError: Cannot read property 'getExtension' of null"
2018-01-04 14:34:15,690 ERROR b'    at Object.get (/home/user2/minos/server/node_modules/sstk/client/js/vendor/three/three.js:19603:22)'
2018-01-04 14:34:15,690 ERROR b'    at new WebGLRenderer (/home/user2/minos/server/node_modules/sstk/client/js/vendor/three/three.js:20039:14)'
2018-01-04 14:34:15,690 ERROR b'    at OffscreenRenderer.Renderer.init (/home/user2/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:45250:39)'
2018-01-04 14:34:15,691 ERROR b'    at OffscreenRenderer.Renderer (/home/user2/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:45246:11)'
2018-01-04 14:34:15,691 ERROR b'    at new OffscreenRenderer (/home/user2/minos/server/node_modules/sstk/ssc/lib/OffscreenRenderer.js:26:23)'
2018-01-04 14:34:15,691 ERROR b'    at createSimulator (/home/user2/minos/server/server.js:41:18)'
2018-01-04 14:34:15,691 ERROR b'    at Socket.<anonymous> (/home/user2/minos/server/server.js:123:13)'
2018-01-04 14:34:15,691 ERROR b'    at emitTwo (events.js:126:13)'
2018-01-04 14:34:15,691 ERROR b'    at Socket.emit (events.js:214:7)'
2018-01-04 14:34:15,691 ERROR b'    at /home/user2/minos/server/node_modules/socket.io/lib/socket.js:503:12'
2018-01-04 14:34:15,709 INFO Finished simserver stderr
2018-01-04 14:34:15,709 INFO Finished simserver stdout
```

when I run the render.js in **sstk**, I got the same error info.

msavva, 2018-01-05:
Hi @XL2013,
The error you have encountered is a bit different -- it indicates that a WebGL context could not be initialized.  Typically, this error occurs when the code is run on a linux machine without an X server.  It is possible to run the code on such machines using the xvfb-run tool.  See https://github.com/stackgl/headless-gl#how-can-headless-gl-be-used-on-a-headless-linux-machine for details, and run `render.js` with the `xvfb-run` tool.  When doing this, please also check that the `LIBGL_ALWAYS_INDIRECT` environment variable is not defined.  If it is defined, you can use `unset LIBGL_ALWAYS_INDIRECT` to undefine it.  If you still have problems, please create a separate issue so we can follow up independently of @nzer0 's issue, which has a different root cause.

devendrachaplot, 2018-01-08:
I had the same issue as @nzer0 and the following from the comment [here](https://github.com/stackgl/headless-gl/issues/65#issuecomment-252742795) worked for me:

> Here's what worked for me in more detail:
> 
> Set up the system dependencies sudo apt-get install -y build-essential libxi-dev libglu1-mesa-dev libglew-dev (That was for Ubuntu, see https://github.com/stackgl/headless-gl#system-dependencies)
> Set up the dev environment, i.e. follow steps 1-6 of https://github.com/stackgl/headless-gl#how-should-i-set-up-a-development-environment-for-headless-gl (Note for step 6 I had to do npm run rebuild instead of npm run build)
> Copy and replace the built webgl.node file into your node module, e.g. cp headless-gl/build/Release/webgl.node node_modules/gl/build/Release/webgl.node

I have an Ubuntu machine with a NVidia GPU.

msavva, 2018-01-09:
@devendrachaplot thanks for posting the details!
@nzer0 do you still have the same issue?
@XL2013 if you are also still experiencing the error, please create a new issue with the details so we can debug it separately from this thread

nzer0, 2018-01-10:
I currently don't have time to test it but I will report here in the future if the solution by @devendrachaplot works

msavva, 2018-01-26:
@nzer0 I will close this issue since it appears the solution has worked for several others.  If you are still having problems after following the steps in this thread, please create a new issue.

closed 2018-01-26 by msavva

dangkh, 2018-03-06:
@msavva i had a same problem as @nzer0, same log file and tried the suggestion from @devendrachaplot but it didn't work.

msavva, 2018-03-06:
@dangkh Thanks for trying out MINOS.  Just to confirm, can you copy your log file as well?  Also, can you try using the latest version of node and building the server again? (see https://github.com/minosworld/minos/issues/12 -- this appears to have resolved the issue for others)

msavva, 2018-03-06:
This error appears to be coming from the socketIO client pip package.  We actually use our own fork of this package to fix some issues with binary mode transport over socket io.

It could be that for some reason when you installed the dependencies through `requirements.txt` the regular pip package got pulled down instead of our fork (it's listed as https://github.com/msavva/socketIO-client-2/zipball/master in the `requirements.txt` file).  Can you check that when you do `pip install -r requirements.txt` the package that is pulled down is indeed from that address?

dangkh, 2018-03-06:
@msavva It works. I do reinstall using pip install -r requirements.txt. Thanks.

ducha-aiki, 2018-10-11:
@msavva I have installed minos on server and now have similar problems. I have rebuilt headless-gl, but this does`t help. By the way, I have no directory "gl" under minos/minos/server/node_modules, so had to create it

>  ./benchmark.js 
Configuration { fileCache: { size: 50 },
  imageCache: { size: 200 },
  imageQueue: { concurrency: 4 },
  base_url: 'file:///home/dmishkin/work',
  assets_url:
   'file:///home/dmishkin/installing_scripts/minos/minos/server/node_modules/sstk/ssc/../server/static' }
p5dScene.0020d9dab70c6c8cfc0564c139c82dce
register ../server/static/data/suncg/suncg.planner5d.models.json
Replace asset group p5d
Registered asset group: p5d
Got 2623 assets
Loaded lightSpecs for 202 models.
register ../server/static/data/suncg/suncg.planner5d.textures.json
Replace asset group p5dTexture
Registered asset group: p5dTexture
Got 67072 assets
Segmentation fault (core dumped)

With xvfb-run , problems are different:

> xvfb-run -s "-ac -screen 0 1280x1024x24" ./benchmark.js 
> Configuration { fileCache: { size: 50 },
>   imageCache: { size: 200 },
>   imageQueue: { concurrency: 4 },
>   base_url: 'file:///home/dmishkin/work',
>   assets_url:
>    'file:///home/dmishkin/installing_scripts/minos/minos/server/node_modules/sstk/ssc/../server/static' }
> p5dScene.0020d9dab70c6c8cfc0564c139c82dce
> register ../server/static/data/suncg/suncg.planner5d.models.json
> Replace asset group p5d
> Registered asset group: p5d
> Got 2623 assets
> Loaded lightSpecs for 202 models.
> register ../server/static/data/suncg/suncg.planner5d.textures.json
> Replace asset group p5dTexture
> Registered asset group: p5dTexture
> Got 67072 assets
> X Error of failed request:  BadRequest (invalid request code or no such operation)
>   Major opcode of failed request:  151 (GLX)
>   Minor opcode of failed request:  206 ()
>   Serial number of failed request:  97
>   Current serial number in output stream:  97

And simserver.log:

> 2018-10-11 02:57:32,322 INFO b'Configuration { fileCache: { size: 50 },'
2018-10-11 02:57:32,323 INFO b'  imageCache: { size: 200 },'
2018-10-11 02:57:32,323 INFO b'  imageQueue: { concurrency: 4 },'
2018-10-11 02:57:32,323 INFO b'  assets_url:'
2018-10-11 02:57:32,323 INFO b"   'file:///home/dmishkin/installing_scripts/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
2018-10-11 02:57:32,586 INFO b'Waiting for client connection on port 27244'
2018-10-11 02:57:32,610 ERROR b'Cannot register unknown asset 2d3ds'
2018-10-11 02:57:32,611 INFO b'register ../server/static/data/matterport/matterport3d.json'
2018-10-11 02:57:32,613 INFO b'Registered asset group: mp3d'
2018-10-11 02:57:32,618 INFO b'Got 90 assets'
2018-10-11 02:57:32,685 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2018-10-11 02:57:32,686 INFO b'Replace asset group p5d'
2018-10-11 02:57:32,686 INFO b'Registered asset group: p5d'
2018-10-11 02:57:32,781 INFO b'Got 2623 assets'
2018-10-11 02:57:33,793 INFO b'Loaded lightSpecs for 202 models.'
2018-10-11 02:57:33,793 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2018-10-11 02:57:33,799 INFO b'Replace asset group p5dTexture'
2018-10-11 02:57:33,799 INFO b'Registered asset group: p5dTexture'
2018-10-11 02:57:34,062 INFO b'Got 67072 assets'
2018-10-11 02:57:34,245 INFO b'Client wO1iJycR-yRRa4HcAAAA connected on port 27244'
2018-10-11 02:57:38,755 INFO Finished simserver stderr
2018-10-11 02:57:38,755 INFO Finished simserver stdout

sumilator.log:

> 2018-10-11 02:57:31,663 INFO {'sim_id': 'sim00', 'machine': 'dmishkin', 'sim_git_hash': 'c66acdb', 'stk_git_hash': 'c66acdb'}
> 2018-10-11 02:57:31,685 INFO sim00:Starting sim server at /home/dmishkin/installing_scripts/minos/minos/server/server.js with port 27244
> 2018-10-11 02:57:34,271 INFO sim00:connect
> 2018-10-11 02:57:39,757 INFO sim00:disconnect
> 

and cli output: 

> 2018-10-11 02:57:31,685 INFO sim00:Starting sim server at /home/dmishkin/installing_scripts/minos/minos/server/server.js with port 27244
> 2018-10-11 02:57:34,271 INFO sim00:connect
> 2018-10-11 02:57:38,755 WARNING localhost:27244/socket.io [connection error] recv disconnected ([Errno 104] Connection reset by peer)
> 2018-10-11 02:57:39,756 WARNING localhost:27244/socket.io [connection error] recv disconnected ([Errno 104] Connection reset by peer)
> 2018-10-11 02:57:39,757 INFO sim00:disconnect
> 2018-10-11 02:57:39,759 WARNING localhost:27244/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=27244): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1539251859757-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f534722e240>: Failed to establish a new connection: [Errno 111] Connection refused',))
> 2018-10-11 02:57:40,762 WARNING localhost:27244/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=27244): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1539251860760-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f534722b240>: Failed to establish a new connection: [Errno 111] Connection refused',))
> 2018-10-11 02:57:41,765 WARNING localhost:27244/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=27244): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1539251861763-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f5347228278>: Failed to establish a new connection: [Errno 111] Connection refused',))

ducha-aiki, 2018-10-11:
I followed your advice here https://github.com/minosworld/minos/issues/6#issuecomment-356667922
and compiled sstk.

It crashes with same message:

> ./render.js
> Configuration { fileCache: { size: 50 },
>   imageCache: { size: 200 },
>   imageQueue: { concurrency: 4 },
>   base_url: 'file:///home/dmishkin/work',
>   assets_url:
>    'file:///home/dmishkin/installing_scripts/sstk/ssc/../server/static' }
> Segmentation fault (core dumped)

and in headless mode:

> xvfb-run -s "-ac -screen 0 1280x1024x24"  ./render.js
> Configuration { fileCache: { size: 50 },
>   imageCache: { size: 200 },
>   imageQueue: { concurrency: 4 },
>   base_url: 'file:///home/dmishkin/work',
>   assets_url:
>    'file:///home/dmishkin/installing_scripts/sstk/ssc/../server/static' }
> X Error of failed request:  BadRequest (invalid request code or no such operation)
>   Major opcode of failed request:  151 (GLX)
>   Minor opcode of failed request:  206 ()
>   Serial number of failed request:  97
>   Current serial number in output stream:  97

angelxuanchang, 2018-10-11:
Can you provide us with the GPU and GPU drivers you are using?  Also

ducha-aiki, 2018-10-11:
And before manual build of headless-gl, I get following in simserver.log:

> 2018-10-11 06:01:36,851 INFO b'Loaded LightSpecs for 202 models'
> 2018-10-11 06:01:44,643 INFO b'THREE.WebGLRenderer 84'
> 2018-10-11 06:01:44,646 ERROR b'THREE.WebGLRenderer: Error creating WebGL context.'
> 2018-10-11 06:01:44,663 ERROR b'/home/dmishkin/minos/minos/server/node_modules/sstk/client/js/vendor/three/three.js:19601'
> 2018-10-11 06:01:44,663 ERROR b"\t\t\t\t\t\textension = gl.getExtension( 'WEBGL_depth_texture' ) || gl.getExtension( 'MOZ_WEBGL_depth_texture' ) || gl.getExtension( 'WEBKIT_WEBGL_depth_texture' );"
> 2018-10-11 06:01:44,664 ERROR b'\t\t\t\t\t\t               ^'
> 2018-10-11 06:01:44,664 ERROR b''
> 2018-10-11 06:01:44,664 ERROR b"TypeError: Cannot read property 'getExtension' of null"

angelxuanchang, 2018-10-11:
Also please try to see what you get with `xvfb-run -s "-ac -screen 0 1280x1024x24" glxinfo` Some information [here](https://github.com/smartscenes/sstk/wiki/Headless-Rendering) may also be helpful

ducha-aiki, 2018-10-11:
lspci -vnn | grep VGA -A 12

> 04:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM200 [GeForce GTX TITAN X] [10de:17c2] (rev a1) (prog-if 00 [VGA controller])
> 	Subsystem: eVga.com. Corp. GM200 [GeForce GTX TITAN X] [3842:2992]
> 	Flags: bus master, fast devsel, latency 0, IRQ 136
> 	Memory at c4000000 (32-bit, non-prefetchable) [size=16M]
> 	Memory at 383fe0000000 (64-bit, prefetchable) [size=256M]
> 	Memory at 383ff0000000 (64-bit, prefetchable) [size=32M]
> 	I/O ports at 5000 [size=128]
> 	[virtual] Expansion ROM at c5000000 [disabled] [size=512K]
> 	Capabilities: <access denied>
> 	Kernel driver in use: nvidia
> 	Kernel modules: nvidiafb, nouveau, nvidia_396, nvidia_396_drm

nvidia-smi    

> +-----------------------------------------------------------------------------+
> | NVIDIA-SMI 396.45                 Driver Version: 396.45                    |
> |-------------------------------+----------------------+----------------------+
> | GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
> | Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
> |===============================+======================+======================|
> |   0  GeForce GTX TIT...  Off  | 00000000:04:00.0 Off |                  N/A |
> | 29%   69C    P2   138W / 250W |   5249MiB / 12212MiB |     85%      Default |
> +-------------------------------+----------------------+----------------------+
>

ducha-aiki, 2018-10-11:
xvfb-run -s "-ac -screen 0 1280x1024x24" glxinfo

> name of display: :99
> X Error of failed request:  BadValue (integer parameter out of range for operation)
>   Major opcode of failed request:  151 (GLX)
>   Minor opcode of failed request:  24 (X_GLXCreateNewContext)
>   Value in failed request:  0x0
>   Serial number of failed request:  35
>   Current serial number in output stream:  36

ducha-aiki, 2018-10-11:
@angelxuanchang thanks. According to your link https://github.com/smartscenes/sstk/wiki/Headless-Rendering , I am not able to get a WebGL context. xvfb-run doesn`t work as well somehow...
Do you have any suggestion for fixing this?

angelxuanchang, 2018-10-11:
I'm not sure.  Usually there needs to be some reinstallation of some nvidia drivers.  One thing to do is make sure they are up-to-date and make sure that you reboot (some posts seems to indicate some problems goes away after reboot).   

@msavva has more experience with getting these things to work.  Also I think there is some change he has for working with specific nvidia gpus that will bypass some of this headaches.  You can wait for him to have some suggestions.

The couple of things I know of are:
1. https://davidsanwald.github.io/2016/11/13/building-tensorflow-with-gpu-support.html
- this was in the wiki page I sent, so it worked for someone at some point.  But there is always new drivers and new combinations (which CUDA version are you using, btw?).  I think the key is the `–no-opengl-files` part for the nvidia drivers. 

2. https://askubuntu.com/questions/893922/ubuntu-16-04-gives-x-error-of-failed-request-badvalue-integer-parameter-out-o
- this one is more similar to the error you are getting (not quite the same) and it also says a crucial step is to say no to nvidia opengl

Not sure if either of those are things you want to try.

ducha-aiki, 2018-10-11:
@angelxuanchang the problem is that I cannot driver installation on cluster, nor even reboot.
I had an experience with `–no-opengl-files ` magic before, it helped on my laptop with dual graphic (internal + nvidia), but here I just have not rights to install drivers :(

angelxuanchang, 2018-10-11:
@ducha-aiki That is very sad.  I'm not sure how much we can help then.  Is there a machine with similar configuration that you can try things out on?

ducha-aiki, 2018-10-11:
@angelxuanchang no. On my desktop minos works out of the box without any problems...

msavva, 2018-10-11:
@ducha-aiki Please try to compile the version of headless-gl in this archive by following the instructions at the top of the README.md file:
https://www.dropbox.com/s/nlh8ofojipghc44/headless-gl-nvidia.zip?dl=1
This is an adapted version of the headless-gl library which uses nvidia's EGL to create an OpenGL context. 

The above archive already contains a compiled `build/Release/webgl.node` but that may not work for your OS and driver combination.  When you copy into the sstk directory, please look carefully to find where the existing `webgl.node` file is found and replace it (the file should exist after doing `npm install` in the minos server directory). Unless you override the existing webgl.node it will likely not work.

Please let us know if you run into any issues.

ducha-aiki, 2018-10-11:
@msavva thank you, I will try it!

ducha-aiki, 2018-10-11:
@msavva That is real miracle. Thank you so much, it works now!