minosworld/minos #4: Get absolute depth

fuenwang, 2018-01-08:
Hi,
Thanks for releasing this environment. When I do some test in MINOS, I found the depth map
from server seems to be normalized to 0~255. Is it possible to get absolute depth map?Or I need to do some hacking?

Thanks.

angelxuanchang, 2018-01-08:
The raw values returned for the depth map should be in meters.    It is naively normalized in the pygame_client.py tool for display purposes. 

The camera parameters for both the color and depth sensor is specified in a configuration file. 
Please see https://github.com/minosworld/minos/blob/python-repackage/minos/config/sensors.yml
for the default configuration.  The depth sensor is configured to have a near of 0.1 meters and a far of 20 meters.  Both the color and depth sensors are configured with a default vertical fov of 45 degrees.  The horizontal fov is a function of the aspect ratio and the vertical fov.

fuenwang, 2018-01-09:
Thank you so much! I may just misunderstand the data I get, sorry for bothering you.

closed 2018-01-09 by fuenwang

romi-seecs, 2019-07-12:
@fuenwang Hi! Could you please tell me, in MINOS where can we get the depth map?

romi-seecs, 2019-07-14:
> The raw values returned for the depth map should be in meters. It is naively normalized in the pygame_client.py tool for display purposes.
>
> The camera parameters for both the color and depth sensor [...]

@angelxuanchang how can we get the raw values for depth?