minosworld/minos #5: running the demo.py on headless Linux machine with GPU

kojimano, 2018-01-09:
I am trying to run MINOS  pygame_client on headless Linux machine with NVIDIA GPU card.
I have two settings, which one works fine and the other have the exact same issue with Linux Machine without Xserver of @XL2013 in Problem with running the deomo.py.

The command I run was below
NODE_BASE_URL="somepath" xvfb-run -s "-ac -screen 0 1280x1024x24" python3 demo.py --env_config objectgoal_suncg_mf

It worked perfectly fine when I run the command on the node of headless Linux server I am sshing, and there appeared the results of random agents exploring. 
However, when I run the same command on the node of headless Linux server with NVIDIA GPU card, I cannot have WebGLRenderer correctly working and the log of simserver.log is attached below, which is equivalent to that of @XL2013. 

If anyone has any ideas to get this working, that is very appreciated.

`2018-01-09 06:30:07,208 INFO b'Configuration { fileCache: { size: 50 },'
  2 2018-01-09 06:30:07,208 INFO b'  imageCache: { size: 200 },'
  3 2018-01-09 06:30:07,208 INFO b'  imageQueue: { concurrency: 4 },'
  4 2018-01-09 06:30:07,208 INFO b"  base_url: '/scratch/jiadeng_fluxoe/kojimano/dataset/SUNGG',"
  5 2018-01-09 06:30:07,208 INFO b"  assets_url: 'file:///home/kojimano/MINOS/minos/server/node_modules/sstk/ssc/../server/static' }"
  6 2018-01-09 06:30:08,493 INFO b'Waiting for client connection on port 24845'
  7 2018-01-09 06:30:08,522 ERROR b'Cannot register unknown asset 2d3ds'
  8 2018-01-09 06:30:08,522 INFO b'register ../server/static/data/matterport/matterport3d.json'
  9 2018-01-09 06:30:08,533 INFO b'Registered asset group: mp3d'
 10 2018-01-09 06:30:08,538 INFO b'Got 90 assets'
 11 2018-01-09 06:30:08,616 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
 12 2018-01-09 06:30:08,650 INFO b'Replace asset group p5d'
 13 2018-01-09 06:30:08,650 INFO b'Registered asset group: p5d'
 14 2018-01-09 06:30:08,735 INFO b'Got 2623 assets'
 15 2018-01-09 06:30:09,805 INFO b'Loaded lightSpecs for 202 models.'
 16 2018-01-09 06:30:09,806 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
 17 2018-01-09 06:30:09,847 INFO b'Replace asset group p5dTexture'
 18 2018-01-09 06:30:09,847 INFO b'Registered asset group: p5dTexture'
 19 2018-01-09 06:30:10,084 INFO b'Got 67072 assets'
 20 2018-01-09 06:30:10,443 INFO b'Client nwy9whNKE2uGLEClAAAA connected on port 24845'
 21 2018-01-09 06:30:10,448 INFO b'Loaded LightSpecs for 202 models'
 22 2018-01-09 06:30:10,590 INFO b'THREE.WebGLRenderer 84'
 23 2018-01-09 06:30:10,591 ERROR b'THREE.WebGLRenderer: Error creating WebGL context.'
 24 2018-01-09 06:30:10,611 ERROR b'/home/kojimano/MINOS/minos/server/node_modules/sstk/client/js/vendor/three/three.js:19603'
 25 2018-01-09 06:30:10,611 ERROR b"\t\t\t\t\t\textension = gl.getExtension( 'WEBGL_depth_texture' ) || gl.getExtension( 'MOZ_WEBGL_depth_texture' ) || gl.getExtension( 'WEBKIT_WEBGL_depth_texture' );"
 26 2018-01-09 06:30:10,611 ERROR b'\t\t\t\t\t\t               ^'
 27 2018-01-09 06:30:10,611 ERROR b''
 28 2018-01-09 06:30:10,611 ERROR b"TypeError: Cannot read property 'getExtension' of null"
 29 2018-01-09 06:30:10,611 ERROR b'    at Object.get (/home/kojimano/MINOS/minos/server/node_modules/sstk/client/js/vendor/three/three.js:19603:22)'
 30 2018-01-09 06:30:10,611 ERROR b'    at new WebGLRenderer (/home/kojimano/MINOS/minos/server/node_modules/sstk/client/js/vendor/three/three.js:20039:14)'
 31 2018-01-09 06:30:10,611 ERROR b'    at OffscreenRenderer.Renderer.init (/home/kojimano/MINOS/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:45250:39)'
 32 2018-01-09 06:30:10,611 ERROR b'    at OffscreenRenderer.Renderer (/home/kojimano/MINOS/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:45246:11)'
 33 2018-01-09 06:30:10,611 ERROR b'    at new OffscreenRenderer (/home/kojimano/MINOS/minos/server/node_modules/sstk/ssc/lib/OffscreenRenderer.js:26:23)'
 34 2018-01-09 06:30:10,611 ERROR b'    at createSimulator (/home/kojimano/MINOS/minos/server/server.js:41:18)'
 35 2018-01-09 06:30:10,611 ERROR b'    at Socket.<anonymous> (/home/kojimano/MINOS/minos/server/server.js:113:13)'
 36 2018-01-09 06:30:10,611 ERROR b'    at emitTwo (events.js:126:13)'
 37 2018-01-09 06:30:10,612 ERROR b'    at Socket.emit (events.js:214:7)'
 38 2018-01-09 06:30:10,612 ERROR b'    at /home/kojimano/MINOS/minos/server/node_modules/socket.io/lib/socket.js:503:12'
 39 2018-01-09 06:30:10,640 INFO Finished simserver stdout
 40 2018-01-09 06:30:10,640 INFO Finished simserver stderr
`

msavva, 2018-01-09:
Hi @kojimano,
Thanks for trying out MINOS!

To help us debug this issue can you provide a few more details about the difference between the two scenarios?  Specifically, when you say "the same server with NVIDIA GPU card", are you still remotely logged in over ssh and running the same command as you quoted?  Can you also check that the `LIBGL_ALWAYS_INDIRECT` environment variable is not set?  (refer to https://github.com/minosworld/minos/issues/3#issuecomment-355613649)

kojimano, 2018-01-09:
Thank you for the quick response and sorry for some confusion.
As I investigate more the issue has the nothing to do with GPU, I assume. (I also checked LIBGL_ALWAYS_INDIRECT is unset.)

I am sshing to the university-wide server in both of the failure case and success case.
Basically, the command I quoted goes successfully when I run the command on login-node of the server. However, when I request another node on the server (no longer login-node) and do experiments (with or without GPU). 
The only difference of two nodes I can think of is 1. the spec of CPU. 2. nvidia driver are installed in the failed node of the server and not installed in the login-node of the server.

The command **NODE_BASE_URL="somepath" xvfb-run -s "-ac -screen 0 1280x1024x24" python3 demo.py --env_config objectgoal_suncg_mf** fails with the following log file (sorry for the confusion the one I attached above was from the different log folder.)
0. simserver.log under /gym
>   1 2018-01-09 14:30:44,836 ERROR b'Cannot register unknown asset 2d3ds'
  2 2018-01-09 14:30:46,601 ERROR b'THREE.WebGLRenderer: Error creating WebGL context.'
  3 2018-01-09 14:30:46,619 ERROR b'/home/kojimano/MINOS/minos/server/node_modules/sstk/client/js/vendor/three/three.js:19603'
  4 2018-01-09 14:30:46,619 ERROR b"\t\t\t\t\t\textension = gl.getExtension( 'WEBGL_depth_texture' ) || gl.getExtension( 'MOZ_WEBGL_depth_texture' ) || gl.getExtension( 'WEBKIT_WEBGL    _depth_texture' );"
  5 2018-01-09 14:30:46,619 ERROR b'\t\t\t\t\t\t               ^'
  6 2018-01-09 14:30:46,619 ERROR b''
  7 2018-01-09 14:30:46,619 ERROR b"TypeError: Cannot read property 'getExtension' of null"
  8 2018-01-09 14:30:46,619 ERROR b'    at Object.get (/home/kojimano/MINOS/minos/server/node_modules/sstk/client/js/vendor/three/three.js:19603:22)'
  9 2018-01-09 14:30:46,620 ERROR b'    at new WebGLRenderer (/home/kojimano/MINOS/minos/server/node_modules/sstk/client/js/vendor/three/three.js:20039:14)'
 10 2018-01-09 14:30:46,620 ERROR b'    at OffscreenRenderer.Renderer.init (/home/kojimano/MINOS/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:45250:39)'
 11 2018-01-09 14:30:46,620 ERROR b'    at OffscreenRenderer.Renderer (/home/kojimano/MINOS/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:45246:11)'
 12 2018-01-09 14:30:46,620 ERROR b'    at new OffscreenRenderer (/home/kojimano/MINOS/minos/server/node_modules/sstk/ssc/lib/OffscreenRenderer.js:26:23)'
 13 2018-01-09 14:30:46,620 ERROR b'    at createSimulator (/home/kojimano/MINOS/minos/server/server.js:41:18)'
 14 2018-01-09 14:30:46,620 ERROR b'    at Socket.<anonymous> (/home/kojimano/MINOS/minos/server/server.js:113:13)'
 15 2018-01-09 14:30:46,620 ERROR b'    at emitTwo (events.js:126:13)'
 16 2018-01-09 14:30:46,620 ERROR b'    at Socket.emit (events.js:214:7)'
 17 2018-01-09 14:30:46,620 ERROR b'    at /home/kojimano/MINOS/minos/server/node_modules/socket.io/lib/socket.js:503:12'

1. The CPU spec of failed-node 
> Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                16
On-line CPU(s) list:   0-15
Thread(s) per core:    1
Core(s) per socket:    8
Socket(s):             2
NUMA node(s):          2
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 79
Model name:            Intel(R) Xeon(R) CPU E5-2620 v4 @ 2.10GHz
Stepping:              1
CPU MHz:               2299.828
CPU max MHz:           3000.0000
CPU min MHz:           1200.0000
BogoMIPS:              4200.16
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              20480K
NUMA node0 CPU(s):     0-7
NUMA node1 CPU(s):     8-15
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 fma cx16 xtpr pdcm pcid dca sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb cat_l3 cdp_l3 invpcid_single intel_pt spec_ctrl ibpb_support tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cqm rdt_a rdseed adx smap xsaveopt cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts

2. CPU spec of the login-node

> Architecture: x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                16
On-line CPU(s) list:   0-15
Thread(s) per core:    1
Core(s) per socket:    8
Socket(s):             2
NUMA node(s):          2
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 79
Model name:            Intel(R) Xeon(R) CPU E5-2620 v4 @ 2.10GHz
Stepping:              1
CPU MHz:               2299.828
CPU max MHz:           3000.0000
CPU min MHz:           1200.0000
BogoMIPS:              4200.16
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              20480K
NUMA node0 CPU(s):     0-7
NUMA node1 CPU(s):     8-15
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aper [... cut ...] 3dnowprefetch epb cat_l3 cdp_l3 invpcid_single intel_pt spec_ctrl ibpb_support tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm cqm rdt_a rdseed adx smap xsaveopt cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local dtherm ida arat pln pts

My apology if these information are not relevant to debug and this case is too specific, and I will also try to figure out the difference of these nodes by contacting the university-wide server service meanwhile.

msavva, 2018-01-09:
This does indeed seem like a machine configuration difference.  Perhaps a more minimal test to try would be something like:
`xvfb-run -s "-ac -screen 0 1280x1024x24" xvinfo` or `xvfb-run -s "-ac -screen 0 1280x1024x24" glxinfo` if `glxinfo` is available.

playerkk, 2018-01-16:
Hi @msavva 
I have exactly the same problem on a remote Linux machine with NVIDIA GPUs. 

xvfb-run -s "-ac -screen 0 1280x1024x24" xvinfo gives
X-Video Extension version 2.2
screen #0
 no adaptors present

xvfb-run -s "-ac -screen 0 1280x1024x24" glxinfo gives
name of display: :99
Error: couldn't find RGB GLX visual or fbconfig
Error: couldn't find RGB GLX visual or fbconfig

Any suggestions would be appreciated.

kvas7andy, 2018-01-18:
Hi @msavva and @kojimano 
Have the same issue as @kojimano, having NVIDIA GPU on remote server and connecting to it via ssh.

xvfb-run -s "-ac -screen 0 1280x1024x24" xvinfo gives

> X-Video Extension version 2.2
> screen #0
> no adaptors present

 xvfb-run -s "-ac -screen 0 1280x1024x24" glxinfo

> name of display: :99
> X Error of failed request:  BadValue (integer parameter out of range for operation)
>   Major opcode of failed request:  151 (GLX)
>   Minor opcode of failed request:  24 (X_GLXCreateNewContext)
>   Value in failed request:  0x0
>   Serial number of failed request:  35
>   Current serial number in output stream:  36

kojimano, 2018-01-18:
I believe @playerkk 's issue is same as mine. But not quite sure @ kvas7andy 's come from the same reason. All the issue come from failing to initialize WebGL contents. Whether to see you successfully initialize WebGL you can run the script below by simple "node test.js". There are mainly two reasons why this is happening 1. we did not  have GLX extension installed along with nvidia driver. (No GLX extension found if you do glxinfo) 2. You are on vnc session, and that is causing the wierd issue. (glxinfo outputs something fine but still WebGL contents will not be initialized). I have currently no clue how to deal with Case 2, but for Case 1, there are similar discussion going on [here.](https://github.com/stackgl/headless-gl/issues/76) (even though it is tricky to install additional packages on linux without root permission, and I haven't made this successfully yet. Some package managers or linuxbrew might be the solution).
 
//test.js
var width   = 100
var height  = 100
var gl = require('gl')(width, height, { preserveDrawingBuffer: true })
gl.clearColor(1, 0, 0, 1)
gl.clear(gl.COLOR_BUFFER_BIT)
var pixels = new Uint8Array(width * height * 4)
gl.readPixels(0, 0, width, height, gl.RGBA, gl.UNSIGNED_BYTE, pixels)
process.stdout.write(['P3\n# gl.ppm\n', width, " ", height, '\n255\n'].join(''))
for(var i=0; i<pixels.length; i+=4) {
  for(var j=0; j<3; ++j) {
     process.stdout.write(pixels[i+j] + ' ')
  }
}
 Let's help out together to solve this issue. I really need to get this working on GPU to save my time.

kojimano, 2018-01-18:
@playerkk  This article might give us some clues. https://davidsanwald.github.io/2016/11/13/building-tensorflow-with-gpu-support.html

playerkk, 2018-01-19:
@kojimano I've solved the problem. Thanks a lot for sharing the article. I also find another one helpful https://gist.github.com/wangruohui/df039f0dc434d6486f5d4d098aa52d07

In summary, add --no-opengl-libs switch is the key to install both NVIDIA driver and CUDA.

kojimano, 2018-01-19:
Thanks @playerkk. I will close this issue since the solution now seems obvious. @kvas7andy please reopen another issue if you cannot solve your problem with the suggestion of @playerkk and mine.

closed 2018-01-19 by kojimano

msavva, 2018-01-23:
Hi @kojimano , @playerkk , @kvas7andy 

Thank you for going through the process of debugging this issue and reporting back to support the community!

If any of you are willing to create a pull request to add a short summary of this issue and its resolution  to the repository README.md that would be great! We can acknowledge your effort and make this part of a "FAQ & Troubleshooting Guide" section in the README so that other users may benefit as well.

kojimano, 2018-01-23:
Hi @msavva ! I am happy to this, unless someone else desires to do.

playerkk, 2018-01-23:
Hi @kojimano , it would be great if you could do it. Thanks a lot.