minosworld/minos #6: Can not connect with simulator

pammirato, 2018-01-09:
Hi.  When I run the basic command, it seems I can not connect with the simulator. Not sure what is happening. I believe the error comes on line 249 in lib/Simulator.py.

Thank you.

`$ python3 -m tools.pygame_client `

`2018-01-09 14:11:00,928 INFO {'sim_id': 'sim00', 'stk_git_hash': '03855da', 'sim_git_hash': '03855da', 'machine': 'bvision4.cs.unc.edu'}
Starting simulator...
2018-01-09 14:11:00,953 INFO sim00:Starting sim server at /playpen/ammirato/Documents/Minos/code/minos/server/server.js with port 47129
2018-01-09 14:11:01,964 INFO sim00:sim server has exited with rv 1
2018-01-09 14:11:01,973 WARNING localhost:47129/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=47129): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1515525061965-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fd55c02c550>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-01-09 14:11:02,978 WARNING localhost:47129/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=47129): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1515525062974-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fd55c02c8d0>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-01-09 14:11:03,983 WARNING localhost:47129/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=47129): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1515525063979-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fd55c0312b0>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-01-09 14:11:04,988 WARNING localhost:47129/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=47129): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1515525064984-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fd55c02c668>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-01-09 14:11:05,992 WARNING localhost:47129/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=47129): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1515525065989-0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fd55c02cef0>: Failed to establish a new connection: [Errno 111] Connection refused',))
`

msavva, 2018-01-09:
Hi @pammirato,
Thanks for trying out MINOS!
Can you look at the simulator server log files (see https://github.com/minosworld/minos/issues/3#issuecomment-354215713) and quote any error you see in the logs?

pammirato, 2018-01-09:
**simserver.log:**

`2018-01-09 14:11:00,963 INFO Finished simserver stdout
2018-01-09 14:11:00,963 INFO Finished simserver stderr
2018-01-09 14:11:01,964 INFO Waiting for simserver stdout to finish
2018-01-09 14:11:01,964 INFO Waiting for simserver stderr to finish`

**simulator.log:**

`2018-01-09 14:11:00,928 INFO {'sim_id': 'sim00', 'stk_git_hash': '03855da', 'sim_git_hash': '03855da', 'machine': 'bvision4.cs.unc.edu'}
2018-01-09 14:11:00,953 INFO sim00:Starting sim server at /playpen/ammirato/Documents/Minos/code/minos/server/server.js with port 47129
2018-01-09 14:11:01,964 INFO sim00:sim server has exited with rv 1
2018-01-09 14:11:06,819 INFO sim00:Stopping the simulator
2018-01-09 14:11:06,819 INFO Counter()
2018-01-09 14:11:06,819 INFO sim00:Stopping child servers
2018-01-09 14:11:06,819 INFO sim00:Stopped child servers
2018-01-09 14:11:06,820 INFO sim00:Simulator killed.`

msavva, 2018-01-09:
The logs seem to indicate that the server module could not boot up.  Can you provide output for the following in the same environment within which you are running MINOS:
1. `node -v`
2. `npm -v`
3. `npm install` in `server` module
4. `./server.js` in `server` module

Can you also confirm that all the steps in https://github.com/minosworld/minos#installing completed without errors?  In particular, the above issue might be caused by an error that occurred in step 3 (building the server module using npm).

msavva, 2018-01-09:
One additional note: if you have already run `npm install` and see a message about removed packages when re-running it, please delete the `node_modules` directory inside `server` and then try running `npm install` again.

pammirato, 2018-01-10:
1.  `node -v` 
v8.9.4

2. `npm -v`
5.6.0

3. `npm install`
$ npm install

> node-expat@2.3.16 install /playpen/ammirato/Documents/Minos/code/minos/server/node_modules/node-expat
> node-gyp rebuild

make: Entering directory `/playpen/ammirato/Documents/Minos/code/minos/server/node_modules/node-expat/build'
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmlparse.o
../deps/libexpat/lib/xmlparse.c: In function 'gather_time_entropy':
../deps/libexpat/lib/xmlparse.c:780:7: warning: variable 'gettimeofday_res' set but not used [-Wunused-but-set-variable]
   int gettimeofday_res;
       ^
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmltok.o
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmlrole.o
  AR(target) Release/obj.target/deps/libexpat/libexpat.a
  COPY Release/libexpat.a
  CXX(target) Release/obj.target/node_expat/node-expat.o
  SOLINK_MODULE(target) Release/obj.target/node_expat.node
  COPY Release/node_expat.node
make: Leaving directory `/playpen/ammirato/Documents/Minos/code/minos/server/node_modules/node-expat/build'

> sstk@0.5.0 postinstall /playpen/ammirato/Documents/Minos/code/minos/server/node_modules/sstk
> npm run bootstrap

> sstk@0.5.0 bootstrap /playpen/ammirato/Documents/Minos/code/minos/server/node_modules/sstk
> lerna bootstrap --hoist

lerna info version 2.5.1
lerna info Bootstrapping 3 packages
lerna info lifecycle preinstall
lerna info Installing external dependencies
lerna info hoist Installing hoisted dependencies into root
lerna info hoist Pruning hoisted dependencies
lerna info hoist Finished pruning hoisted dependencies
lerna info hoist Finished installing in root
lerna info Symlinking packages and binaries
lerna info lifecycle postinstall
lerna info lifecycle prepublish
lerna info lifecycle prepare
lerna success Bootstrapped 3 packages
added 476 packages in 95.606s

4. `./server.js`

 $ ./server.js 
Configuration { fileCache: { size: 50 },
  imageCache: { size: 200 },
  imageQueue: { concurrency: 4 },
  base_url: 'file:///home/ammirato/work',
  assets_url: 'file:///playpen/ammirato/Documents/Minos/code/minos/server/node_modules/sstk/ssc/../server/static' }
Waiting for client connection on port 1234
Cannot register unknown asset 2d3ds
register ../server/static/data/matterport/matterport3d.json
Registered asset group: mp3d
Got 90 assets
register ../server/static/data/suncg/suncg.planner5d.models.json
Replace asset group p5d
Registered asset group: p5d
Got 2623 assets
Loaded lightSpecs for 202 models.
register ../server/static/data/suncg/suncg.planner5d.textures.json
Replace asset group p5dTexture
Registered asset group: p5dTexture
Got 67072 assets
Loaded LightSpecs for 202 models

5. `./server.js` then just hangs there. Running `python3 -m tools.pygame_client` while server.js hangs gives the same error message as before. 

6. One other thing to note is that when running `source ./bashrc` after install `nvm` I get:

nvm is not compatible with the npm config "prefix" option: currently set to "/playpen/ammirato/.nvm/versions/node/v8.9.4"
Run `npm config delete prefix` or `nvm use --delete-prefix v8.9.4 --silent` to unset it

I just added `nvm use --delete-prefix v8.9.4` to my ~/.bashrc and it seems to fix this issue

msavva, 2018-01-10:
The output of steps 1 - 5 are as expected. `./server.js` appears to boot up successfully and wait for a connection on port 1234.  The `python3 -m tools.pygame_client` command should boot up a server on a random port and attempt to connect to it, but the log files you copied don't indicate any connection.  The issue in 6. appears unrelated.

Can you try a couple more things:
1. Run `./benchmark.js` in the server module.  This should take a minute or so and report frame rate statistics at the end.  If this runs successfully then the server module is working properly by itself, and the issue must be with the client module not being able to establish a connection to the server.
2. If `./benchmark.js` fails, try checking out http://github.com/smartscenes/sstk which is the core library on which the server module depends, build it using the `build.sh` script in the root of the repository, go into the `ssc` directory, and try to run `./render.js`. This is a more minimal test of the rendering functionality that the server module requires.  If this step fails, we can isolate the problem to the core library and an inability to render simulation frames in your environment.

pammirato, 2018-01-10:
**tldr**
It works following [https://github.com/stackgl/headless-gl/issues/65](https://github.com/stackgl/headless-gl/issues/65). The steps from the last post. 

1. `./benchmark.js` fails with: 

$ ./benchmark.js 
Configuration { fileCache: { size: 50 },
  imageCache: { size: 200 },
  imageQueue: { concurrency: 4 },
  base_url: 'file:///home/ammirato/work',
  assets_url: 'file:///playpen/ammirato/Documents/Minos/code/minos/server/node_modules/sstk/ssc/../server/static' }
p5dScene.0020d9dab70c6c8cfc0564c139c82dce
register ../server/static/data/suncg/suncg.planner5d.models.json
Replace asset group p5d
Registered asset group: p5d
Got 2623 assets
Loaded lightSpecs for 202 models.
register ../server/static/data/suncg/suncg.planner5d.textures.json
Replace asset group p5dTexture
Registered asset group: p5dTexture
Got 67072 assets
node: symbol lookup error: /playpen/ammirato/Documents/Minos/code/minos/server/node_modules/sstk/node_modules/gl/build/Release/webgl.node: undefined symbol: _Z15XextFindDisplayP15_XExtensionInfoP9_XDisplay

2. I was able to successfully run the code at [ http://github.com/smartscenes/sstk]( http://github.com/smartscenes/sstk) in the same environment. I was able to access the server through http://localhost:8010 in my browser. 

3. I tried to rebuild sstk by running `./build.sh` in the minos project, but that did not work.

4. I followed the steps from [https://github.com/stackgl/headless-gl/issues/65](https://github.com/stackgl/headless-gl/issues/65) and it works. 

thanks for your help!

msavva, 2018-01-10:
Great to hear.  It appears in the end this was the same issue as https://github.com/minosworld/minos/issues/3 which is resolved by rebuilding headless-gl.

closed 2018-01-10 by msavva