minosworld/minos #7: Error: Cannot find module '../ssc/stk-ssc'

devendrachaplot, 2018-01-14:
I am trying to run the 'server/visualize_path.js' script by running the following command
```
NODE_BASE_URL=~/work node --max-old-space-size=2048 visualize_path.js --compress_png --config_file config/visualize_traces.json --color beige --output_dir simpath --input ./data/episode_states.mp3d.csv --ids dc251799d2b7cbc1fee86fbf524f2bec --allow_diag false
```
 and I am getting this error:

```
module.js:540
    throw err;
    ^

Error: Cannot find module '../ssc/stk-ssc'
    at Function.Module._resolveFilename (module.js:538:15)
    at Function.Module._load (module.js:468:25)
    at Module.require (module.js:587:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/Labs/DeepRL/Environments/minos/visualize_path.js:6:11)
    at Module._compile (module.js:643:30)
    at Object.Module._extensions..js (module.js:654:10)
    at Module.load (module.js:556:32)
    at tryModuleLoad (module.js:499:12)
    at Function.Module._load (module.js:491:3)
```

Any idea what's wrong?

msavva, 2018-01-14:
Hi @devendrachaplot,
Thank you for reporting this -- the error is due to incorrect require paths in the script.  The issue should be fixed with commit  d6512d7 .  Please pull the master branch and let us know if you have any problems.

devendrachaplot, 2018-01-14:
That worked. Thanks.

closed 2018-01-14 by devendrachaplot