minosworld/minos #10: Bug: fail to run gym/demo.py with advanced args

playerkk, 2018-01-22:
When I replaced the add_sim_args_basic with add_sim_args, I got the following error
>** (demo.py:4496): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-NpYiBpNtPY: Connection refused
/home/hzjiang/.local/lib/python3.4/site-packages/matplotlib/backends/backend_gtk3agg.py:17: UserWarning: The Gtk3Agg backend is known to not work on Python 3.x with pycairo. Try installing cairocffi.
  "The Gtk3Agg backend is known to not work on Python 3.x with pycairo. "
val:7 episodes
test:11 episodes
train:35 episodes
Traceback (most recent call last):
  File "gym/demo.py", line 46, in <module>
    main()
  File "gym/demo.py", line 42, in main
    run_gym(sim_args)
  File "gym/demo.py", line 16, in run_gym
    env.configure(sim_args)
  File "/data/hzjiang/Code/minos/gym/gym_minos/envs/indoor_env.py", line 25, in configure
    self._sim = RoomSimulator(sim_args)
  File "/data/hzjiang/Code/minos/lib/RoomSimulator.py", line 50, in __init__
    self.sim = Simulator(params)
  File "/data/hzjiang/Code/minos/lib/Simulator.py", line 62, in __init__
    if 'port' not in params.audio or params.audio.port is None:
TypeError: argument of type 'bool' is not iterable
Exception ignored in: <bound method Simulator.__del__ of <lib.Simulator.Simulator object at 0x7f5326375cf8>>
Traceback (most recent call last):
  File "/data/hzjiang/Code/minos/lib/Simulator.py", line 164, in __del__
    if not self.killed:
AttributeError: 'Simulator' object has no attribute 'killed'

When I explicitly set params.audio = {} in Simulator's constructor following the default setting of add_sim_args_basic, I was unable to connect to the server.

msavva, 2018-01-23:
Hi @playerkk 
Thank you for trying out MINOS and for contributing to resolving issues!

Can you please provide some more context about your use case, and how you are replacing add_sim_args_basic with add_sim_args?

playerkk, 2018-01-23:
I simply replaced add_sim_args_basic in https://github.com/minosworld/minos/blob/master/gym/demo.py#L34 by add_sim_args, imported from config.sim_args.

msavva, 2018-01-26:
@playerkk ,

Apologies for the long delay.  We have fixed this issue (this was due to several arguments in sim_args that have been deprecated).  Let us know if you have any other issues.

closed 2018-01-26 by msavva

playerkk, 2018-01-26:
It works well right now. Thanks for fixing this.

playerkk, 2018-01-27:
Unfortunately, I am unable to run python3 -m tools.pygame_client --empty_room after updating.

msavva, 2018-01-28:
@playerkk ,
Can you send us the errors you observe when running the above command? In particular, the log files in `logs/<timestamp>/simserver.log` and `logs/<timestamp>/simulator.log` might contain relevant information about what is going wrong.

playerkk, 2018-01-29:
python3 -m tools.pygame_client --empty_room gave me
>Traceback (most recent call last):
  File "/usr/lib/python3.4/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.4/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/data/hzjiang/Code/minos/tools/pygame_client.py", line 453, in <module>
    main()
  File "/data/hzjiang/Code/minos/tools/pygame_client.py", line 409, in main
    args = parse_sim_args(parser)
  File "/data/hzjiang/Code/minos/config/sim_args.py", line 195, in parse_sim_args
    args.observations = {'color': True, 'depth': args.depth, 'forces': args.forces, 'audio': args.audio}
AttributeError: 'Namespace' object has no attribute 'audio'

I think this is related to the last update.

msavva, 2018-01-31:
@playerkk 
Thanks for catching this, the last update did indeed break some of the parameter parsing.  This should now be fixed.  Let us know if you still see issues.

playerkk, 2018-02-01:
@msavva Thanks for the fix. Everything works well expect a minor point. The --navmap switch is not supported. But simply uncommenting https://github.com/minosworld/minos/blob/master/config/sim_args.py#L83-L85 fixes this problem.

commit e9692074 (2018-01-26): remove another old sim_args argument. Fixes #10 (when using sim_args instead of sim_args_basic with gym demo)