minosworld/minos #11: Seting up the initial vertical view (angle) in the indoor environment of SUNCG

kojimano, 2018-01-23:
When I add 'LookUp' and 'LookDown' to the action space of the agent in the indoor environment of SUNCG, the agents often start their views looking up directly at the ceiling. I would like to fix this vertical view angle at the starting point. Are there any convenient ways to this. (Angle argument in episode_states.suncg.csv under data folder is the Horizontal angle at the starting point but I would like to fix the Vertical Angle.)

Thank you for your help in advance.

P.S.
I understand this is a technically tricky to do. But agents actually can look through the wall if they rotate themselves near the wall with some vertical view angle. This is nicer to get fixed in the future, since this will confuzes the agents learned behaivor.

msavva, 2018-01-26:
@kojimano Thank you for reporting this -- unfortunately we do not currently have API support for resetting the vertical angle.  We will add this functionality.

With regards to the "looking through the wall" issue, that is most likely due to clipping of the wall plane when it is closer to the agent than the camera `near` plane value (see https://github.com/minosworld/minos/blob/python-repackage/minos/config/sensors.yml#L25).  You can try setting a `near` value closer to zero if you want to avoid this issue.

msavva, 2018-03-17:
@kojimano the recent updates introduce the ability to make calls of the form `env.simulator.move_to(tilt=0)` where `tilt` is the angle from the horizontal in radians.  Please try this and let us know if you experience any issues.

closed 2018-03-23 by kojimano