minosworld/minos #12: Problem with running SUNCG models

hzyjerry, 2018-01-25:
Hi,

When I run SUNCG models using command `python3 -m tools.pygame_client --source suncg --scene_ids xxx`

I get warning:
`WARNING localhost:52755/socket.io [connection error] recv disconnected (Connection is already closed.)`  
then repeatedly warnings of 
`WARNING localhost:52755/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=52755): Max retries exceeded with url: /socket.io/?t=1516855384738-0&transport=polling&EIO=3 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x108be3780>: Failed to establish a new connection: [Errno 61] Connection refused',))`

This **only** happens with SUNCG models, not with MP3D models.

simulator.log seems fine. Inside simserver.log:
> 2018-01-24 20:43:12,374 INFO b'Loading scene  suncg.0004dd3cb11e50530676f77b55262d38'
> 2018-01-24 20:43:12,391 ERROR b'/Users/cvgl/Desktop/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:29694'
> 2018-01-24 20:43:12,392 ERROR b'\t        assetType = assetGroup.type;'
> 2018-01-24 20:43:12,392 ERROR b'\t                               ^'
> 2018-01-24 20:43:12,392 ERROR b''
> 2018-01-24 20:43:12,392 ERROR b"TypeError: Cannot read property 'type' of undefined"
> 2018-01-24 20:43:12,392 ERROR b'    at AssetManager.loadAssetAsScene (/Users/cvgl/Desktop/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:29694:33)'
> 2018-01-24 20:43:12,392 ERROR b'    at Simulator.__loadScene (/Users/cvgl/Desktop/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:104631:22)'
> 2018-01-24 20:43:12,392 ERROR b'    at Simulator.<anonymous> (/Users/cvgl/Desktop/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:104578:12)'
> 2018-01-24 20:43:12,392 ERROR b'    at /Users/cvgl/Desktop/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:51282:19'
> 2018-01-24 20:43:12,392 ERROR b'    at Array.forEach (<anonymous>)'
> 2018-01-24 20:43:12,392 ERROR b'    at Simulator.PubSub.Publish (/Users/cvgl/Desktop/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:51280:12)'
> 2018-01-24 20:43:12,392 ERROR b'    at Simulator.__removeWaiting (/Users/cvgl/Desktop/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:104007:11)'
> 2018-01-24 20:43:12,392 ERROR b'    at Simulator.<anonymous> (/Users/cvgl/Desktop/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:104034:15)'
> 2018-01-24 20:43:12,392 ERROR b'    at /Users/cvgl/Desktop/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:105508:11'
> 2018-01-24 20:43:12,392 ERROR b'    at /Users/cvgl/Desktop/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:38210:10'
> 2018-01-24 20:43:12,425 INFO Finished simserver stderr
> 2018-01-24 20:43:12,425 INFO Finished simserver stdout

msavva, 2018-01-25:
Hi @hzyjerry ,

Thanks for trying out MINOS!  You shouldn't need to specify `--source` as SUNCG is the default environment source, so the command would be:
`python3 -m tools.pygame_client --scene_ids 0004dd3cb11e50530676f77b55262d38`

Let us know if you still have issues.

hzyjerry, 2018-01-25:
@msavva thanks for the response!

This time I get error in simserver.log:

> 2018-01-25 09:38:28,751 ERROR b'node: symbol lookup error: /home/zhiyang/Desktop/minos/server/node_modules/sstk/node_modules/gl/build/Release/webgl.node: undefined symbol: _Z15XextFindDisplayP15_XExtensionInfoP9_XDisplay'

I was able to install sstk and run the test script inside it though.

msavva, 2018-01-25:
This appears to be the headless-gl nvidia gpu issue referenced in #3 .  Can you try following the steps linked at the end of that thread?

msavva, 2018-01-25:
Another option that might be easier, is to install the newest version of node (using `nvm install v9.4.0` and then nvm alias default v9.4.0`).  This will automatically force compilation of the `headless-gl` package on your machine (since precompiled binaries are not available for the newest node versions).  If you follow this, please remove the `node_modules` directory in `server` before you rerun `npm install`.

hzyjerry, 2018-01-25:
That does the fix! Thanks

closed 2018-01-25 by hzyjerry