minosworld/minos #13: Explicitly changing field of view

hzyjerry, 2018-01-25:
Hi,

I'm wondering if there's an explicit way to change the default fov through MINOS code? Currently it seems that it's fixed to around 45~50 degrees.

Thanks

msavva, 2018-01-25:
There is a related discussion in #4 . The `fov` parameter in `config/sensors.yml` controls the vertical field of view.  We are currently working on improving the documentation and structure of the codebase to clarify such configuration parameters.  Thanks for bearing with us in the meantime. :)

hzyjerry, 2018-01-25:
Thanks! Will play around with that. Is there a way to also modify the horizontal field of view?

msavva, 2018-01-25:
The horizontal field of view is currently determined from the aspect ratio of the image and the vertical field of view.

hzyjerry, 2018-01-25:
Nice. It seems that instruction `python3 -m tools.pygame_client --source mp3d --scene_ids 17DRP5sb8fy` no longer works after python repackaging. What would be the correct way to kick start minos?

msavva, 2018-01-26:
If you use the the `python-repackage` branch, the command will be of the form `python3 -m minos.tools.pygame_client`.  The main README.md on that branch is updated with these commands

closed 2018-01-26 by hzyjerry