minosworld/minos #14: Definition of roomId, roomType, objectId, and objectType in SUNCG

playerkk, 2018-01-27:
What can I find their definitions? I didn't find much information on SUNCG's website. Thanks.

playerkk, 2018-01-28:
If I understand correctly, I can get instance segmentation by combining objectId and objectType. Is it correct?

I see there is objectGoal task, is there detailed description of this task? For example, how is the object specified, an image?

Thanks.

playerkk, 2018-01-28:
If I understand correctly, I can get instance segmentation by combining objectId and objectType. Is it correct?

Thanks.

angelxuanchang, 2018-01-28:
The objectId gives the instance segmentation without the semantic type.  If you include the objectType, then you will get the instance semantic segmentation.

playerkk, 2018-01-28:
Thanks a lot. Will close the issue.

closed 2018-01-28 by playerkk