minosworld/minos #15: Support of the objectGoal task

playerkk, 2018-01-28:
I was wondering if the objectGoal is supported. Thanks.

msavva, 2018-01-28:
We support the object goal task.  For example, https://github.com/minosworld/minos/blob/master/config/envs/objectgoal_suncg_mf.py specifies that the navigation goal is a randomly selected door or arch object in each episode with this parameter:
`'goal': {'categories': ['arch', 'door'], 'select': 'random', 'dist_from_bbox': True},`

You can select other object categories as the goal by changing the above parameter.  Please let us know if you have any issues running with the example environment configuration (e.g. using `python3 demo.py --env_config objectgoal_suncg_mf`).

playerkk, 2018-01-28:
Sorry, I should have made the question clearer.

How is the object specified, by an image or by an instruction like "find a door"? I can see there may be some duplicate objects in a scene. How to solve the ambiguity of the target object?

Thanks.

msavva, 2018-01-29:
The object goal can be specified in a variety of ways described here: https://github.com/minosworld/minos/blob/master/API.md#goal-configuration

We don't currently support image input or text input like the example you gave.  Though for the latter, "find a door" is exactly what the previous example parameter is intended to specify (the goal is a randomly selected instance of a door).

If you want to specify a particular object instance (instead of a random instance from a category as in the previous example), that is possible using the `'objectIds': ['0_12']` parameter.  The string must then match the object id of the specific object in the SUNCG house.json file of a given scene.

playerkk, 2018-01-29:
For the specified target, is there any way to provide information of the target object (eg, image or category name)  to an agent? For the roomGoal, an one-hot embedding vector is available.

And how to determine if an agent reaches the target object?

msavva, 2018-01-31:
We didn't run experiments with a target image for the goal.  However, the functionality to return target goal images from the Simulator is implemented (target goal images are visualized on the web demo views).

We plan to update the python API so that target images and other goal information can be retrieved.  In the meantime, to get goal images you need to change a bit of code to pipe the data through:
- Comment out https://github.com/minosworld/minos/blob/master/lib/RoomSimulator.py#L141
- In the gym `demo.py` you can then access the current episode information through `env._sim.start_config_this_episode` . This will contain `goalObservations` that have an image for the target goal.

Please note that we didn't test the above since we didn't use it in any of our experiments.  However, please do let us know if you notice issues.

playerkk, 2018-02-02:
@msavva 
I got the target image as follows
```python
parser = argparse.ArgumentParser(description='MINOS gym wrapper')
add_sim_args_basic(parser)
parser.add_argument('--env_config',
                        default='objectgoal_suncg_sf',
                        help='Environment configuration file')
args = parser.parse_args()

sim = RoomSimulator(sim_args)
sim.init()

res = sim.start_config_this_episode
obj_im = res['goalObservations'][0]['observation']['sensors']['color']['data']. # 128*128*4
smisc.imsave('obj_im.png', obj_im[:,:,:3])
```

But what I got was random images, unrelated to the target (I tried door and chair with select as 'closest'). Would it be possible to help check this problem? Thanks.

msavva, 2018-02-07:
@playerkk , apologies for the delay.

The `goalObservations` images are currently implemented as views taken at the position of the goal object (at the object centroid), facing in a random direction.  Are the output images you are seeing of this type, or is the randomness you observe something different?