minosworld/minos #16: how to generate the input of export_navgrid.js

yscoffee, 2018-02-01:
Hi 
could you give some guide about how to generate the input of "export_navgrid.js" ?
it seems that the "simscenes." file cannot be produced by scripts in the tools directory.

> export_navgrid.js - Exports precomputed navigation maps (will need to be computed and exported for scene sets other than the ones provided in the repo):
> NODE_BASE_URL=~/work ./export_navgrid.js --config_file config/visualize_traces.json --output_dir navgrid --input simscenes. --arch_only

msavva, 2018-02-01:
Hi @yscoffee ,
Thanks for trying out MINOS!

The description of the example command in the README.md was a bit confusing -- apologies for that.  You can either specify `--input simscenes.txt` where `simscenes.txt` is a file with one SUNCG scene id per line, or specify `--ids <id>` where `<id>` is a SUNCG scene id.  We updated the example to be the latter since that is simpler to try out.

Let us know if you still have any trouble.

yscoffee, 2018-02-01:
Hi Manolis
Now I can successfuly regenerate some data.
Many thanks!

closed 2018-02-01 by yscoffee