minosworld/minos #17: The format of navigation maps and suncg.1.0.2 data

yscoffee, 2018-02-02:
Hi
I am trying to map 2D navigation maps back to its original 3D mesh data
but found that, in SUNCG.1.0.2,  "room" directory is removed and there is new "wall" directory, which includes : 
```
004667ffa2594fa780e84d7b979f25ab/004667ffa2594fa780e84d7b979f25ab.arch.json
004667ffa2594fa780e84d7b979f25ab/004667ffa2594fa780e84d7b979f25ab.wall
filetype wall
wall 0 39.28 37.5 0 39.28 37.71 2.7 0 0
wall 1 39.28 38.67 0 39.28 44.71 2.7 0 0
...
```
I guess navigation maps are normalized based the bbox of floor but still not sure how floor, ceiling, and walls are rendering/accessing.  
Could MINOS team give some guides about this part ? 
Thanks

msavva, 2018-02-07:
Hi @yscoffee 
Apologies for the long delay.

The wall, floor and ceiling elements are now specified in the `*.arch.json` files.  We have added a high level description and example of the format in the wiki at https://github.com/minosworld/minos/wiki/SUNCG-Architecture-Format
The default element type is `Wall`, and the points specify the 2D line segments that generate each wall segment.

The `.wall` files are a separate simpler format that specifies only wall segments (it is also documented in the above wiki page).

Let us know if you have any other questions.

yscoffee, 2018-02-09:
Thanks for your detail description and the example.
This may also help a bit:
https://github.com/minosworld/minos/blob/d6512d785364ba8a65a981f6160e726e2d9fae54/server/visualize_path.js#L140

closed 2018-02-09 by yscoffee

msavva, 2018-02-09:
On a related note, we have precomputed navigation maps for all SUNCG scenes (in addition to the set of train/val/test scenes provided as part of the initial release).  You can obtain an archive with all the SUNCG navigation maps from: http://suncg.cs.princeton.edu/data/sim/navgrid_10.suncg.v2.zip (1.2GB)

msavva, 2018-02-09:
We also wrote up a brief description of the navigation grid format: https://github.com/minosworld/minos/wiki/NavGrid-format

Hopefully this further clarifies this data and how it is used