minosworld/minos #18: Train/val/test split

playerkk, 2018-02-09:
Hi, where can I find the train/val/test split? How can I navigate in a particular apartment of a split? Thanks.

msavva, 2018-02-09:
The splits are stored as part of the presampled episode files:
- [data/episodes_states.suncg.csv.bz2](https://github.com/minosworld/minos/blob/master/data/episode_states.suncg.csv.bz2)
- [data/episodes_states.mp3d.csv.bz2](https://github.com/minosworld/minos/blob/master/data/episode_states.mp3d.csv.bz2)

To set the split from which episodes will be picked, you can either pass an `episode_schedule` parameter to the `RoomSimulator` constructor, or use the `set_episode_schedule` call to the `RoomSimulator`.  The strings `train`, `val`, and `test` are the three valid options, and they will correspondingly select the subset of episodes for the given split (from the above two files by default, or other episode_states files if you set the `states_file` parameter to a different file).

You can further restrict the episodes that will be used by a particular `RoomSimulator` through the `episode_filter` parameter in the env config files.  This is an arbitrary python function that can check the value of any of the columns in the episode_states csv files and return a boolean for whether the episode should be included.  For example, you can filter on `sceneId` and `roomId` to select only episodes in a specific house and room.

playerkk, 2018-02-09:
Thanks for the detailed explanation of choosing a split.

If a simulator is created according to the setting of roomgoal_suncg_sf, a random scene with two rooms will be selected. The agent will only move in this particular scene (apartment)?

I am also confused about using pre-sampled episodes. The agent can only access to some particular episodes, in a similar way of experience replay?

msavva, 2018-03-08:
That's right.  The condition filters `scene_filter` and `episode_filter` are applied to the scenes and episodes, and the first 10 remaining episodes for each remaining test or val set scene are used in the evaluation.  This is the default value, but it can be controlled by the `episodes_per_scene_test` parameter.

msavva, 2018-03-15:
@playerkk are there any remaining issues with the train/val/test split? If not, shall we close this issue?

kvas7andy, 2018-03-16:
@msavva 
There are two questions, I would like to ask:
1) Is it necessary to filter first 10 episodes while training? To change `'episode_filter'` for filtering first episodes?
Or everything is preconfigured  train/val/test split?

2) It is, of course, very useful and common to save chekpoints, but the problem is, that training 13.2 million of episodes can be crushed and the first episode after restart seems to be the first of the whole training episodes (from `train/val/test` split). 
Is there any opportunity to filter all scenes before relevant one, which comes after the last saved checkpoint? For example, I can manually search for a `sceneId` and `EPISODE:Number`, but how to start with this concrete episode, which could be one of 10 episodes for the same sceneId?
The only approach which I see is to make another `episode_states.suncg.csv` file with cutted first part. But even with this approach, different simulators will start with the same episode, which they have already simulated before the crash (as some simulator can "outrun" others).

msavva, 2018-03-18:
1. The `episode_filter` setting controls which presampled train/val/test split episodes are used during `val` or `test` schedules. During `train` schedules, random episodes are sampled from the available train split scenes (`episodes_per_scene_train` controls how many random episodes are sampled for each randomly sampled train scene before moving to the next randomly sampled train scene).  The presampled `train` episodes in the split files are there for completeness and are not currently used during training.
2. Since the training schedule randomly samples a train scene and episode within the train scene, "resuming" a training run should not be impacted by what episodes have been sampled thus far.

kojimano, 2018-04-02:
A follow question. What exactly is pathNumDoors specified in episode_states_suncg.csv? This is used to filter episodes from the scenes default envs files. I initially thought this is the parameter to remove trivial episodes by enforcing episodes to go through the number of doors specified in pathNumDoors. However, this is not True. For example row 33302 of episode_states_suncg.csv. This episode with pathNumDoors 1 and pathDist 1.273 apparently does have zero doors from a starting point to a goal point. Another problem I have been observing within some episodes is that the agents are never able to reach from start to goal. This is due to all free spaces are not necessarily connected to each others on the ground truth map, and since the start and end point is randomly sampled from free spaces, some bad episodes are generated. I think this can be confirmed and removed when the a-star search finds no pathes given map and start and endpoint.