minosworld/minos #24: What is the angle of view of the agent and how to fix this

kojimano, 2018-03-01:
Hello, as written in the paper I got the field of view for the agent is 90, but what is the angle of view of the agent? Is this also 90 degrees, and where exactly on the code are they specified (could you probably point me to the script or text file)? Thanks for your response in advance.

msavva, 2018-03-01:
Hi @kojimano ,
If by angle of view you mean the pitch angle of the camera, that is initialized to 0 (i.e. looking straight at the horizon).  The relevant code which creates the camera is at https://github.com/smartscenes/sstk/blob/master/client/js/lib/sim/sensors/CameraSensor.js . Each camera sensor object is added the Agent class (https://github.com/smartscenes/sstk/blob/master/client/js/lib/sim/Agent.js#L309) under a proxy object which represents the pitch orientation.  If you look at the implementation of the lookUp and lookDown actions in the Agent class, you'll see they modify the state of the pitch proxy object (and since the camera is a direct child, influence the pitch of the camera as well).

If you make changes to the code in the sstk library, you will probably want to specify it as a local dependency in the npm packages (https://github.com/minosworld/minos/blob/master/minos/server/package.json#L19), putting there the path to your local checked out copy (see https://stackoverflow.com/questions/15806241/how-to-specify-local-modules-as-npm-package-dependencies).  You will also need to run the build.sh script for the sstk library if you make a change in that code.

kojimano, 2018-03-01:
Thank you for your response. I meant the angle of view in the sense of the attached image below. Is this set to be 90 degrees?[
[image]
](url)

msavva, 2018-03-01:
I see -- apologies for misunderstanding.  What we call "field of view" in our code is the angle of view in this diagram.  It is defined in the sensors config file in the minos codebase: https://github.com/minosworld/minos/blob/master/minos/config/sensors.yml#L27 , and given  in degrees -- the default there is 45 degrees (which is defined in the vertical image direction from center to edge of image).  You can change the value in the sensors config file if you want something other than 90.

kojimano, 2018-03-23:
Hi Thank you for clarifying the vertical field of view is set to 90 degrees. But is the horizontal view of view is also set to 90 degrees in default? Also according to  https://github.com/minosworld/minos/blob/master/minos/config/sensors.yml, the hight of the camera seems to be set to 0.6 m. Is my understanding correct? (I am slightly confused about the behavior of the camera.) Thank you.

msavva, 2018-03-23:
The horizontal angle of view is determined from the vertical one and the aspect ratio of the image resolution (i.e. ratio of width/height).  The position coordinates for the camera (and all other sensors) are expressed in the coordinate frame of the agent body, with the origin being at half the height of the agent (see https://github.com/minosworld/minos/blob/master/minos/config/agent_continuous.yml for the agent `eyeHeight` parameter -- it's a bit confusing that this is the agent height, not the camera height, we will try to improve this naming in an upcoming API update).

kojimano, 2018-03-23:
I see. Thank you for your responses. Just a little clarification on the definition of angle of view. When you define the angle of view in degrees, isn't it actually from the edge of the image to the edge of the image instead of the center of the images to the edge of those as you mentioned in the previous comment? (As far as I observe) Also for the position of the camera, the eye height parameter you mentioned is the camera height in the world coordinate. Is my understanding correct>

msavva, 2018-03-23:
The definition of the angle of view is image edge to edge. The position of
the camera height is 0.6 + (h/2) where the h value is the eyeHeight
parameter of the agent (this the confusing parameter naming I was referring
to that we will fix in a future update)

msavva, 2018-04-02:
Are there any follow up questions regarding the angle of view parameter? If not, let's close this issue thread.

closed 2018-04-02 by kojimano