minosworld/minos #25: Change to complex environment in Gym

dangkh, 2018-03-12:
Hi, 
How can I change a simple environment in Gym to another complex? that environment comprises some stuff such as chair, washing machine, table ,... Thanks

msavva, 2018-03-13:
@dangkh , can you clarify what you mean by "another complex environment"? Do you mean how you would specify a different set of scenes for a given experiment?

dangkh, 2018-03-13:
@msavva sorry for my confuse question. I mean, the environment in gym is empty house and there are no objects in that one. I want to add more objects into the environment like that 
[image]

msavva, 2018-03-14:
You can specify the environment configuration through the `--env_config` command line argument. For example: `python3 demo.py --env_config objectgoal_suncg_mf` will use the `objectgoal_suncg_mf` environment configuration.  See the configuration files in https://github.com/minosworld/minos/tree/master/minos/config/envs for the details of each preset.  Files with the `_sf` or `_mf` indicate furnished SUNCG houses.

closed 2018-03-14 by msavva

dangkh, 2018-03-15:
@msavva Thanks for your reply. By the way, I'm trying to get depth image from observation in gym. The sensors.yml had already configuration for depth camera. Returned data in Simulator.py is only color image. Can you give me some suggestions. Thanks again.

msavva, 2018-03-16:
@dangkh To enable depth output you need to specify the `--depth` command line argument, or set `"depth":True` in the `observations` variable in `sim_config.py`.