minosworld/minos #27: 'Distance' property undefined for scene

kvas7andy, 2018-03-13:
While training unreal model on medium-sized SUNCG dataset I found problem with one scene (wish only one, as I couldn't continue my train after an error occurred). 
I don't know concrete reason but while running .js script for scene parsing it couldn't establish scene and its distance. 
Please, look here at _sim00/simserver.log_ traceback (other 3 servers has the same error, when loaded scene).
Is it due to partial data, included in suncg.v2.zip archive? It was provided to me by the authors of SUNCG alone for minos task.

> 2018-03-13 20:10:45,387 INFO b'preloadData { loadSceneStatistics: true, loadSounds: false }'
2018-03-13 20:10:45,387 INFO b'Loading scene  p5dScene.cf85f3aafa09bb6fc7c228835ef25a33'
2018-03-13 20:10:45,392 INFO b'preloading /root/data/suncg//navgrid_10/cf85f3aafa09bb6fc7c228835ef25a33/cf85f3aafa09bb6fc7c228835ef25a33.empty.grid.json'
2018-03-13 20:10:45,392 INFO b'preloading /root/data/suncg//wall/cf85f3aafa09bb6fc7c228835ef25a33/cf85f3aafa09bb6fc7c228835ef25a33.arch.json'
2018-03-13 20:10:45,392 INFO b'preloading /root/data/suncg//wall/cf85f3aafa09bb6fc7c228835ef25a33/cf85f3aafa09bb6fc7c228835ef25a33.arch.json'
2018-03-13 20:10:45,397 INFO b'Removing subscribers for event drain [ 0 ]'
2018-03-13 20:10:45,549 INFO b'Timing loadScene: 159.492ms'
2018-03-13 20:10:45,559 INFO b'Loaded p5dScene.cf85f3aafa09bb6fc7c228835ef25a33 bbdims: [5.282054875069392,2.7800000104308142,8.214471497729662]'
2018-03-13 20:10:45,559 INFO b'Retexture scene'
2018-03-13 20:10:48,012 INFO b'buildBVH.scene: 5.133ms'
2018-03-13 20:10:48,012 INFO b'SimState.sample: 10.781ms'
2018-03-13 20:10:48,012 INFO b'Timing for initial is 0'
2018-03-13 20:10:48,012 INFO b'Parsing grid for p5dScene.cf85f3aafa09bb6fc7c228835ef25a33 navmap'
2018-03-13 20:10:48,012 INFO b'Refining grid'
2018-03-13 20:10:48,013 INFO b'Populating occupancy'
2018-03-13 20:10:48,013 INFO b'set active level 0'
2018-03-13 20:10:48,017 INFO b'started... waiting for all images to load'
2018-03-13 20:10:48,084 INFO b'Removing subscribers for event drain [ 0 ]'
2018-03-13 20:10:48,085 INFO b'Timing waitImages: 68.464ms'
2018-03-13 20:10:48,126 ERROR b'/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:32592'
2018-03-13 20:10:48,126 ERROR b'\t          if (c.distance < room.distance) {'
2018-03-13 20:10:48,128 ERROR b'\t                ^'
2018-03-13 20:10:48,128 ERROR b''
2018-03-13 20:10:48,128 ERROR b"TypeError: Cannot read property 'distance' of undefined"
2018-03-13 20:10:48,128 ERROR b'    at SceneState.__getIntersectedRoomAt (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:32592:18)'
2018-03-13 20:10:48,129 ERROR b'    at SceneState.getIntersectedRoomAt (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:32607:20)'
2018-03-13 20:10:48,129 ERROR b'    at Simulator.getObservations (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:104361:48)'
2018-03-13 20:10:48,130 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:104290:12'
2018-03-13 20:10:48,131 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:35281:10'
2018-03-13 20:10:48,131 ERROR b'    at eachOfArrayLike (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:35215:10)'
2018-03-13 20:10:48,131 ERROR b'    at eachOf (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:35263:6)'
2018-03-13 20:10:48,131 ERROR b'    at _asyncMap (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:35279:6)'
2018-03-13 20:10:48,131 ERROR b'    at Object.map (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:35268:17)'
2018-03-13 20:10:48,131 ERROR b'    at Simulator.getGoalObservations (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:104282:10)'
2018-03-13 20:10:48,243 INFO Finished simserver stderr
2018-03-13 20:10:48,243 INFO Finished simserver stdout

kojimano, 2018-03-14:
I encountered the same issue, I just evades using that specific room.

kvas7andy, 2018-03-16:
@kojimano 
Thnx for solution.
I would like to ask you for an example of valid object_suncg_mf.py file, considering filtering that particular scene, and 10 first scenes, which you and msavva [noticed](https://github.com/minosworld/minos/issues/18#issuecomment-371360927) to be in evaluation (or they are already filtered in some code in main repository or in [unreal](https://github.com/minosworld/unreal))?

kvas7andy, 2018-03-16:
I am asking, because I tried to filter scene with new
```
    'scene_filter': lambda s: 2 < s['nrooms'] < 6 or s['sceneId'] == 'cf85f3aafa09bb6fc7c228835ef25a33',
```
which spends eternity to load scene and not even try to initialize server.

msavva, 2018-03-16:
@kvas7andy The filter you copied will not work because you need to use 'id' not 'sceneId' (refer to the column headers of the file specified in "scenes_file" for valid filter columns).  You are likely observing freezing on initialization because the process is throwing an error.  Also, you probably want the filter logic to use an "and" not an "or" if you want to exclude episodes from the specific scene.

I am currently investigating the reason for the distance undefined error you observe in that scene and will let you know when that is resolved.

kvas7andy, 2018-03-16:
Thank you, @msavva 
I was not neat enough. 
Similar issue comes from scene number "0c818d79073de53af1b620d950f0ad75"

msavva, 2018-03-17:
@kvas7andy the commits to master should help to resolve this issue.  Please follow the installation steps to rebuild the server module, and let us know if you still observe any crashes.

msavva, 2018-04-02:
@kvas7andy did you get a chance to check whether the current master code resolves the issue on your end?

kvas7andy, 2018-04-12:
@msavva 
No similar error came up till now, thank you!