minosworld/minos #28: Error with pygame_client.py

nina124, 2018-03-14:
After a long long trouble shooting, finally I could run the step 7 in Installation `python3 -m minos.tools.pygame_client`. Thanks for the detailed instruction in issues.
Still, I met the following error. It seems there is some bug in pygame_client.py
> ➜  minos git:(master) ✗ python3 -m minos.tools.pygame_client
2018-03-14 10:24:18,758 INFO {'stk_git_hash': '966f84d', 'machine': 'nina-lab', 'sim_git_hash': '966f84d', 'sim_id': 'sim00'}
Starting simulator...
2018-03-14 10:24:18,769 INFO sim00:Starting sim server at /home/nina/work/minos/minos/server/server.js with port 42001
2018-03-14 10:24:20,433 INFO sim00:connect
observation_space {'measurements': {'direction_to_goal': BoxSpace(range=None, shape=[0, 3]), 'offset_to_goal': BoxSpace(range=None, shape=[0, 3]), 'distance_to_goal': BoxSpace(range=None, shape=[0, 1])}, 'sensors': {'color': BoxSpace(range=[0, 255], shape=[256, 256, 4]), 'force': BoxSpace(range=[None, None], shape=[4, 1])}}
Simulator started.
>***
>***
>CONTROLS: WASD+Arrows = move agent, R = respawn, N = next state/scene, O = print observation, Q = quit
>***
>***
>episode_info {'start': {'room': '0_1', 'angle': 1.376749071468897, 'position': [-45.54027536022024, 0.5949999970197678, -43.15533064966463]}, 'task': 'object_goal', 'sceneId': 'p5dScene.00a76592d5cc7d92eef022393784a2de'}
Traceback (most recent call last):
  File "/home/nina/work/minos/minos/tools/pygame_client.py", line 442, in main
    interactive_loop(sim, args)
  File "/home/nina/work/minos/minos/tools/pygame_client.py", line 376, in interactive_loop
    display_response(response, display_surf, camera_outputs['curr'], print_observation=print_next_observation, write_video=True)
  File "/home/nina/work/minos/minos/tools/pygame_client.py", line 106, in display_response
    draw_offset(measurements.get('offset_to_goal'), display_surf, camera_outputs['offset']['area'])
  File "/home/nina/work/minos/minos/tools/pygame_client.py", line 72, in draw_offset
    dir = (offset[0], offset[2])
IndexError: list index out of range
Error running simulator. Aborting.
2018-03-14 10:24:22,770 INFO sim00:Stopping the simulator
2018-03-14 10:24:22,771 INFO Counter({'frames_received': 2})
2018-03-14 10:24:22,772 INFO sim00:disconnect
2018-03-14 10:24:22,772 INFO sim00:Stopping child servers
2018-03-14 10:24:22,772 INFO sim00:Killing sim pid 9036
2018-03-14 10:24:22,797 INFO sim00:Stopped child servers
2018-03-14 10:24:22,797 INFO sim00:Simulator killed.

msavva, 2018-03-14:
Hi @nina124 , thank you for trying MINOS!

The error is likely due to a failure in loading the scene. Within the working directory from which you start the client, a set of log files will be generated in logs/<timestamp>/simserver.log and logs/<timestamp>/simulator.log. Can you please inspect both and quote here any errors that you see in these log files (in particular simserver.log)?

nina124, 2018-03-14:
@msavva 
I just find I made a stupid mistake. The $NODE_BASE_URL pointed to the original SUNCG dataset instead of the minos package.
Thank you so much for always kindly answering these questions. So helpful for beginners like me.

closed 2018-03-14 by nina124