minosworld/minos #30: Fail to import gym_minos

nina124, 2018-03-15:
I have installed the `gym_minos` package by  running `pip3 install -e .` in the `gym` directory. It runs successfully, but the `gym_minos` package is not installed properly. Running `python3 demo.py --env_config objectgoal_suncg_mf` gives the following error.
> ➜  gym git:(master) ✗ python3 demo.py --env_config objectgoal_suncg_mf
Traceback (most recent call last):
  File "demo.py", line 5, in <module>
    import gym_minos
  File "/home/nina/work/minos/gym/gym_minos/__init__.py", line 2, in <module>
    from gym.envs.registration import register
ImportError: No module named 'gym.envs'

Is it due to the gym version change? My gym version is 0.10.3. Could anyone importing gym_minos successfully tell me your gym version?

dangkh, 2018-03-15:
My gym version is 0.9.4. Did you run pygame_client to make sure you install package successfully ??

kvas7andy, 2018-03-15:
I think this is because go last pull request of @johny-c.
"__init__.py" inside gym directory make python find closest gym library, which is the exact minos/gym folder, where "demo.py" is.
See this [commit](https://github.com/minosworld/minos/commit/f17f41f9d8c6f5a1a82e2d81352f02a5c9c6ff53)

nina124, 2018-03-16:
Yes, @kvas7andy is right.
Importing `gym`, it turns out the gym module from `minos/gym` folder. And after deleting the "__init\__.py" file, the problem solved.

closed 2018-03-16 by nina124