minosworld/minos #31: Error in pygame client

gliese581gg, 2018-03-15:
Hi, 

When I run 'python3 -m minos.tools.pygame_client --empty_room',

I got this error 

```
2018-03-15 16:19:06,384 INFO {'sim_id': 'sim00', 'sim_git_hash': '966f84d', 'machine': 'user-com', 'stk_git_hash': '966f84d'}
Starting simulator...
2018-03-15 16:19:06,400 INFO sim00:Starting sim server at /media/user/NAS_1/Dataset/minos/minos/server/server.js with port 38268
2018-03-15 16:19:07,412 INFO Starting new HTTP connection (1): localhost
2018-03-15 16:19:07,413 WARNING localhost:38268/socket.io [waiting for connection] HTTPConnectionPool(host='localhost', port=38268): Max retries exceeded with url: /socket.io/?transport=polling&EIO=3&t=1521098347406-0 (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnection object at 0x7f39f46b1cc0>: Failed to establish a new connection: [Errno 111] Connection refused',))
2018-03-15 16:19:08,415 INFO Starting new HTTP connection (2): localhost
2018-03-15 16:19:09,492 INFO sim00:connect
Traceback (most recent call last):
  File "/media/user/NAS_1/Dataset/minos/minos/tools/pygame_client.py", line 437, in main
    ep_info = sim.start()
  File "/media/user/NAS_1/Dataset/minos/minos/lib/Simulator.py", line 321, in start
    self._rpc('start', self.params, self.on_started)
  File "/media/user/NAS_1/Dataset/minos/minos/lib/Simulator.py", line 179, in _rpc
    return rpc.call(name, data, callback, seconds, check_wait=lambda: self.running)
  File "/media/user/NAS_1/Dataset/minos/minos/lib/util/RpcCall.py", line 27, in call
    self.sio.wait_for_callbacks(seconds=seconds)    # wait for response
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/__init__.py", line 450, in wait_for_callbacks
    self.wait(seconds, for_callbacks=True)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/__init__.py", line 245, in wait
    self._process_packets()
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/__init__.py", line 269, in _process_packets
    self._process_packet(engineIO_packet)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/__init__.py", line 496, in _process_packet
    delegate(packet)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/__init__.py", line 523, in _on_ack
    ack_callback(*packet.args)
  File "/media/user/NAS_1/Dataset/minos/minos/lib/util/RpcCall.py", line 94, in _handle_response
    self.result = self.callback(self.response)
  File "/media/user/NAS_1/Dataset/minos/minos/lib/Simulator.py", line 552, in on_started
    self.step({'name': 'idle'}, 1)  # take a first step to fill last observation
  File "/media/user/NAS_1/Dataset/minos/minos/lib/Simulator.py", line 376, in step
    return self._rpc('action', action, self.on_observation)
  File "/media/user/NAS_1/Dataset/minos/minos/lib/Simulator.py", line 179, in _rpc
    return rpc.call(name, data, callback, seconds, check_wait=lambda: self.running)
  File "/media/user/NAS_1/Dataset/minos/minos/lib/util/RpcCall.py", line 27, in call
    self.sio.wait_for_callbacks(seconds=seconds)    # wait for response
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/__init__.py", line 450, in wait_for_callbacks
    self.wait(seconds, for_callbacks=True)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/__init__.py", line 245, in wait
    self._process_packets()
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/__init__.py", line 269, in _process_packets
    self._process_packet(engineIO_packet)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/__init__.py", line 471, in _process_packet
    self.placeholder.add(engineIO_packet_data)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 35, in add
    self.replace_placeholders()
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 44, in replace_placeholders
    self.args = traverse(deepcopy(self.args), predicate, fn)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 57, in traverse
    obj[i] = traverse(value, predicate, fn)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 52, in traverse
    obj[key] = traverse(value, predicate, fn)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 52, in traverse
    obj[key] = traverse(value, predicate, fn)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 52, in traverse
    obj[key] = traverse(value, predicate, fn)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 52, in traverse
    obj[key] = traverse(value, predicate, fn)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 52, in traverse
    obj[key] = traverse(value, predicate, fn)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 57, in traverse
    obj[i] = traverse(value, predicate, fn)
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 48, in traverse
    if predicate(obj):
  File "/usr/local/lib/python3.5/dist-packages/socketIO_client/parsers.py", line 39, in predicate
    return '_placeholder' in obj and 'num' in obj
TypeError: argument of type 'int' is not iterable
Error running simulator. Aborting.
2018-03-15 16:19:11,946 INFO sim00:Stopping the simulator
2018-03-15 16:19:11,946 INFO Counter()
```

Does anyone got the same error? How can I solve this?

dangkh, 2018-03-15:
Reinstall using pip install -r requirements.txt
It works for me. In requirements.txt, there is a different socketIO package you should install to replace the regular pip package.

gliese581gg, 2018-03-15:
@dangkh It works!!! Thank you very much!

closed 2018-03-15 by gliese581gg