minosworld/minos #32: Is there  plane segmentation  result ?

huangkuns, 2018-03-15:
I tried the demo, the plane seems well segmented, like walls, windows, cainets etc. I'm wondering is there any plane segmentation result from SUNCG or mattport3d data.

msavva, 2018-03-15:
@huangkuns can you clarify what you mean by plane segmentation? Do you mean the semantic segmentation mask indicating category labels for the objects?

huangkuns, 2018-03-17:
@msavva  Thanks for replying. By plane/surface segmentation, I mean areas that  belong to the same plane/surface. For example, windows/doors are on the same plane as one side of wall. Ceiling and wall are on two orthogonal plane.

msavva, 2018-03-18:
We do not currently have a plane segmentation mode.  However, you can probably use the surface normal sensor to extract planar segments.  You can enable this sensor by using the `--sensor normal` command line argument, or by manually setting `normal: True` in the `observations` parameter.

msavva, 2018-04-02:
@huangkuns did this information answer your question? If so, let's mark the issue as resolved.

msavva, 2018-04-18:
Please re-open issue if additional information is required.

closed 2018-04-18 by msavva