minosworld/minos #34: ObjectType is irelevant

kojimano, 2018-03-17:
Also, I am getting a very random result for ObjectType channel of sensor input. I was expecting the ObjectType to be relevant to ObjectType.csv, but this does not likely to be the case. Also, I was observing the same object types (such as floor) are getting a very different assignment of semantic labels across different scenes.

kojimano, 2018-03-17:
I am getting this as a not a bag since server/node_modules/sstk/client/js/lib/scene/SceneUtils.py is just assigning the index to the object in the order which exists in the scene, but I think it is more convenient for the uses to have the index constant over same object types.

msavva, 2018-03-18:
@kojimano the `objectType` sensor is designed to provide stable indices that correspond to category labels.  The mapping from object category to index is specified through the `--objectttypes_file` argument (see https://github.com/minosworld/minos/issues/14 for a discussion on which predefined mapping files are available).  Instance segmentation masks are provided by the `objectId` and `roomId` sensors -- the behavior of these id sensors is what you are describing (i.e. randomly assigned unique index for each object or room instance).  I have summarized this information in the [FAQ](https://github.com/minosworld/minos/blob/master/FAQ.md#getting-semantic-segmentation-masks).  Please let me know if anything is not clear.

Can you check if you observe inconsistent/unstable index values for the same object category by running `python3 demo.py -s objectType --objecttypes_file ../minos/server/node_modules/sstk/server/static/data/suncg/objectTypes.csv --save_observations`? (I have updated the gym wrapper demo.py with a `--save_observations` debug argument to dump all available observations to png files).  If the output does not match the values in the csv file, please follow up with an example output so that we can track down any bugs.

kojimano, 2018-03-23:
Thank you for all the clarifications!

closed 2018-03-23 by kojimano