minosworld/minos #35: The depth set to be 0.1 for the outside.

kojimano, 2018-03-24:
It came across my experiment that the outside view is set to be the depth of 0.1, and this is pretty confusing for the perspective of the agent (admittingly there is some way to filter this using camera pose and etc on the user side). Is there any way to set this infinity instead of 0.1? Below is a snapshot of the online simulator which shows such an instance, but the recent gihub release shows the exact same behavior. (you can see the door connected to the outside has a very shallow depth.) Sorry if the issue is too specific but I thought it would be more convenient if this has been fixed.
[image]

msavva, 2018-03-26:
@kojimano thank you for reporting this issue.  We have pushed a version update that now properly returns zero values for parts of the image with no depth.  The case of no depth was being clamped to the camera near value (which is 0.1m by default) due to the depth buffer unpacking logic.  Please reinstall the minos server module and let us know if you still observe a problem.

kojimano, 2018-03-29:
Looks like it is working good. Thank you!

kojimano, 2018-03-31:
This issue seems to also happen as in Object Type as well. (The outside view or empty door is also categorized as a floor.)

kojimano, 2018-03-31:
Misunderstanding. This issue never exists.

closed 2018-03-31 by kojimano