minosworld/minos #36: map "roomId" back to node in the SUNCG json files

yscoffee, 2018-03-30:
is it possible to map "roomId" or the roomIndex in the *.arch.json back to the node in the *.json file (SUNCG) ?
Looks "roomId" is not assigned based on the ID (e.g. '1_1') specified in the SUNCG annotation. 
Thanks

msavva, 2018-04-01:
Yes, the roomId value is a linearized contiguous index of the room node ids sorted in ascending order of the first id part `x` then the second id part `y` in `x_y`.  The zeroth value corresponds to null, and room nodes are numbered with index 1 and larger.  The logic that constructs this index is in https://github.com/smartscenes/sstk/blob/master/client/js/lib/scene/SceneState.js#L1398 .  If you want to map from the index value back to the room node you can construct the same room id array and index into it.

msavva, 2018-04-02:
Yes, that's correct.  Note however that the linearized index is stable for a given house since the contiguous sorted order of room node ids does not change.