minosworld/minos #37: Default set up for UNREAL agent in MINOS technical report

kojimano, 2018-03-31:
Hi where can I find the default set up for the UNREAL agent used for MINOS technical report? (Max Speed and Max angular speed). It seems like none of .yml file corresponds to the default set up when I run MINOS in the default setting, and I was wondering which setting was used in the experiment and where I can find this default setting.

msavva, 2018-04-02:
The agent configuration specified in `config/agent_continuous.yml` is the one we used for the experiments reported in the MINOS technical report. The values specified in that file match the ones reported in the writeup.  The default agent configuration uses this parameter set.  Please let us know if you discover any discrepancy.

kojimano, 2018-04-02:
Thank you for your response. It seems like the default max angular speed specified in that .yml is maxAngularSpeed: 12.5663706144 rad/s. Since MINOS's timestamp is every 0.1s. I was expecting the agent angular speed to be 1.25663706144 rad/s. However, I am observing it to be 0.125663706144 rad/s. Other configuration seems to be relevant with config/agent_continuous.yml.

msavva, 2018-04-02:
The `maxAngularSpeed` parameter is a max value -- it is not reached because the values for the acceleration and friction parameters (`turnAcceleration` and `angularFriction` respectively) result in the lower effective angular velocity that you observe (i.e. `d_theta = (a * dt) * dt = (12.56 rad/s * 0.1 s) * 0.1 s = 0.1256 rad`).  Note that the value of `angularFriction: 1.0` means that there is no angular velocity accumulation over subsequent steps.

kojimano, 2018-04-02:
Thank you for the clarification!

closed 2018-04-02 by kojimano

kojimano, 2018-04-03:
So for the linear speed, we will get a slipping? Since linearFriction set to be 0.5, If the agent keep taking forward step until timestamp t and stop at t+1, the agent will have the linear speed of 2.0 m/s at t, 1.0 m/s at t+1 and 0 m/s at t+2 . Is my understanding correct?

kojimano, 2018-04-03:
Also can you clarify the case of the collision? Firstly, if the agent hits objects did it get push back due to physics in this set up of agent_continuous.yml? Secondly, if the agent is touching the object, will this cause any extra friction between the agent and objects and affect the angular or linear acceleration of the agent?

msavva, 2018-04-03:
The agent physics computations are implemented in https://github.com/smartscenes/sstk/blob/master/client/js/lib/sim/Agent.js 

The `linearFriction = k` parameter is applied through `F = - k * m * v / dt` where `v` is the linear velocity of the agent, `m` is the mass of the agent and `dt` is the time step.  Then the computed frictional force F acts against current velocity through `v_{t+1} = v_t + (F * dt / m)` Therefore, `v_{t+1} = (1 - k) * v_t`.

Collisions involve two components: normal to the collision surface, and tangential to the collision surface.  The implementation is in the `narrowphase` function of the CollisionProcessor class at https://github.com/smartscenes/sstk/blob/master/client/js/lib/sim/CollisionProcessorRaycast.js#L322 .  The degree to which there is "push back" is controlled by the `coeffRestitution` parameter. This is 0 by default so that there is no restitution (i.e. velocity normal to the collision surface is zero after collision)

The linear and angular acceleration of the agent are not affected by collisions. Instead, collisions introduce a collision response force which is added to the linear acceleration force.  This means that there is "friction" against collision surfaces (determined by how close the collision surface normal is to the agent's velocity vector and the `linearFriction` parameter).