minosworld/minos #39: IndexError: list index out of range

dabadee, 2018-04-05:
I am getting the following error
omer@OmersLaptop:~/minos$ NODE_BASE_URL=/mnt/16789BF0789BCD3D/work python3 -m minos.tools.pygame_client
2018-04-05 10:42:31,666 INFO {'sim_id': 'sim00', 'stk_git_hash': 'be173c1', 'machine': 'OmersLaptop', 'sim_git_hash': 'be173c1'}
Starting simulator...
2018-04-05 10:42:31,680 INFO sim00:Starting sim server at /home/omer/minos/minos/server/server.js with port 28320
2018-04-05 10:42:33,594 INFO sim00:connect
observation_space {'measurements': {'offset_to_goal': BoxSpace(range=None, shape=[0, 3]), 'direction_to_goal': BoxSpace(range=None, shape=[0, 3]), 'distance_to_goal': BoxSpace(range=None, shape=[0, 1])}, 'sensors': {'force': BoxSpace(range=[None, None], shape=[4, 1]), 'color': BoxSpace(range=[0, 255], shape=[256, 256, 4])}}
Simulator started.
***
***
CONTROLS: WASD+Arrows = move agent, R = respawn, N = next state/scene, O = print observation, Q = quit
***
***
episode_info {'start': {'angle': 3.3845193805761737, 'room': '0_0', 'position': [-39.99729332368484, 0.5949999970197678, -41.7213434747472]}, 'sceneId': 'p5dScene.00a76592d5cc7d92eef022393784a2de', 'task': 'object_goal'}
Traceback (most recent call last):
  File "/home/omer/minos/minos/tools/pygame_client.py", line 443, in main
    interactive_loop(sim, args)
  File "/home/omer/minos/minos/tools/pygame_client.py", line 377, in interactive_loop
    display_response(response, display_surf, camera_outputs['curr'], print_observation=print_next_observation, write_video=True)
  File "/home/omer/minos/minos/tools/pygame_client.py", line 107, in display_response
    draw_offset(measurements.get('offset_to_goal'), display_surf, camera_outputs['offset']['area'])
  File "/home/omer/minos/minos/tools/pygame_client.py", line 72, in draw_offset
    dir = (offset[0], offset[2])
IndexError: list index out of range
Error running simulator. Aborting.
2018-04-05 10:42:36,368 INFO sim00:Stopping the simulator
2018-04-05 10:42:36,368 INFO Counter({'frames_received': 2})
2018-04-05 10:42:36,370 INFO sim00:disconnect
2018-04-05 10:42:36,370 INFO sim00:Stopping child servers
2018-04-05 10:42:36,370 INFO sim00:Killing sim pid 6922
2018-04-05 10:42:36,389 INFO sim00:Stopped child servers
2018-04-05 10:42:36,389 INFO sim00:Simulator killed.

I have attached my logs

Please help
[simserver.log](https://github.com/minosworld/minos/files/1878710/simserver.log)
[simulator.log](https://github.com/minosworld/minos/files/1878712/simulator.log)

dabadee, 2018-04-06:
Yes you are right, I had the incorrect SUNCG dataset, it was also bigger in size.

closed 2018-04-06 by dabadee

wqynew, 2018-10-13:
My problem is different. The error is listed below:
Starting simulator...
2018-10-13 15:18:59,573 INFO sim00:Starting sim server at /home/exbot/minos/minos/server/server.js with port 39142
Traceback (most recent call last):
  File "/home/exbot/minos/minos/tools/pygame_client.py", line 438, in main
    ep_info = sim.start()
  File "/home/exbot/minos/minos/lib/Simulator.py", line 321, in start
    started = self.start_child_servers()
  File "/home/exbot/minos/minos/lib/Simulator.py", line 255, in start_child_servers
    self._sio = SocketIO(self.params.host, self.params.port)
  File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/__init__.py", line 345, in __init__
    resource, hurry_interval_in_seconds, **kw)
  File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/__init__.py", line 58, in __init__
    self._transport
  File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/__init__.py", line 66, in _transport
    self._engineIO_session = self._get_engineIO_session()
  File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/__init__.py", line 80, in _get_engineIO_session
    transport.recv_packet())
  File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/transports.py", line 92, in recv_packet
    for engineIO_packet in decode_engineIO_content(response.content):
  File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/parsers.py", line 95, in decode_engineIO_content
    content, content_index)
  File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/parsers.py", line 202, in _read_packet_length
    while get_byte(content, content_index) not in [0, 1]:
  File "/home/exbot/anaconda3/lib/python3.7/site-packages/socketIO_client/symmetries.py", line 28, in get_byte
    return six.indexbytes(x, index)
IndexError: index out of range
Error running simulator. Aborting.
2018-10-13 15:19:02,499 INFO sim00:Stopping the simulator
2018-10-13 15:19:02,499 INFO Counter()
2018-10-13 15:19:02,499 INFO sim00:Stopping child servers
2018-10-13 15:19:02,499 INFO sim00:Killing sim pid 8022
2018-10-13 15:19:02,506 INFO sim00:Stopped child servers
2018-10-13 15:19:02,506 INFO sim00:Simulator killed.
**the simserver.log:**
2018-10-13 15:19:00,195 INFO b'Configuration { fileCache: { size: 50 },'
2018-10-13 15:19:00,195 INFO b'  imageCache: { size: 200 },'
2018-10-13 15:19:00,195 INFO b'  imageQueue: { concurrency: 4 },'
2018-10-13 15:19:00,195 INFO b"  base_url: '/home/exbot/work/',"
2018-10-13 15:19:00,195 INFO b'  assets_url:'
2018-10-13 15:19:00,195 INFO b"   'file:///home/exbot/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
2018-10-13 15:19:00,454 INFO b'Waiting for client connection on port 39142'
2018-10-13 15:19:00,476 ERROR b'Cannot register unknown asset 2d3ds'
2018-10-13 15:19:00,477 INFO b'register ../server/static/data/matterport/matterport3d.json'
2018-10-13 15:19:00,480 INFO b'Registered asset group: mp3d'
2018-10-13 15:19:00,485 INFO b'Got 90 assets'
2018-10-13 15:19:00,558 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2018-10-13 15:19:00,559 INFO b'Replace asset group p5d'
2018-10-13 15:19:00,560 INFO b'Registered asset group: p5d'
2018-10-13 15:19:00,681 INFO b'Got 2623 assets'
2018-10-13 15:19:01,681 INFO b'Loaded lightSpecs for 202 models.'
2018-10-13 15:19:01,681 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2018-10-13 15:19:01,683 INFO b'Replace asset group p5dTexture'
2018-10-13 15:19:01,683 INFO b'Registered asset group: p5dTexture'
2018-10-13 15:19:02,151 INFO b'Got 67072 assets'
2018-10-13 15:19:02,499 INFO Waiting for simserver stdout to finish
2018-10-13 15:19:02,505 INFO Finished simserver stdout
2018-10-13 15:19:02,506 INFO Waiting for simserver stderr to finish
2018-10-13 15:19:02,506 INFO Finished simserver stderr
**the simulator.log**
2018-10-13 15:18:59,553 INFO {'sim_id': 'sim00', 'machine': '123', 'sim_git_hash': 'c66acdb', 'stk_git_hash': 'c66acdb'}
2018-10-13 15:18:59,573 INFO sim00:Starting sim server at /home/exbot/minos/minos/server/server.js with port 39142
2018-10-13 15:19:02,499 INFO sim00:Stopping the simulator
2018-10-13 15:19:02,499 INFO Counter()
2018-10-13 15:19:02,499 INFO sim00:Stopping child servers
2018-10-13 15:19:02,499 INFO sim00:Killing sim pid 8022
2018-10-13 15:19:02,506 INFO sim00:Stopped child servers
2018-10-13 15:19:02,506 INFO sim00:Simulator killed.

Could you please give me some advice?