minosworld/minos #41: Bug in demo.py after update

kvas7andy, 2018-04-12:
I have this bug, when using `roomgoal_` and `objectgoal_` tasks in `--env_config`
```
root@iitp-04:~/repos/minos/gym# python3 demo.py --env_config roomgoal_mp3d_s
train:60 episodes
test:30 episodes
val:10 episodes
2018-04-12 13:13:26,940 INFO {'stk_git_hash': 'a2ead4c', 'sim_git_hash': 'a2ead4c', 'machine': 'iitp-04', 'sim_id': 'sim00'}
2018-04-12 13:13:26,957 INFO sim00:Starting sim server at /root/repos/minos/minos/server/server.js with port 21906
2018-04-12 13:13:28,979 INFO sim00:connect
2018-04-12 13:13:29,137 INFO sim00:inited
Traceback (most recent call last):
  File "demo.py", line 80, in <module>
    main()
  File "demo.py", line 76, in main
    run_gym(args)
  File "demo.py", line 13, in run_gym
    env.configure(sim_args)
  File "/root/repos/minos/gym/gym_minos/envs/indoor_env.py", line 29, in configure
    self._sim_obs_space = self._sim.get_observation_space(sim_args['outputs'])
  File "/root/repos/minos/minos/lib/RoomSimulator.py", line 227, in get_observation_space
    self.init()
  File "/root/repos/minos/minos/lib/RoomSimulator.py", line 158, in init
    return self.new_episode()
  File "/root/repos/minos/minos/lib/RoomSimulator.py", line 129, in new_episode
    if 'roomType' in result['goal'] and hasattr(self.sim, 'roomTypes'):
KeyError: 'goal'
2018-04-12 13:15:08,820 INFO sim00:disconnect
2018-04-12 13:15:08,820 INFO sim00:Stopping the simulator
2018-04-12 13:15:08,820 INFO Counter({'frames_received': 1})
2018-04-12 13:15:08,820 INFO sim00:Stopping child servers
2018-04-12 13:15:08,821 INFO sim00:Killing sim pid 5915
2018-04-12 13:15:08,872 INFO sim00:Stopped child servers
2018-04-12 13:15:08,872 INFO sim00:Simulator killed.
```

kvas7andy, 2018-04-28:
I will close issue, and reopen it, if the same bug occurs.

closed 2018-04-28 by kvas7andy

khanhptnk, 2018-06-04:
I have the same bug

msavva, 2018-06-06:
Hi @khanhptnk , thank you for trying out MINOS!

Can you try the code updates in the `dev` branch? They contain several bug fixes which we are currently testing.  They should resolve issues like the above.  Let us know if you run into any problems (ideally by opening new issues).

ZhuFengdaaa, 2018-09-03:
I have the same bug in master branch.