minosworld/minos #44: SUNCG scene bug

kvas7andy, 2018-04-16:
While retrieving `objectType` data via `demo.py` new error occurred with one of the scenes:
```
2018-04-16 19:03:31,787 INFO b'preloading /root/data/suncg//navgrid_10/58306be361138410c5383bfc6af2b843/58306be361138410c5383bfc6af2b843.furnished.grid.json'
2018-04-16 19:03:31,787 INFO b'preloading /root/data/suncg//wall/58306be361138410c5383bfc6af2b843/58306be361138410c5383bfc6af2b843.arch.json'
2018-04-16 19:03:31,787 INFO b'preloading /root/data/suncg//wall/58306be361138410c5383bfc6af2b843/58306be361138410c5383bfc6af2b843.arch.json'
2018-04-16 19:03:31,790 INFO b'Removing subscribers for event drain [ 0 ]'
2018-04-16 19:03:33,589 ERROR b'/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:2555'
2018-04-16 19:03:33,589 ERROR b'\t    this.__remove(lru.id);'
2018-04-16 19:03:33,589 ERROR b'\t                      ^'
2018-04-16 19:03:33,589 ERROR b''
2018-04-16 19:03:33,589 ERROR b"TypeError: Cannot read property 'id' of null"
2018-04-16 19:03:33,590 ERROR b'    at AssetCache.set (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:2555:24)'
2018-04-16 19:03:33,590 ERROR b'    at Array.<anonymous> (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:73642:22)'
2018-04-16 19:03:33,590 ERROR b'    at LoadingAsset.done (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:73571:25)'
2018-04-16 19:03:33,590 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:73652:25'
2018-04-16 19:03:33,590 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/ssc/lib/file-util.js:117:9'
2018-04-16 19:03:33,590 ERROR b'    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:528:3)'
2018-04-16 19:03:33,745 INFO Finished simserver stdout
2018-04-16 19:03:33,745 INFO Finished simserver stderr
```

msavva, 2018-04-18:
@kvas7andy can you also copy the corresponding log files as well?

kvas7andy, 2018-04-18:
Of course,

I attached log files in this comment. 
[simulator.log](https://github.com/minosworld/minos/files/1925204/simulator.log)
[simserver.log](https://github.com/minosworld/minos/files/1925205/simserver.log)
This error with the same file occured while training `unreal` on suncg dataset, without using any `objectType` data (with almost the same config file, except a typo in `objectgoal_suncg_mf.py` - `empty_room: True`, but should be `False`):
```
2018-04-18 13:45:56,241 INFO b'starting memory= { rss: 1183367168,'
2018-04-18 13:45:56,241 INFO b'  heapTotal: 690302976,'
2018-04-18 13:45:56,241 INFO b'  heapUsed: 329192120,'
2018-04-18 13:45:56,242 INFO b'  external: 392962798 }'
2018-04-18 13:45:56,242 INFO b'preloadData { loadSceneStatistics: true, loadSounds: false }'
2018-04-18 13:45:56,243 INFO b'Loading scene  p5dScene.58306be361138410c5383bfc6af2b843'
2018-04-18 13:45:56,261 INFO b'preloading /root/data/suncg//navgrid_10/58306be361138410c5383bfc6af2b843/58306be361138410c5383bfc6af2b843.furnished.grid.json'
2018-04-18 13:45:56,261 INFO b'preloading /root/data/suncg//wall/58306be361138410c5383bfc6af2b843/58306be361138410c5383bfc6af2b843.arch.json'
2018-04-18 13:45:56,261 INFO b'preloading /root/data/suncg//wall/58306be361138410c5383bfc6af2b843/58306be361138410c5383bfc6af2b843.arch.json'
2018-04-18 13:45:56,264 INFO b'Removing subscribers for event drain [ 0 ]'
2018-04-18 13:45:58,322 ERROR b'/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:2555'
2018-04-18 13:45:58,322 ERROR b'\t    this.__remove(lru.id);'
2018-04-18 13:45:58,323 ERROR b'\t                      ^'
2018-04-18 13:45:58,323 ERROR b''
2018-04-18 13:45:58,323 ERROR b"TypeError: Cannot read property 'id' of null"
2018-04-18 13:45:58,323 ERROR b'    at AssetCache.set (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:2555:24)'
2018-04-18 13:45:58,323 ERROR b'    at Array.<anonymous> (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:73642:22)'
2018-04-18 13:45:58,323 ERROR b'    at LoadingAsset.done (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:73571:25)'
2018-04-18 13:45:58,323 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:73652:25'
2018-04-18 13:45:58,323 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/ssc/lib/file-util.js:117:9'
2018-04-18 13:45:58,324 ERROR b'    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:528:3)'
2018-04-18 13:45:58,443 INFO Finished simserver stdout
2018-04-18 13:45:58,444 INFO Finished simserver stderr
```

kvas7andy, 2018-04-18:
The same error with another scene:
[simserver.log](https://github.com/minosworld/minos/files/1925206/simserver.log)
[simulator.log](https://github.com/minosworld/minos/files/1925207/simulator.log)
```
2018-04-16 23:21:18,792 INFO b'starting memory= { rss: 1821786112,'
2018-04-16 23:21:18,792 INFO b'  heapTotal: 947347456,'
2018-04-16 23:21:18,792 INFO b'  heapUsed: 566072544,'
2018-04-16 23:21:18,792 INFO b'  external: 367284817 }'
2018-04-16 23:21:18,792 INFO b'preloadData { loadSceneStatistics: true, loadSounds: false }'
2018-04-16 23:21:18,792 INFO b'Loading scene  p5dScene.25a19fa80aac90b4cf22aea24e4b7032'
2018-04-16 23:21:18,849 INFO b'preloading /root/data/suncg//navgrid_10/25a19fa80aac90b4cf22aea24e4b7032/25a19fa80aac90b4cf22aea24e4b7032.furnished.grid.json'
2018-04-16 23:21:18,849 INFO b'preloading /root/data/suncg//wall/25a19fa80aac90b4cf22aea24e4b7032/25a19fa80aac90b4cf22aea24e4b7032.arch.json'
2018-04-16 23:21:18,849 INFO b'preloading /root/data/suncg//wall/25a19fa80aac90b4cf22aea24e4b7032/25a19fa80aac90b4cf22aea24e4b7032.arch.json'
2018-04-16 23:21:18,851 INFO b'Removing subscribers for event drain [ 0 ]'
2018-04-16 23:21:20,114 ERROR b'/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:2555'
2018-04-16 23:21:20,114 ERROR b'\t    this.__remove(lru.id);'
2018-04-16 23:21:20,115 ERROR b'\t                      ^'
2018-04-16 23:21:20,115 ERROR b''
2018-04-16 23:21:20,115 ERROR b"TypeError: Cannot read property 'id' of null"
2018-04-16 23:21:20,115 ERROR b'    at AssetCache.set (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:2555:24)'
2018-04-16 23:21:20,115 ERROR b'    at Array.<anonymous> (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:73642:22)'
2018-04-16 23:21:20,115 ERROR b'    at LoadingAsset.done (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:73571:25)'
2018-04-16 23:21:20,115 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:73652:25'
2018-04-16 23:21:20,115 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/ssc/lib/file-util.js:117:9'
2018-04-16 23:21:20,115 ERROR b'    at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:528:3)'
2018-04-16 23:21:20,253 INFO Finished simserver stdout
2018-04-16 23:21:20,254 INFO Finished simserver stderr
```

kvas7andy, 2018-04-23:
One more scene, with the same problem: "f3bb9c723c8f8f50827f92741a463b24"

[simserver.log](https://github.com/minosworld/minos/files/1940093/simserver.log)
[simulator.log](https://github.com/minosworld/minos/files/1940094/simulator.log)

Do you have any ideas, what the problem is?

kvas7andy, 2018-04-24:
After update, one more run of `objectgoal_suncg_mf.py` for `unreal` algorithm didn't throw any error!

But, as noticed before (in previous comment), running `demo.py` after `node_modules` rebuild throw such error with another scene (I had to sort out `p5dScene.58306be361138410c5383bfc6af2b843` scene, but error occured with new scene).

msavva, 2018-04-25:
@kvas7andy This issue was due to over-eager cache clearing logic which would occasionally fail when the cache was empty.  The v0.5.3 update resolves this bug.  I will close the issue but please re-open if you still encounter the same problem.

kvas7andy, 2018-04-27:
Hi,

I haven't checked before update, but now occurred another problem, while running `unreal`:
```
2018-04-27 22:05:35,401 INFO b'Timing waitImages: 161.092ms'
2018-04-27 22:05:35,577 ERROR b'THREE.WebGLRenderer: EXT_texture_filter_anisotropic extension not supported.'
2018-04-27 22:05:35,757 INFO b'Timing start: 35099.751ms'
2018-04-27 22:05:42,296 INFO b'Seeding simulator rng with 46722566'
2018-04-27 22:06:07,592 INFO b'SimState.sample: 2284.321ms'
2018-04-27 22:06:07,592 INFO b'set active level 0'
2018-04-27 22:06:07,682 ERROR b'/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:106471'
2018-04-27 22:06:07,683 ERROR b'\t      callback(err, scope.state.sceneState);'
2018-04-27 22:06:07,683 ERROR b'\t      ^'
2018-04-27 22:06:07,683 ERROR b''
2018-04-27 22:06:07,683 ERROR b'TypeError: callback is not a function'
2018-04-27 22:06:07,683 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:106471:8'
2018-04-27 22:06:07,683 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:35013:17'
2018-04-27 22:06:07,683 ERROR b'    at next (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:39855:30)'
2018-04-27 22:06:07,683 ERROR b'    at /root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:35498:17'
2018-04-27 22:06:07,683 ERROR b'    at Timeout._onTimeout (/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:106467:35)'
2018-04-27 22:06:07,684 ERROR b'    at ontimeout (timers.js:458:11)'
2018-04-27 22:06:07,684 ERROR b'    at tryOnTimeout (timers.js:296:5)'
2018-04-27 22:06:07,684 ERROR b'    at Timer.listOnTimeout (timers.js:259:5)'
2018-04-27 22:06:07,764 INFO Finished simserver stdout
2018-04-27 22:06:07,765 INFO Finished simserver stderr
```
[simserver.log](https://github.com/minosworld/minos/files/1956824/simserver.log)
[simulator.log](https://github.com/minosworld/minos/files/1956825/simulator.log)

After killing the process with Ctrl+C in terminal, `python` process still works, while node processes exit.

kvas7andy, 2018-04-28:
Hi @msavva,
Unfortunately, I am not good at debugging javascript. However, I think this is not a difficult mistake.
I cannot re-open my own issues, if a repo collaborator closed it.

msavva, 2018-04-29:
@kvas7andy This looks like a different issue -- we'll look into it from our side.  Do you observe this consistently soon after starting the simulation?

kvas7andy, 2018-04-29:
I thought the problem is linked to the update of version "v0.5.3", that's why commented it here.

It seems to occur consistently, as I tried different envs with `unreal` training, and, while running `demo.py`. I think, particularly, when environment resets.

commit f3f115fb (2018-04-30): fix simulator server reset crash due to regression in v0.5.3 update (closes #44, closes #49)
closed 2018-04-30 by msavva