minosworld/minos #47: Trouble with running npm install command

lionminhu, 2018-04-24:
Hello, I have been trying to install MINOS for a while, and I have been following the installation steps in the README.md. However, running `npm install` command inside the `server` directory (located at `/home/mwlee/minos/minos/server` in this case) under `sudo -s` gives the following error message.
```
root@doota:~/mwlee/minos/minos/server# npm install
npm ERR! code 1
npm ERR! Command failed: /usr/bin/git clone -q git://github.com/smartscenes/sstk.git /home/.npm/_cacache/tmp/git-clone-33dd48c3
npm ERR! /home/.npm/_cacache/tmp/git-clone-33dd48c3/.git: Permission denied
npm ERR! 

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/.npm/_logs/2018-04-24T04_47_15_520Z-debug.log
```

Running the command with another `sudo` in front of the command gives a very long error message (more than 4000 lines), and I am not quite sure which part is relevant. The first ~170 lines of this error message is the following.
```
root@doota:~/mwlee/minos/minos/server# sudo npm install
npm ERR! code 1
npm ERR! Command failed: /usr/bin/git clone -q git://github.com/smartscenes/sstk.git /home/.npm/_cacache/tmp/git-clone-33dd48c3
npm ERR! /home/.npm/_cacache/tmp/git-clone-33dd48c3/.git: Permission denied
npm ERR! 

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/.npm/_logs/2018-04-24T04_47_15_520Z-debug.log
root@doota:~/mwlee/minos/minos/server# sudo npm install
loadDep:sstk -> resolveWi | |####################################-----------------------------------------------------------------------------------------------------------------------------------------|
npm WARN deprecated coffee-script@1.12.7: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
loadDep:node-expat -> 304 - |########################################-------------------------------------------------------------------------------------------------------------------------------------|
loadDep:hoek -> resolveWi | |########################################-------------------------------------------------------------------------------------------------------------------------------------|
loadDep:topo -> 304       - |########################################-------------------------------------------------------------------------------------------------------------------------------------|
loadDep:punycode -> 304   - |########################################-------------------------------------------------------------------------------------------------------------------------------------|
loadDep:hoek -> resolveWi / |########################################-------------------------------------------------------------------------------------------------------------------------------------|
loadDep:source-map -> 304 \ |########################################-------------------------------------------------------------------------------------------------------------------------------------|
loadDevDep:pixel-diff ->  | |########################################################---------------------------------------------------------------------------------------------------------------------|
npm WARN deprecated to-iso-string@0.0.2: to-iso-string has been deprecated, use @segment/to-iso-string instead.
npm WARN deprecated jade@0.26.3: Jade has been renamed to pug, please install the latest version of pug instead of jade
npm WARN deprecated minimatch@0.3.0: Please update to minimatch 3.0.2 or higher to avoid a RegExp DoS issue
loadDep:uuid -> cache add \ |#######################################################################------------------------------------------------------------------------------------------------------|

> node-expat@2.3.16 install /home/mwlee/minos/minos/server/node_modules/node-expat
> node-gyp rebuild

make: Entering directory '/home/mwlee/minos/minos/server/node_modules/node-expat/build'
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmlparse.o
../deps/libexpat/lib/xmlparse.c: In function ‘gather_time_entropy’:
../deps/libexpat/lib/xmlparse.c:780:7: warning: variable ‘gettimeofday_res’ set but not used [-Wunused-but-set-variable]
   int gettimeofday_res;
       ^
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmltok.o
  CC(target) Release/obj.target/expat/deps/libexpat/lib/xmlrole.o
  AR(target) Release/obj.target/deps/libexpat/libexpat.a
  COPY Release/libexpat.a
  CXX(target) Release/obj.target/node_expat/node-expat.o
  SOLINK_MODULE(target) Release/obj.target/node_expat.node
  COPY Release/node_expat.node
make: Leaving directory '/home/mwlee/minos/minos/server/node_modules/node-expat/build'

> sstk@0.5.2 postinstall /home/mwlee/minos/minos/server/node_modules/sstk
> npm run bootstrap

> sstk@0.5.2 bootstrap /home/mwlee/minos/minos/server/node_modules/sstk
> lerna bootstrap --hoist

lerna info version 2.10.2
lerna info Bootstrapping 3 packages
lerna info lifecycle preinstall
lerna info Installing external dependencies
lerna info hoist Installing hoisted dependencies into root
lerna info hoist Pruning hoisted dependencies
lerna info hoist Finished pruning hoisted dependencies
lerna ERR! execute callback with error
lerna ERR! Error: Command failed: npm install
[... 110 lines cut ...]
lerna ERR! npm ERR! Please include the following file with any support request:
lerna ERR! npm ERR!     /home/mwlee/minos/minos/server/node_modules/sstk/npm-debug.log
```

I have Node v9.11.1, and `npm version` gives the following result.
```
root@doota:~/mwlee/minos/minos/server# npm version
{ 'sstk-sim': '0.5.2',
  npm: '5.6.0',
  ares: '1.13.0',
  cldr: '33.0',
  http_parser: '2.8.0',
  icu: '61.1',
  modules: '59',
  napi: '3',
  nghttp2: '1.29.0',
  node: '9.11.1',
  openssl: '1.0.2o',
  tz: '2018c',
  unicode: '10.0',
  uv: '1.19.2',
  v8: '6.2.414.46-node.23',
  zlib: '1.2.11' }
```
If there are other information I would need to provide, or if I would have to raise this issue in npm repository instead, please let me know.

msavva, 2018-04-25:
@lionminhu Thank you for trying out MINOS.
The log indicates the issue is with permissions during git cloning of the dependencies.  There should be no need to run any of the installation steps using `sudo`.  Can you confirm whether you installed `nvm` and run `npm install` under a regular user account, not using sudo?  If you perform all the installation steps from beginning to end without elevating to a sudo user there should be no issue like the one you see.

msavva, 2018-04-26:
@lionminhu Besides the permissions issues, at the end of the log it looks like you might also be missing a package for building one of the dependencies. In Ubuntu you can install necessary packages for building all dependencies using `sudo apt-get install build-essential libxi-dev libglu1-mesa-dev libglew-dev libvips`.

lionminhu, 2018-04-27:
Uninstalling and reinstalling `nvm` and rerunning `npm install` without `sudo` seems to have worked successfully. I installed `libvips` package, which was the only one missing, according to your advice.

I have yet to check that everything works after installing the SUNCG/Matterport3D data set, but since `npm install` part is settled, I will close this issue now.

Thank you for your help.

closed 2018-04-27 by lionminhu