minosworld/minos #50: TypeError: Cannot read property 'split' of null" in sstk

kibeomKim, 2018-05-01:
Excuse me.

I am trying to install MINOS on docker container in my headless server
and I tried install https://github.com/smartscenes/sstk and headless-gl package

and I reached under the message and got the successes.

"One additional note: if you have already run npm install and see a message about removed packages when re-running it, please delete the node_modules directory inside server and then try running npm install again. There should be a lerna success Bootstrapped 3 packages message at the end of this command."

but, After this trial, I got the error messages when I execute ./server.js or demo.py

2018-05-01 14:32:07,663 INFO b'Configuration { fileCache: { size: 50 },'
2018-05-01 14:32:07,663 INFO b'  imageCache: { size: 200 },'
2018-05-01 14:32:07,664 INFO b'  imageQueue: { concurrency: 4 },'
2018-05-01 14:32:07,664 INFO b"  base_url: '/root/work/',"
2018-05-01 14:32:07,664 INFO b"  assets_url: 'file:///home/minos/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
2018-05-01 14:32:07,949 INFO b'Waiting for client connection on port 6006'
2018-05-01 14:32:07,974 ERROR b'Cannot register unknown asset 2d3ds'
2018-05-01 14:32:07,976 INFO b'register ../server/static/data/matterport/matterport3d.json'
2018-05-01 14:32:07,979 INFO b'Registered asset group: mp3d'
2018-05-01 14:32:07,985 INFO b'Got 90 assets'
2018-05-01 14:32:08,062 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2018-05-01 14:32:08,064 INFO b'Replace asset group p5d'
2018-05-01 14:32:08,064 INFO b'Registered asset group: p5d'
2018-05-01 14:32:08,155 ERROR b'/home/minos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:48216'
2018-05-01 14:32:08,155 ERROR b"\t          m[k] = v.split(',');"
2018-05-01 14:32:08,155 ERROR b'\t                   ^'
2018-05-01 14:32:08,155 ERROR b''
2018-05-01 14:32:08,155 ERROR b**"TypeError: Cannot read property 'split' of null"**
2018-05-01 14:32:08,155 ERROR b'    at /home/minos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:48216:21'
2018-05-01 14:32:08,155 ERROR b'    at /home/minos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:8934:16'
2018-05-01 14:32:08,156 ERROR b'    at baseForOwn (/home/minos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:7019:25)'
2018-05-01 14:32:08,156 ERROR b'    at /home/minos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:8903:19'
2018-05-01 14:32:08,156 ERROR b'    at Function.forEach (/home/minos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:13351:15)'
2018-05-01 14:32:08,156 ERROR b'    at AssetsDb.__loadAssetInfoFromCsvData (/home/minos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:48214:10)'
2018-05-01 14:32:08,156 ERROR b'    at AssetsDb.loadAssetInfoFromData (/home/minos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:48240:24)'
2018-05-01 14:32:08,156 ERROR b'    at registerCustomAssetGroupSync (/home/minos/minos/minos/server/node_modules/sstk/ssc/stk-ssc.js:230:12)'
2018-05-01 14:32:08,156 ERROR b'    at Object.registerCustomAssetGroupsSync (/home/minos/minos/minos/server/node_modules/sstk/ssc/stk-ssc.js:271:5)'
2018-05-01 14:32:08,156 ERROR b'    at Object.<anonymous> (/home/minos/minos/minos/server/server.js:33:12)'
2018-05-01 14:32:08,172 INFO Finished simserver stderr
2018-05-01 14:32:08,172 INFO Finished simserver stdout
2018-05-01 14:32:08,669 INFO Waiting for simserver stdout to finish
2018-05-01 14:32:08,669 INFO Waiting for simserver stderr to finish

How do I fix this error?

thanks!

kvas7andy, 2018-05-01:
Hi,

Not sure about precise solution, but I had the same problem when tried to degrade to 0.5.2 sstk version.

Please, make sure, that you have last version of `master` branch installed from main repository.

If that doesn't help, please, look at this issue thread #12 (and some other referenced thread), which might help you.

kibeomKim, 2018-05-02:
Thank you for your response!

I delete all directories related to MINOS and reinstall step by step in front of the installing process.
and I reached new..error

If I enter the command like this  `xvfb-run -s "-ac -screen 0 1280x1024x24" python benchmark.py --port 6006` I can get under the log message.

simserver.log

> 2018-05-02 04:54:35,995 INFO b'Configuration { fileCache: { size: 50 },'
> 2018-05-02 04:54:35,996 INFO b'  imageCache: { size: 200 },'
> 2018-05-02 04:54:35,996 INFO b'  imageQueue: { concurrency: 4 },'
> 2018-05-02 04:54:35,996 INFO b"  base_url: '/root/work/',"
> 2018-05-02 04:54:35,996 INFO b"  assets_url: 'file:///home/minos/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
> 2018-05-02 04:54:36,269 INFO b'Waiting for client connection on port 6006'
> 2018-05-02 04:54:36,290 ERROR b'Cannot register unknown asset 2d3ds'
> 2018-05-02 04:54:36,291 INFO b'register ../server/static/data/matterport/matterport3d.json'
> 2018-05-02 04:54:36,294 INFO b'Registered asset group: mp3d'
> 2018-05-02 04:54:36,298 INFO b'Got 90 assets'
> 2018-05-02 04:54:36,357 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
> 2018-05-02 04:54:36,359 INFO b'Replace asset group p5d'
> 2018-05-02 04:54:36,359 INFO b'Registered asset group: p5d'
> 2018-05-02 04:54:36,429 INFO b'Got 2623 assets'
> 2018-05-02 04:54:37,394 INFO b'Loaded lightSpecs for 202 models.'
> 2018-05-02 04:54:37,395 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
> 2018-05-02 04:54:37,398 INFO b'Replace asset group p5dTexture'
> 2018-05-02 04:54:37,398 INFO b'Registered asset group: p5dTexture'
> 2018-05-02 04:54:37,635 INFO b'Got 67072 assets'
> 2018-05-02 04:54:37,828 INFO b'Client agEQWstUDLW86gUlAAAA connected on port 6006'
> 2018-05-02 04:54:37,834 INFO b'Loaded LightSpecs for 202 models'
> 2018-05-02 04:54:37,909 ERROR b'X Error of failed request:  BadRequest (invalid request code or no such operation)'
> 2018-05-02 04:54:37,909 ERROR b'  Major opcode of failed request:  151 (GLX)'
> 2018-05-02 04:54:37,909 ERROR b'  Minor opcode of failed request:  206 ()'
> 2018-05-02 04:54:37,909 ERROR b'  Serial number of failed request:  100'
> 2018-05-02 04:54:37,909 ERROR b'  Current serial number in output stream:  100'
> 2018-05-02 04:54:37,924 INFO Finished simserver stderr
> 2018-05-02 04:54:37,924 INFO Finished simserver stdout

simulator.log

> 2018-05-02 04:54:35,227 INFO {'sim_id': 'sim00', 'machine': '4ff50be5170f', 'sim_git_hash': '399f4a0', 'stk_git_hash': '399f4a0'}
> 2018-05-02 04:54:35,246 INFO sim00:Starting sim server at /home/minos/minos/minos/server/server.js with port 6006
> 2018-05-02 04:54:37,849 INFO sim00:connect
> 2018-05-02 04:54:38,926 INFO sim00:disconnect

If I command just `python benchmark.py` and do not includ `xvfv-run` ,  I can get WebGL error like  #3 

Thank you !

kibeomKim, 2018-05-02:
Hi!

I reinstalled all MINOS and related projects (`headless-gl`, https://github.com/smartscenes/sstk#getting-started, `sstk` [https://github.com/smartscenes/sstk#getting-started](url) ) and tried your response.
And now, maybe, I can train on MINOS.
yet, I have some errors. I think it's just visualization problem and is not critical for me if it is not a problem related to learning some tasks.

Thank you for your help!

> 2018-05-02 15:27:47,264 INFO b'Waiting for client connection on port 6006'
> 2018-05-02 15:27:47,289 ERROR b'Cannot register unknown asset 2d3ds'
> 2018-05-02 15:27:47,290 INFO b'register ../server/static/data/matterport/matterport3d.json'
> 2018-05-02 15:27:47,293 INFO b'Registered asset group: mp3d'
> 2018-05-02 15:27:47,297 INFO b'Got 90 assets'
> 2018-05-02 15:27:47,387 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
> 2018-05-02 15:27:47,389 INFO b'Replace asset group p5d'
> 2018-05-02 15:27:47,390 INFO b'Registered asset group: p5d'
> 2018-05-02 15:27:47,487 INFO b'Got 2623 assets'
> 2018-05-02 15:27:48,920 INFO b'Loaded lightSpecs for 202 models.'
> 2018-05-02 15:27:48,920 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
> 2018-05-02 15:27:48,930 INFO b'Replace asset group p5dTexture'
> 2018-05-02 15:27:48,930 INFO b'Registered asset group: p5dTexture'
> 2018-05-02 15:27:49,224 INFO b'Got 67072 assets'
> 2018-05-02 15:27:49,543 INFO b'Client IbgqGs65blowHj0sAAAA connected on port 6006'
> 2018-05-02 15:27:49,546 INFO b'Loaded LightSpecs for 202 models'
> 2018-05-02 15:27:49,650 INFO b'THREE.WebGLRenderer 84'
> 2018-05-02 15:27:49,652 ERROR b'THREE.WebGLRenderer: WEBGL_depth_texture extension not supported.'
> 2018-05-02 15:27:49,652 ERROR b'THREE.WebGLRenderer: OES_texture_float extension not supported.'
> 2018-05-02 15:27:49,652 ERROR b'THREE.WebGLRenderer: OES_texture_float_linear extension not supported.'
> 2018-05-02 15:27:49,653 ERROR b'THREE.WebGLRenderer: OES_texture_half_float extension not supported.'
> 2018-05-02 15:27:49,653 ERROR b'THREE.WebGLRenderer: OES_texture_half_float_linear extension not supported.'
> 2018-05-02 15:27:49,653 ERROR b'THREE.WebGLRenderer: OES_standard_derivatives extension not supported.'
> 2018-05-02 15:27:49,653 ERROR b'THREE.WebGLRenderer: OES_element_index_uint extension not supported.'
> 2018-05-02 15:27:49,669 INFO b"Creating Simulator with options { simulator: 'room_simulator',"
> 2018-05-02 15:27:49,670 INFO b'  numSimulators: 1,'
> 2018-05-02 15:27:49,670 INFO b"  modalities: [ 'color', 'measurements' ],"
> 2018-05-02 15:27:49,670 INFO b"  outputs: [ 'color', 'measurements', 'rewards', 'terminals' ],"
> 2018-05-02 15:27:49,670 INFO b'  resolution: [ 84, 84 ],'
> 2018-05-02 15:27:49,670 INFO b'  frameSkip: 1,'
> 2018-05-02 15:27:49,670 INFO b"  host: 'localhost',"
> 2018-05-02 15:27:49,670 INFO b'  logActionTrace: false,'
> 2018-05-02 15:27:49,670 INFO b'  autoStart: true,'
> 2018-05-02 15:27:49,670 INFO b"  collisionDetection: { mode: 'raycast' },"
> 2018-05-02 15:27:49,670 INFO b'  navmap:'
> 2018-05-02 15:27:49,671 INFO b'   { refineGrid: true,'
> 2018-05-02 15:27:49,671 INFO b'     autoUpdate: true,'
> 2018-05-02 15:27:49,671 INFO b'     allowDiagonalMoves: true,'
> 2018-05-02 15:27:49,671 INFO b'     reverseEdgeOrder: false },'
> 2018-05-02 15:27:49,671 INFO b"  rewardType: 'dist_time',"
> 2018-05-02 15:27:49,671 INFO b'  observations:'
> 2018-05-02 15:27:49,671 INFO b'   { color: true,'
> .....

msavva, 2018-05-04:
@kibeomKim The remaining error messages are reporting unsupported rendering features and are safe to ignore.

@kvas7andy Thank you for helping out on the issue threads -- community support is deeply appreciated.  You are correct that headless rendering through `xvfb-run` relies on mesa.  An alternative but more complex approach for headless rendering uses  TurboVNC + VirtualGL for hardware-accelerated headless rendering (see http://carla.readthedocs.io/en/latest/carla_headless/ for a description of how to set up this configuration)

kvas7andy, 2018-05-04:
@msavva 
hank you, I am highly involved in MINOS project right now, and happy to help with some questions from community. 

If anyone could try this solution to render on headless machine with NVIDIA libs that would be great!