minosworld/minos #51: How to save the navigational path video?

hasifast, 2018-05-04:
I want to record the video of navigation similar to the one given in the demo video of MINOS.
Please guide me if there is any option available to record it without using desktop screen recorder etc.

latifAnjum, 2018-05-24:
By using the command "python3 -m minos.tools.pygame_client --save_video test.avi" i get the error 

2018-05-24 11:11:40,688 INFO {'sim_git_hash': '3bf7fb6', 'machine': 'latif-Lenovo-ideapad-310-15IKB', 'stk_git_hash': '3bf7fb6', 'sim_id': 'sim00'}
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/latif/Desktop/minos/minos/tools/pygame_client.py", line 457, in <module>
    main()
  File "/home/latif/Desktop/minos/minos/tools/pygame_client.py", line 423, in main
    VIDEO_WRITER = VideoWriter(filename, framerate=24, resolution=(args.width, args.height), rgb=is_rgb)
  File "/home/latif/Desktop/minos/minos/lib/util/VideoWriter.py", line 33, in __init__
    self.proc = sp.Popen(command, stdin=sp.PIPE, stderr=sp.PIPE)
  File "/usr/lib/python3.5/subprocess.py", line 947, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.5/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: 'ffmpeg'
Exception ignored in: <bound method VideoWriter.__del__ of <minos.lib.util.VideoWriter.VideoWriter object at 0x7fe6ed2a7390>>
Traceback (most recent call last):
  File "/home/latif/Desktop/minos/minos/lib/util/VideoWriter.py", line 44, in __del__
    self.close()
  File "/home/latif/Desktop/minos/minos/lib/util/VideoWriter.py", line 39, in close
    self.proc.stdin.close()
AttributeError: 'VideoWriter' object has no attribute 'proc'
2018-05-24 11:11:42,355 INFO sim00:Stopping the simulator
2018-05-24 11:11:42,355 INFO Counter()
2018-05-24 11:11:42,355 INFO sim00:Stopping child servers
2018-05-24 11:11:42,355 INFO sim00:Stopped child servers
2018-05-24 11:11:42,355 INFO sim00:Simulator killed.

msavva, 2018-06-06:
The above error is due to the system not having `ffmpeg` installed.  Please install ffmpeg through `apt-get`, `brew` or other package manager depending on your OS.