minosworld/minos #56: Is it possible to filter out scenes without a particular object?

antonilo, 2018-05-11:
I am wondering whether is possible to filter out scenes / episodes if a particular category of objects (e.g. stairs) is absent. I assume that the configuration file has to be changed, but I did not find any evident way to do it.
Moreover, I am not entirely sure what the difference between scene and episode is. Could you please comment on it?

Thanks a lot in advance!

antonilo, 2018-05-12:
Hi @msavva, thanks for your comment. 
How can I get all objects that are present in a scene? 
I guess I need to apply a filter in StateSet.py, am I wrong?

I have made a little modification to Roomsimulator.py to change scene if the object is not found. I am basically checking whether there is a path to the goal. Unfortunately this can't be passed as a "scene filter", so I guess performance could be improved.
I can make a pull request if that is of interest for you guys.

msavva, 2018-05-13:
The code at https://github.com/minosworld/minos/issues/22#issuecomment-387626308 shows an example of how to get all object nodes present in a given scene.  You are right that if you want to generate a pre-filtered set of scenes and episodes such that they are guaranteed to not contain a particular object you would do that by creating new `scenes.multiroom.csv` and `episode_states.suncg.csv`.  The first specifies the scene train/val/test split, and the second specifies pre-sampled episodes in those scenes for val and test runs.

Pull requests with new features are encouraged and highly appreciated!

closed 2018-05-21 by antonilo