minosworld/minos #57: OOM error: javascript heap out of memory

kvas7andy, 2018-05-11:
Hi,

I encountered OOM failaure of JS heap. Maybe that is related to this issue? https://github.com/nodejs/node/issues/17878

Part of the log:
```
/2a3aef3a6ce9e7aa7cea01b55236202f.arch.json'
2018-05-11 03:06:41,313 INFO b'preloading /root/hdd/data/suncg//wall/2a3aef3a6ce9e7aa7cea01b55236202f/2a3aef3a6ce9e7aa7cea01b55236202f.arch.json'
2018-05-11 03:06:41,315 INFO b'Removing subscribers for event drain [ 0 ]'
2018-05-11 03:10:15,343 INFO b'Timing loadScene: 214031.864ms'
2018-05-11 03:11:04,280 INFO b'Loaded p5dScene.2a3aef3a6ce9e7aa7cea01b55236202f bbdims: [10.260000616312027,2.7400000095367445,9.187790033035824]'
2018-05-11 03:11:04,281 INFO b'Retexture scene'
2018-05-11 03:11:30,318 INFO b''
2018-05-11 03:11:30,318 ERROR b'FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory'
2018-05-11 03:11:30,318 INFO b'<--- Last few GCs --->'
2018-05-11 03:11:30,318 ERROR b' 1: node::Abort() [node]'
2018-05-11 03:11:30,318 INFO b''
2018-05-11 03:11:30,319 ERROR b' 2: 0x8c807c [node]'
2018-05-11 03:11:30,319 INFO b'[17404:0x2f01900] 15396056 ms: Mark-sweep 3731.4 (4286.9) -> 3731.5 (4287.4) MB, 6260.1 / 0.0 ms  allocation failure GC in old space requested'
2018-05-11 03:11:30,319 ERROR b' 3: v8::Utils::ReportOOMFailure(char const*, bool) [node]'
2018-05-11 03:11:30,319 INFO b'[17404:0x2f01900] 15402331 ms: Mark-sweep 3731.5 (4287.4) -> 3731.6 (4242.4) MB, 6275.1 / 0.0 ms  last resort GC in old space requested'
2018-05-11 03:11:30,319 ERROR b' 4: v8::internal::V8::FatalProcessOutOfMemory(char const*, bool) [node]'
2018-05-11 03:11:30,319 INFO b'[17404:0x2f01900] 15408715 ms: Mark-sweep 3731.6 (4242.4) -> 3731.1 (4230.9) MB, 6383.7 / 0.0 ms  last resort GC in old space requested'
2018-05-11 03:11:30,319 INFO b''
2018-05-11 03:11:30,320 INFO b''
2018-05-11 03:11:30,320 INFO b'<--- JS stacktrace --->'
2018-05-11 03:11:30,320 INFO b''
2018-05-11 03:11:30,320 ERROR b' 5: v8::internal::Factory::NewFixedArray(int, v8::internal::PretenureFlag) [node]'
2018-05-11 03:11:30,320 INFO b'==== JS stack trace ========================================='
2018-05-11 03:11:30,320 INFO b''
2018-05-11 03:11:30,320 INFO b'Security context: 0xb3defb254d9 <JSObject>'
2018-05-11 03:11:30,320 INFO b'    1: add(this=0x172998aa09a9 <Set map = 0x1db526384cf9>,0x23282f791039 <String[41]: ut/6c0b1af96e05430a/a5d9816ea9cbec566.jpg>)'
2018-05-11 03:11:30,320 INFO b'    2: new Set(aka Set)(this=0x2cdad1302311 <the_hole>)'
2018-05-11 03:11:30,320 INFO b'    5: /* anonymous */ [/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:47939] [bytecode=0x14aab3a9bc51 offset=36](this=0xdcf83e8c419 <JSGlobal Object>,g=0x24e896465231 <JSArray[40...'
2018-05-11 03:11:30,320 ERROR b' 6: v8::internal::OrderedHashTable<v8::internal::OrderedHashSet, 1>::Allocate(v8::internal::Isolate*, int, v8::internal::PretenureFlag) [node]'
2018-05-11 03:11:30,320 INFO b''
2018-05-11 03:11:30,321 ERROR b' 7: v8::internal::OrderedHashTable<v8::internal::OrderedHashSet, 1>::Rehash(v8::internal::Handle<v8::internal::OrderedHashSet>, int) [node]'
2018-05-11 03:11:30,321 ERROR b' 8: v8::internal::OrderedHashTable<v8::internal::OrderedHashSet, 1>::EnsureGrowable(v8::internal::Handle<v8::internal::OrderedHashSet>) [node]'
2018-05-11 03:11:30,322 ERROR b' 9: v8::internal::Runtime_SetGrow(int, v8::internal::Object**, v8::internal::Isolate*) [node]'
2018-05-11 03:11:30,322 ERROR b'10: 0x2f0266c042fd'
2018-05-11 03:11:30,671 INFO Finished simserver stdout
2018-05-11 03:11:30,672 INFO Finished simserver stderr
``` 

[simserver.log](https://github.com/minosworld/minos/files/1994860/simserver.log)
[simulator.log](https://github.com/minosworld/minos/files/1994861/simulator.log)

Don't know exact problem.

I used my own code from forked repository, replacing parts of neural network.
https://github.com/kvas7andy/minos 
I started 4 threads which rendered 88x88 RGB images and segmentation labels (images with only one non zero B channel) 
I couldn't follow the memory consumption, but nodes worked on CPU and GPU at the same time. Can it be because of GPU memory consumption, or some memory leak in scene loader, could you please check this?

msavva, 2018-05-11:
Hi @kvas7andy ,

If this OOM issue occurs consistently for specific scenes, it might just be that the default heap size is insufficient.  You can change this parameter at https://github.com/minosworld/minos/blob/master/minos/lib/Simulator.py#L217 (`max-old-space-size=4096` sets the heap size in MB).

In case you observe the issue occurring for different scenes and after training for a while, please follow up as that could be evidence of a memory leak.

kvas7andy, 2018-05-12:
Not sure, but the error occured second time on the other scene. 

```
2018-05-12 03:11:56,512 INFO b'preloadData { loadSceneStatistics: true, loadSounds: false }'
2018-05-12 03:11:56,514 INFO b'Loading scene  p5dScene.93d2d55f161d256112636d371c99bd9d'
2018-05-12 03:11:56,535 INFO b'preloading /root/data/suncg//navgrid_10/93d2d55f161d256112636d371c99bd9d/93d2d55f161d256112636d371c99bd9d.furnished.grid.json'
2018-05-12 03:11:56,535 INFO b'preloading /root/data/suncg//wall/93d2d55f161d256112636d371c99bd9d/93d2d55f161d256112636d371c99bd9d.arch.json'
2018-05-12 03:11:56,535 INFO b'preloading /root/data/suncg//wall/93d2d55f161d256112636d371c99bd9d/93d2d55f161d256112636d371c99bd9d.arch.json'
2018-05-12 03:11:56,536 INFO b'Removing subscribers for event drain [ 0 ]'
2018-05-12 03:12:19,234 INFO b'Timing loadScene: 22702.211ms'
2018-05-12 03:12:33,868 INFO b'Loaded p5dScene.93d2d55f161d256112636d371c99bd9d bbdims: [5.929473876953125,2.7400000095367445,4.847501322627068]'
2018-05-12 03:12:33,869 INFO b'Retexture scene'
2018-05-12 03:13:04,168 INFO b''
2018-05-12 03:13:04,168 ERROR b'FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory'
2018-05-12 03:13:04,168 INFO b'<--- Last few GCs --->'
2018-05-12 03:13:04,169 INFO b''
2018-05-12 03:13:04,169 INFO b'[29292:0x41bd900] 11434036 ms: Mark-sweep 3649.7 (4257.9) -> 3649.7 (4258.4) MB, 8004.7 / 0.0 ms  allocation failure GC in old space requested'
2018-05-12 03:13:04,169 INFO b'[29292:0x41bd900] 11441533 ms: Mark-sweep 3649.7 (4258.4) -> 3649.5 (4217.9) MB, 7496.3 / 0.0 ms  last resort GC in old space requested'
2018-05-12 03:13:04,169 INFO b'[29292:0x41bd900] 11448959 ms: Mark-sweep 3649.5 (4217.9) -> 3649.5 (4205.9) MB, 7425.8 / 0.0 ms  last resort GC in old space requested'
2018-05-12 03:13:04,169 INFO b''
2018-05-12 03:13:04,169 INFO b''
2018-05-12 03:13:04,169 INFO b'<--- JS stacktrace --->'
2018-05-12 03:13:04,169 INFO b''
2018-05-12 03:13:04,169 INFO b'==== JS stack trace ========================================='
2018-05-12 03:13:04,169 INFO b''
2018-05-12 03:13:04,169 INFO b'Security context: 0x1b9d538254d9 <JSObject>'
2018-05-12 03:13:04,169 INFO b'    1: add(this=0x1a8f9725d509 <Set map = 0x267f2b884cf9>,0x217a7c3d05d9 <String[41]: ut/6c0b1af96e05430a/a5d9816ea9cbec566.jpg>)'
2018-05-12 03:13:04,169 INFO b'    2: new Set(aka Set)(this=0x36cd75502311 <the_hole>)'
2018-05-12 03:13:04,169 INFO b'    5: /* anonymous */ [/root/repos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:47939] [bytecode=0x2f2ca1a1a3b9 offset=36](this=0x12bd63d0c419 <JSGlobal Object>,g=0x1a8f9725d301 <JSArray[...'
2018-05-12 03:13:04,170 INFO b''
2018-05-12 03:13:04,171 ERROR b' 1: node::Abort() [node]'
2018-05-12 03:13:04,172 ERROR b' 2: 0x8c807c [node]'
2018-05-12 03:13:04,172 ERROR b' 3: v8::Utils::ReportOOMFailure(char const*, bool) [node]'
2018-05-12 03:13:04,173 ERROR b' 4: v8::internal::V8::FatalProcessOutOfMemory(char const*, bool) [node]'
2018-05-12 03:13:04,173 ERROR b' 5: v8::internal::Factory::NewFixedArray(int, v8::internal::PretenureFlag) [node]'
2018-05-12 03:13:04,174 ERROR b' 6: v8::internal::OrderedHashTable<v8::internal::OrderedHashSet, 1>::Allocate(v8::internal::Isolate*, int, v8::internal::PretenureFlag) [node]'
2018-05-12 03:13:04,174 ERROR b' 7: v8::internal::OrderedHashTable<v8::internal::OrderedHashSet, 1>::Rehash(v8::internal::Handle<v8::internal::OrderedHashSet>, int) [node]'
2018-05-12 03:13:04,175 ERROR b' 8: v8::internal::OrderedHashTable<v8::internal::OrderedHashSet, 1>::EnsureGrowable(v8::internal::Handle<v8::internal::OrderedHashSet>) [node]'
2018-05-12 03:13:04,176 ERROR b' 9: v8::internal::Runtime_SetGrow(int, v8::internal::Object**, v8::internal::Isolate*) [node]'
2018-05-12 03:13:04,176 ERROR b'10: 0x1e594ea842fd'
2018-05-12 03:13:30,895 INFO Finished simserver stdout
2018-05-12 03:13:30,895 INFO Finished simserver stderr
```

kvas7andy, 2018-05-14:
I have an update!
I found an issue about bad cache management of tensorflow, and consecutive advice to use `tcmalloc`. https://github.com/tensorflow/tensorflow/issues/6779

But using stated environmental variables interrupts node from executing `server.js`.
CPUPROFILE prevents from executing node itself, and `libtcmalloc_and_profiles.so.4` leads to the error of simserver:
```2018-05-14 00:17:49,126 INFO {'sim_git_hash': '3e31e16', 'stk_git_hash': '3e31e16', 'sim_id': 'sim01', 'machine': 'Ares'}
2018-05-14 00:17:49,156 INFO sim01:Starting sim server at /root/repos/minos/minos/server/server.js with port 58363
2018-05-14 00:17:50,282 INFO sim01:sim server has exited with rv -27```

Maybe here will be any workaround?
I still haven't seen knew OOM JS errors, after enlarging heap-size twice.

kvas7andy, 2018-05-15:
This can state that the problem is still because of JS.
Nodes are working hard  and after 900k steps rapidly increase the time of _rpc call.
This can be seen here in this log file (search with '_rpc call action lasted')
[rg_sun_mf_stnd_1x_pretr_norp_nstepTD100_localtmax100_1505.log](https://github.com/minosworld/minos/files/2004015/rg_sun_mf_stnd_1x_pretr_norp_nstepTD100_localtmax100_1505.log)

If looking from the ending try to find a place with column of such outputs with values of 16-18 seconds per _rpc call, which states its durarion!

Tensorflow is restricted to close if its own sess.run operation lasts more than 30 seconds.

kvas7andy, 2018-05-16:
@msavva 
There was no problem with nodes heap-size until I started extensively requesting `objectType` data along with `color` from nodes. May that be a problem?

kvas7andy, 2018-05-26:
Unfortunately, I get the same error all the time, have to reset environment after some delay.
At the same time nodes expand a lot RAM memory used. Maybe, the problem is with caching, that tensorflow doesn't catch up with node rendering?
Still hope you can advice me any workaround.

kvas7andy, 2018-05-27:
Checked with and without requesting `objectType` data from simulators and it seems, that RAM for nodes in increasing rapidly, maybe because `objectType` data is not deleted after scene reloading?

I can provide some screenshots from htop. Please, follow `MEM%` tag and top 10 process: `python3 main.py ...` main process; `node --max-old-space-size=4096 ...` - node simulators. I reduced `--max-old-space-size`, when generating `objectType`, a lot of RAM memory is relocated to swap, which is totally undesirable.

With 'objectType: False' in observations (so without request to generate semantic segmentation):
At start:
[image]
[image]

Middle of training (8 hours)
[image]

Close to 14 hours
[image]

With `objectType: True`:
At start:
[image]
After 2 hours:
[image]
2,5-3 hours and fast growing of RAM memory taken by node simulators:
[image]

kvas7andy, 2018-05-28:
@msavva 
Your help is much needed, because this issue halts all conducted research.

msavva, 2018-05-28:
@kvas7andy Apologies for the delay.  We are looking into this but meanwhile here are a couple of notes:
- As a workaround, you can increase the frequency of simulator process restarts by setting the `num_episodes_per_restart` parameter to be lower than the default 1000 (try a value that sets it below the time where you observe a significant memory increase). This way, each simulator node process will be restarted more frequently, alleviating the increase in the memory footprint.
- The memory usage in the above screenshots is actually not unexpectedly high.  In all cases, the total memory per process is less than the 4GB maximum set through the `max-old-space-size` parameter.  The node.js garbage collector can let the memory usage rise up to the max allocatable heap space.  If you are running 8 simulator node threads in parallel and set a 4GB maximum, then on a machine with 32GB it is very likely that you will end up swapping.

kvas7andy, 2018-05-29:
Hi @msavva,

Thank you, I will check this solution today. Of course, you are right about swap memory, I didn't notice that. I used to work on another workstation with 250 Gb RAM, which was enough in memory, but still learning process ended up with OOM JS error or after increasing space_size, node simulators slowed down with slow `rpc` calls.

kibeomKim, 2018-07-22:
Hi

I have the same issues with  @kvas7andy 
I read the above messages, unfortunately, it didn't solve yet.

I run MINOS with 15 multiprocesses after 1 and half hour, it occurred.
Is there a new method to solve it?
if so, please let me know

thanks

```
2018-07-22 07:03:26,303 INFO b'<--- Last few GCs --->'
2018-07-22 07:03:26,303 INFO b''
2018-07-22 07:03:26,303 INFO b'[10363:0x2a04660]  3885177 ms: Mark-sweep 3719.7 (4179.5) -> 3704.6 (4180.0) MB, 4163.4 / 0.0 ms  (average mu = 0.260, current mu = 0.205) allocation failure scavenge might not succeed'
2018-07-22 07:03:26,303 INFO b'[10363:0x2a04660]  3890225 ms: Mark-sweep 3720.6 (4180.5) -> 3705.6 (4181.0) MB, 3995.5 / 0.0 ms  (average mu = 0.235, current mu = 0.208) allocation failure scavenge might not succeed'
2018-07-22 07:03:26,303 INFO b''
2018-07-22 07:03:26,303 INFO b''
2018-07-22 07:03:26,303 INFO b'<--- JS stacktrace --->'
2018-07-22 07:03:26,303 INFO b''
2018-07-22 07:03:26,303 INFO b'==== JS stack trace ========================================='
2018-07-22 07:03:26,303 INFO b''
2018-07-22 07:03:26,303 INFO b'    0: ExitFrame [pc: 0xa4a51841bd]'
2018-07-22 07:03:26,303 INFO b'Security context: 0xe126c19e6c9 <JSObject>'
2018-07-22 07:03:26,304 INFO b'    1: /* anonymous */ [0x1714b45ed6a9] [/home/minos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:~112012] [pc=0xa4a5dab192](this=0x4db5e757331 <SimState map = 0x2d594ec7eb11>)'
2018-07-22 07:03:26,304 INFO b'    2: /* anonymous */ [0x25faab0520f1] [/home/minos/minos/minos/server/node_modules/sstk/client/build/STK-core.bundle.js:109853] [bytecode=0x3ffa96adaab1 offset=...'
2018-07-22 07:03:26,304 INFO b''
2018-07-22 07:03:26,304 ERROR b'FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory'
2018-07-22 07:03:26,344 ERROR b' 1: 0x89c2f0 node::Abort() [node]'
2018-07-22 07:03:26,345 ERROR b' 2: 0x89c33c  [node]'
2018-07-22 07:03:26,355 ERROR b' 3: 0xa8f05e v8::Utils::ReportOOMFailure(v8::internal::Isolate*, char const*, bool) [node]'
2018-07-22 07:03:26,367 ERROR b' 4: 0xa8f278 v8::internal::V8::FatalProcessOutOfMemory(v8::internal::Isolate*, char const*, bool) [node]'
2018-07-22 07:03:26,368 ERROR b' 5: 0xe7cad2  [node]'
2018-07-22 07:03:26,370 ERROR b' 6: 0xe88a2a v8::internal::Heap::PerformGarbageCollection(v8::internal::GarbageCollector, v8::GCCallbackFlags) [node]'
2018-07-22 07:03:26,370 ERROR b' 7: 0xe89394 v8::internal::Heap::CollectGarbage(v8::internal::AllocationSpace, v8::internal::GarbageCollectionReason, v8::GCCallbackFlags) [node]'
2018-07-22 07:03:26,373 ERROR b' 8: 0xe8bcc5 v8::internal::Heap::AllocateRawWithRetry(int, v8::internal::AllocationSpace, v8::internal::AllocationAlignment) [node]'
2018-07-22 07:03:26,374 ERROR b' 9: 0xe541a4 v8::internal::Factory::NewFillerObject(int, bool, v8::internal::AllocationSpace) [node]'
2018-07-22 07:03:26,377 ERROR b'10: 0x10f563e v8::internal::Runtime_AllocateInNewSpace(int, v8::internal::Object**, v8::internal::Isolate*) [node]'
2018-07-22 07:03:26,377 ERROR b'11: 0xa4a51841bd'
```

ifyum, 2019-09-16:
in package.json change 
"webpack-dev-server": "node --max_old_space_size=4096 to 2096
"webpack": "node --max_old_space_size=4096 to 2096 

result:
  "webpack-dev-server": "node --max_old_space_size=2096 node_modules/webpack-dev-server/bin/webpack-dev-server.js",
    "webpack": "node --max_old_space_size=2096 node_modules/webpack/bin/webpack.js"

problem solved.