minosworld/minos #60: Socket connection problem

antonilo, 2018-05-21:
After installation I am experiencing the following problem when running one of the basic testing examples.
I was not able yet to figure out a solution. It would be great if you could provide some hints on how to solve it!

```
2018-05-21 23:30:22,307 WARNING localhost:16666/socket.io [engine.io waiting for connection] 
HTTPConnectionPool(host='localhost', port=16666): Max retries exceeded with url: /socket.io/?EIO=3&transport=polling&t=1526909422290-0
(Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f6b3e616048>:
 Failed to establish a new connection: [Errno 111] Connection refused',))
```

closed 2018-05-21 by antonilo

CMinhBui, 2018-07-01:
What happened with this issue? I ran into the same problems, have you solved it?

mcimpoi, 2018-07-20:
Did you check the logs? I get this when trying to debug the code in Pycharm, but work when I run the pygame script from terminal.

antonilo, 2018-07-20:
Yes checking the logs was the solution. I've realized it shortly after opening the issue so I've closed it myself.

antonilo, 2018-12-06:
Every time you run the simulator, some logs file are saved (usually under the logs folder). Try to have a look there to see what is going wrong with your setup!