minosworld/minos #61: some questions related to RoomSimulator

nina124, 2018-05-21:
Hi,
I have been investigating minos, I have some questions about RoomSimulator.
1. It seems RoomSimulator is a wrapper of Simulator: it enables some auxiliary functions, like scene filtering, episode_schedular and so on. Am I right?
2. By default, RoomSimulator reads scenes and episode states from `minos_basedir/minos/data/scenes.multiroom.csv` and `minos_basedir/minos/data/episode_states.suncg.csv`, how are these scenes and episode states generated?
3. In `minos_basedir/minos/data/episode_states.suncg.csv`, it sets infos of `startX/Y/Z, goalX/Y/Z`, but I cannot figure out how they are used in RoomSimulator. I just found that the RoomSimulator calls `episode_schedular` when starting a new episode, and `episode_schedular` only returns the sceneId, ignoring other infos like `startX/Y/Z, goalX/Y/Z`.
4. I found two files about roomType, one is `minos/server/node_modules/sstk/server/static/data/suncg/roomTypes.csv`, the other is `data/roomTypes.suncg.csv`. I am confused their contents are different. Which one should be used? Are they used in different scenarios?

Looking forward to an early reply. Thanks!

msavva, 2018-05-22:
Hi @nina124 

Here are some brief pointers for each of the questions that should hopefully clarify things:
1. Yes, that is correct.  RoomSimulator wraps Simulator to add episode scheduling and other auxiliary functionality.
2. The `episode_states` files are generated using the `minos/tools/generate_episodes.py` script.  The `scenes.multiroom.csv` files contain manually chosen scene ids that we chose to create our benchmark train/val/test split.  The columns are populated (first automatically with tools that are not part of this codebase) and then manually verified.
3. The goal specification is passed through the RoomSimulator onto Simulator and then eventually used to configure the backing node.js Simulator instance.  The actual parsing of the goal specification is handled by the `SceneState` class once the configuration is set (see https://github.com/smartscenes/sstk/blob/master/client/js/lib/sim/SimState.js for the logic that handles this configuration).  Please note that there is a pending issue (#38) with handling goal setting, the resolution of which we plan to test and incorporate soon.
4. The `minos/server/node_modules/sstk/server/static/data/suncg/roomTypes.csv` file is the full category set for SUNCG room types.  In contrast, the `data/roomTypes.suncg.csv` file is part of the MINOS codebase and specifies a simplified set of room types (by collapsing several types of rooms into coarser categories) which we used for the RoomGoal experiments.  The latter is what is used by default.

Let us know if anything is unclear.

nina124, 2018-05-22:
Thanks @msavva 
So currently, I cannot set goal positions, but it will be incorpated soon. And the suncg_minos dataset uses roomTypes defined in `data/roomTypes.suncg.csv`, and I can totally ignore `minos/server/node_modules/sstk/server/static/data/suncg/roomTypes.csv`. 
Right?

nina124, 2018-05-22:
I am checking because in #14 , u mentioned the roomtype and objecttype files are 
https://github.com/smartscenes/sstk/blob/master/server/static/data/suncg/roomTypes.csv
https://github.com/smartscenes/sstk/blob/master/server/static/data/suncg/objectTypes.csv