minosworld/minos #62: Can I set the agent start position in some specific room?

nina124, 2018-05-21:
Hi,
Is it possible to configure the agent to start from specific room?
Can it be achieved by setting the agent starting position and making the position located in the specific room?

msavva, 2018-05-22:
The currently exposed API allows for setting the agent to a specific position and angle through the `move_to` call.  Information on what room a given position falls in can be retrieved using the function https://github.com/smartscenes/sstk/blob/master/client/js/lib/sim/SimState.js#L487 .  However, please note that this function is not currently exposed to the python API.  This position-to-room information is available in the NavGrid files (i.e. `.grid.json` format files) as the `roomIndex` field at a set of precomputed position cells for the entire navigable space of each scene. See https://github.com/minosworld/minos/wiki/NavGrid-format for details on the format of these files.  If you want to create a set of starting and goal positions in particular rooms as a preprocess, you can probably load this metadata outside the simulator API.  Otherwise, you would need to expose the `SimState. getPositionMetadata` function to the Python API by adding an rpc wrapper method similator to the existing ones for other functions.

nina124, 2018-05-22:
Can I use `tools/generate_episodes.py` to generate some presampled episodes and keep episodes with starting points in the room I set? 
Will `tools/generate_episodes.py` generated episodes always be navigatable? Do I need to check navigability by `minos/server/check_episode_states.js`?
Meanwhile, I cannot generate episodes by `tools/generate_episodes.py`, it gets stuck when resetting. The command is `python3 generate_episodes.py --output test_generate_episode_states.csv --samples_per_scene 3`.