minosworld/minos #65: Minos installation problem

lulienhsi, 2018-06-12:
Environment
OS: Ubuntu 18.04 LTS
Python: 3.6.5
node: v10.4.0
only SUNCG data is installed in the $HOME/work directory.

Hi, I have followed the installation instruction in https://github.com/minosworld/minos. 
While I am running   `python3 -m minos.tools.pygame_client`  , I have confronted some problems. 
```
fatal: Needed a single revision
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/lulienhsi/Downloads/minos-master/minos/tools/pygame_client.py", line 457, in <module>
    main()
  File "/home/lulienhsi/Downloads/minos-master/minos/tools/pygame_client.py", line 415, in main
    sim = Simulator(vars(args))
  File "/home/lulienhsi/Downloads/minos-master/minos/lib/Simulator.py", line 109, in __init__
    cwd=stk_sim_path).rstrip(),
  File "/usr/lib/python3.6/subprocess.py", line 336, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.6/subprocess.py", line 418, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['git', 'rev-parse', '--short', 'HEAD']' returned non-zero exit status 128.
2018-06-12 11:24:18,925 INFO sim00:Stopping the simulator
2018-06-12 11:24:18,926 INFO Counter()
2018-06-12 11:24:18,926 INFO sim00:Stopping child servers
2018-06-12 11:24:18,926 INFO sim00:Stopped child servers
2018-06-12 11:24:18,927 INFO sim00:Simulator killed.

```
Have someone confront this problem or anyone can help me to solve this? Thanks!

msavva, 2018-06-12:
Hi @lulienhsi ,
Thank you for trying out MINOS!

This issue appears to be due to a behavior change in recent git versions.  You can confirm whether that is the problem by running `git rev-parse --short HEAD` in your local minos git repository directory and seeing if it produces an error message.  If that is indeed the case, you can temporarily workaround this issue by replacing the two lines at https://github.com/minosworld/minos/blob/master/minos/lib/Simulator.py#L108 with empty strings instead to avoid the system call to the git binary.  This hash information is only used for log keeping purposes and is safe to remove.  Let us know if you confirm this to be the issue and send us the version of git that you are using on your system.

lulienhsi, 2018-06-12:
lulienhsi@lulienhsi-Lenovo-ideapad-700-15ISK:~/Downloads/minos-master$ ```
git rev-parse --short HEAD
fatal: Needed a single revision
```
After I replace the two lines at https://github.com/minosworld/minos/blob/master/minos/lib/Simulator.py#L108
The fatal problem has gone, but still cannot open the simulator. :(

```
`lulienhsi@lulienhsi-Lenovo-ideapad-700-15ISK:~/Downloads/minos-master$ python3 -m minos.tools.pygame_client`
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/lulienhsi/Downloads/minos-master/minos/tools/pygame_client.py", line 457, in <module>
    main()
  File "/home/lulienhsi/Downloads/minos-master/minos/tools/pygame_client.py", line 415, in main
    sim = Simulator(vars(args))
  File "/home/lulienhsi/Downloads/minos-master/minos/lib/Simulator.py", line 109, in __init__
    cwd=stk_sim_path).rstrip(),
  File "/usr/lib/python3.6/subprocess.py", line 336, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.6/subprocess.py", line 403, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.6/subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1344, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: ''
2018-06-12 15:04:25,055 INFO sim00:Stopping the simulator
2018-06-12 15:04:25,055 INFO Counter()
2018-06-12 15:04:25,056 INFO sim00:Stopping child servers
2018-06-12 15:04:25,056 INFO sim00:Stopped child servers
2018-06-12 15:04:25,056 INFO sim00:Simulator killed.

```

msavva, 2018-06-12:
From the error above it looks like there is still a system call on that line -- perhaps you replaced the arguments of the call with an empty string, instead of setting the variable in the line to an empty string?
Try the following:
```python
stk_sim_path = ''
sim_git_hash = ''
```

msavva, 2018-06-12:
Actually, setting the above to empty strings you will run into an issue in the following lines which try to index into the string.  You can remove the above two variables entirely and just set the `info` members directly to any string below at https://github.com/minosworld/minos/blob/master/minos/lib/Simulator.py#L115

lulienhsi, 2018-06-13:
```
lulienhsi@lulienhsi-Lenovo-ideapad-700-15ISK:~/Downloads/minos-master$ python3 -m minos.tools.pygame_client

Starting simulator...
2018-06-13 10:31:59,281 INFO sim00:Starting sim server at /home/lulienhsi/Downloads/minos-master/minos/server/server.js with port 35482
Traceback (most recent call last):
  File "/home/lulienhsi/Downloads/minos-master/minos/tools/pygame_client.py", line 438, in main
    ep_info = sim.start()
  File "/home/lulienhsi/Downloads/minos-master/minos/lib/Simulator.py", line 311, in start
    started = self.start_child_servers()
  File "/home/lulienhsi/Downloads/minos-master/minos/lib/Simulator.py", line 245, in start_child_servers
    self._sio = SocketIO(self.params.host, self.params.port)
  File "/home/lulienhsi/.local/lib/python3.6/site-packages/socketIO_client/__init__.py", line 345, in __init__
    resource, hurry_interval_in_seconds, **kw)
  File "/home/lulienhsi/.local/lib/python3.6/site-packages/socketIO_client/__init__.py", line 58, in __init__
    self._transport
  File "/home/lulienhsi/.local/lib/python3.6/site-packages/socketIO_client/__init__.py", line 66, in _transport
    self._engineIO_session = self._get_engineIO_session()
  File "/home/lulienhsi/.local/lib/python3.6/site-packages/socketIO_client/__init__.py", line 80, in _get_engineIO_session
    transport.recv_packet())
  File "/home/lulienhsi/.local/lib/python3.6/site-packages/socketIO_client/transports.py", line 92, in recv_packet
    for engineIO_packet in decode_engineIO_content(response.content):
  File "/home/lulienhsi/.local/lib/python3.6/site-packages/socketIO_client/parsers.py", line 95, in decode_engineIO_content
    content, content_index)
  File "/home/lulienhsi/.local/lib/python3.6/site-packages/socketIO_client/parsers.py", line 202, in _read_packet_length
    while get_byte(content, content_index) not in [0, 1]:
  File "/home/lulienhsi/.local/lib/python3.6/site-packages/socketIO_client/symmetries.py", line 36, in get_byte
    return six.indexbytes(x, index)
IndexError: index out of range
Error running simulator. Aborting.
2018-06-13 10:32:01,298 INFO sim00:Stopping the simulator
2018-06-13 10:32:01,298 INFO Counter()
2018-06-13 10:32:01,298 INFO sim00:Stopping child servers
2018-06-13 10:32:01,298 INFO sim00:Killing sim pid 25094
2018-06-13 10:32:01,305 INFO sim00:Stopped child servers
2018-06-13 10:32:01,305 INFO sim00:Simulator killed.

```

simserver.log 
```
2018-06-13 10:31:59,842 INFO b'Configuration { fileCache: { size: 50 },'
2018-06-13 10:31:59,842 INFO b'  imageCache: { size: 200 },'
2018-06-13 10:31:59,842 INFO b'  imageQueue: { concurrency: 4 },'
2018-06-13 10:31:59,842 INFO b"  base_url: '/home/lulienhsi/work/',"
2018-06-13 10:31:59,842 INFO b'  assets_url:'
2018-06-13 10:31:59,842 INFO b"   'file:///home/lulienhsi/Downloads/minos-master/minos/server/node_modules/sstk/ssc/../server/static' }"
2018-06-13 10:32:00,064 INFO b'Waiting for client connection on port 35482'
2018-06-13 10:32:00,080 ERROR b'Cannot register unknown asset 2d3ds'
2018-06-13 10:32:00,081 INFO b'register ../server/static/data/matterport/matterport3d.json'
2018-06-13 10:32:00,082 INFO b'Registered asset group: mp3d'
2018-06-13 10:32:00,085 INFO b'Got 90 assets'
2018-06-13 10:32:00,140 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
2018-06-13 10:32:00,141 INFO b'Replace asset group p5d'
2018-06-13 10:32:00,141 INFO b'Registered asset group: p5d'
2018-06-13 10:32:00,198 INFO b'Got 2623 assets'
2018-06-13 10:32:00,994 INFO b'Loaded lightSpecs for 202 models.'
2018-06-13 10:32:00,994 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
2018-06-13 10:32:00,997 INFO b'Replace asset group p5dTexture'
2018-06-13 10:32:00,997 INFO b'Registered asset group: p5dTexture'
2018-06-13 10:32:01,152 INFO b'Got 67072 assets'
2018-06-13 10:32:01,298 INFO Waiting for simserver stdout to finish
2018-06-13 10:32:01,305 INFO Finished simserver stderr
2018-06-13 10:32:01,305 INFO Finished simserver stdout
2018-06-13 10:32:01,305 INFO Waiting for simserver stderr to finish
```

Some new problems after the steps above is done, thanks again!

msavva, 2018-06-13:
The above errors appear to be due to an incorrect socketio_client package version.  Try to follow the instructions here: https://github.com/minosworld/minos/blob/master/FAQ.md#i-get-an-error-from-the-socketio_client-package to manually install the specific package version that is required as a dependency

mcimpoi, 2018-07-19:
Which version of node are you using? And socket.io? (check in package-lock.json, in server folder).
The version we got it working with was 1.7.4 (same as in the git repo).

With node 10.7.0, and npm audit fix /npm audit fix --force, socket.io was upgraded to 2.1.1, which doesn't seem to work with the python socketIO-client-2. 

Hope this helps.