minosworld/minos #66: How to retrieve camera depth?

kanwal41g, 2018-06-17:
I am a beginner at MINOS. 
In sim_config.py, in following line
observations': {'color': True, 'forces': False, 'audio': False, 'objects': False, 'depth': True, 'map': False},
if i turn the depth True, why am i not getting the display of camera depth? 

Moreover, in what file are all these variables information is getting stored?

msavva, 2018-06-18:
Hi @kanwal41g ,

Thank you for trying out MINOS!

The gym demo code provides examples of how to retrieve depth data (as well as the other sensor types): https://github.com/minosworld/minos/blob/master/gym/demo.py
If you want to see an interactive visualization including the depth sensor, please check out the code on the `dev` branch, follow the build instructions and then try the command `python3 -m minos.tools.pygame_client --depth -s normal -s objectId -s objectType -s map --navmap --width 128 --height 128`

efyphil, 2018-08-16:
Can I ask some additional question? How i can add some depth noise, I found --depth_noise solution but I don't know which argument I can add