minosworld/minos #70: Terminal Display Modification

kanwal41g, 2018-07-15:
I have seen that the following file is responsible for most of the display on the Terminal as the simulator works.... 

pygame_client.py

However, i dont see  print "orientation" anywhere in this file. Can you please guide as how to access that?

msavva, 2018-07-18:
The `pygame_client.py` implements the interactive control interface.  If you want to look at an example of reading back agent state data programmatically, the `gym/demo.py` example is probably a much better start.  In that file, you can pull out the agent's orientation using `info['agent_state']['orientation']`.

kanwal41g, 2018-07-18:
I have checked the .demo file but it has the following commands: 
color = observation["observation"]["sensors"]["color"]["data"]

It doesnt have this command anywhere. info['agent_state']['orientation'] which you stated in your reply. 
kindly guide how do i add this command.

Also, i have the GYM folder downloaded but when i give the command .demo.py/ it gives me following error. bash: .demo.py/: No such file or directory

msavva, 2018-07-18:
Please make sure you have followed the gym demo setup instructions described in https://github.com/minosworld/minos/tree/master/gym 
You should then be able to run from within the gym directory through `python3 demo.py`
What I stated is code that you would have to write yourself inline with the demo code.  For example, you could `print(info['agent_state']['orientation'])` after line https://github.com/minosworld/minos/blob/master/gym/demo.py#L23 which performs the simulation step and returns the observation and info variables

kanwal41g, 2018-07-28:
This command do work but the simulatuon speed is very high. What can i do to slow it down. Or control it myself?