minosworld/minos #72: Collision = True, but 'forces': {'type': 'force', 'data': array([0., 0., 0., 0.]

ducha-aiki, 2018-07-17:
It usually happens, when agent diagonally moves into chairs or something like this. I often cannot see obstacle visually, because it is too low. But high enough to block the way.
Is is intentional behaviour or a bug?

msavva, 2018-07-18:
The `forces` field is an array which reports a contact force value for each force sensor.  The contact force for each sensor is computed by detecting colliding surfaces within a threshold distance from the sensor position in the direction of the sensor orientation.  The default force sensor array in `minos/config/sensors.yml` is composed of four sensors placed on the surface of a cylinder (pointing outward from the center).  Thus, it is possible to collide with a surface while all four sensors report no contact force, especially if the motion is diagonal to the agent orientation.

ducha-aiki, 2018-07-18:
Thank you for clarification! 
I was under impression, that force sensor is not point, but whole sector

closed 2018-07-18 by ducha-aiki