minosworld/minos #73: Agent pose doesn`t reset in new episode

ducha-aiki, 2018-07-18:
Hi,

Agent pose doesn`t reset in new episode. Say, agent finish episode looking down - and is respawn in new episode also looking down. It makes life quite hard :) Is such behavior intentional?

msavva, 2018-07-18:
Resetting the view tilt is indeed something that should happen on a new episode.  Thanks for reporting this issue.  Looking up and down were not initially in the action space of the agent, thus the reset does not account for tilt changes. :( I will mark this as a bug and push a fix to the `dev` branch soon.  In the meantime, you can reset the tilt angle on new episodes using `env.simulator.move_to(tilt=0)`.

ducha-aiki, 2018-07-18:
@msavva Thanks a lot! 
I use lookDown for some ad-hoc scale calibration.