minosworld/minos #75: How to paste a picture on the wall on MINOS environment

kanwal41g, 2018-07-22:


msavva, 2018-08-04:
Can you clarify what you mean by "pasting a picture"?  Do you mean changing the texture of the whole wall, or do you mean adding a picture object to the wall?

kanwal41g, 2018-08-05:
i mean adding a picture object to the wall

msavva, 2018-08-05:
Adding a picture object to a wall is the same functionality as #74 -- we will plan to implement this functionality for an upcoming release.
Changing the texture of the whole wall is possible in a hacky way currently by modifying the `suncg/wall/<house_id>/arch.json` file for specific houses that you are interested in.  If you look in those files, elements of type `wall` contain a `materials` array where the `texture` is the basename of a texture file in the `suncg/texture` directory.  You can introduce your own texture images this way.

Usama323, 2018-08-18:
Can you Kindly explain what I have to edit in arch.jason file ? do I need to add name of my image or what ? kindly explain !! Also I need to add the image in texture directory ?

msavva, 2018-08-18:
Hi @Usama323 , thanks for trying out MINOS.  Here is an example for file `wall/0004d52d1aeeb8ae6de39d6bd993e992/0004d52d1aeeb8ae6de39d6bd993e992.arch.json`.  The first `Wall` element with id `0_0_0` has the following materials entry:
```json
            "materials": [
                {
                    "name": "inside",
                    "texture": "wallp_0",
                    "diffuse": "#e61c52"
                },
                {
                    "name": "outside",
                    "texture": "bricks_1",
                    "diffuse": "#ffffff"
                }
            ],
```
You can change the `"texture": "wallp_0",` entry to `"texture": "TEXTURE_FILE_BASENAME_NO_EXTENSION"` where `TEXTURE_FILE_BASENAME_NO_EXTENSION` refers to either an existing texture image file in the `texture` directory, or a new texture image file you copied into there.

Usama323, 2018-08-19:
I have tried this solution , the texture of wall, floor or a ceiling is changing but the problem is that the image i have selected to be displayed (as texture) is not coming on that wall , I have tried many options but was unable to get my texture/image on walls. Instead a dummy image/texture came on wall every time i run the minos with same house id . Can you kindly tell me what else do I need to do ?, as image i wanted to be my texture is there in the said directory as well . Can you kindly help me regarding this issue @msavva ??

msavva, 2018-08-19:
Have you tried changing the texture file base name to one that already
exists? For example, copy one of the other texture names in the json file.
If that works, then the issue must be due to the new texture image that you
copied into the texture folder. Check that the base name (no extension) of
the filename is what you have copied into the json file.  If that still
does not work, check that the image format is the same as other textures
(png or jpg).

Usama323, 2018-08-19:
1. Yes I have tired texture already in the folder but I am getting same results, that is another dummy image/texture instead of the selected one 

2. I am copying base name of texture file with out its extension in .json file

3. I have copied its name by clicking on rename then copy it and also double check its base name befor executing the code .

4. The image I have copied my self (my texture) as same extension as other images have that is .jpg

5. Funny thing about this even if I change texture for one wall , multiple walls in multiple rooms are changed with random texture on them .

Kindly help me out , looking forward for your reply !

msavva, 2018-08-21:
I am not able to reproduce the issue you are having unfortunately.  The example below shows before and after images for house id `00a76592d5cc7d92eef022393784a2de`.  The "after" image on the right shows the changed wall texture when replacing wall `0_0_0`'s inside material with `bricks_1` (instead of the original `wallp_1_2` as shown on the left)

[image]

Please try to do the above modification in the `00a76592d5cc7d92eef022393784a2de.arch.json` file and check whether you observe the same change or not.

Usama323, 2018-08-27:
Thanks a lot @msavva I have tried it and it is successfully changed after changing in following two files 
sim_config.py and sim_args.py , which are located in "minos/minos/config" folder. I have changed retexture=False to retexture=True in sim_config.py file and set  const='False', default=False in retexture portion of sim_args.py file.