minosworld/minos #79: Smooth movements with  generate_episodes.py

CaptainTrunky, 2018-08-02:
I would like to sample from MINOS random motion tracks, representing an agent moving and sensing surroundings. However, my application requires that the agent should move as smooth as possible.

I'm running generate_episodes.py as follows:

`python  minos/tools/generate_episodes.py -s objectType  --samples_per_scene 10 --scene_ids d0c3971878bfb42e81876c6a6501bda8 --scenes minos/data/scenes.multiroom.csv --output train.txt --save_png --split train --agent_config agent_firstperson --forces false --width 640 --height 480`

Firstly, generated images show that agent **warps** quite severely, presumably, traversing navgrid and ignoring *--agent_config agent_firstperson*.

Is there a way to suppress such warping behavior and generate more or less smooth traversals in non-iteractive fashion?

I use *dev* branch, rev e0a805598668

msavva, 2018-08-05:
Hi @CaptainTrunky ,
Thanks for trying out MINOS!  The `generate_episodes.py` script generates pairs of start and goal configurations (i.e. what constitutes the definition of one episode).  So the images that you see in the output in fact correspond to the start configuration for each new episode (thus the severe warping).

In order to generate motion trajectories, you might want to adapt an example like the `gym/demo.py` to save out the agent sensor frame at each step in a sequence.  The existing `--save_observations` argument will save frames but it overwrites the files at each step (it was just intended as an example of how to retrieve the data).

CaptainTrunky, 2018-08-08:
Hi @msavva 
Thanks for pointing me at `gym/demo.py`, it looks like this is the perfect starting point for me. The script kind of works, but I had to comment out the following lines:
```
if 'goalObservations' in result:
    del result['goalObservations']
```
`result` occasionally equals `None` at some cases. Currently the script is working, except that an agent behaves pretty chaotic and often stuck at walls and corners.

closed 2018-08-10 by CaptainTrunky