minosworld/minos #80: [bug] When using look up and down, camera can go through walls

ducha-aiki, 2018-08-03:
When using look up and down near wall, camera can go through walls. Agent cannot "walk" through them although.

Here is video from online minos sim, showing such behavior

https://drive.google.com/open?id=1D-sK9iZ2m_V5Z3OGUCpZq5EG9KeFuC2H

msavva, 2018-08-04:
This is indeed a bug in handling look up/down actions.  We are investigating and will commit a fix to the dev branch.  Thank you for reporting this bug.

msavva, 2018-08-05:
@ducha-aiki ,
We have committed a fix -- the bug was in the dependency library sstk (see https://github.com/smartscenes/sstk/commit/63b0980b7b33116f922a2152189584b7a1756ba0).  If you pull the `dev` branch of minos, and re-run `npm install` in `minos/server` (remove the `node_modules` directory if it exists prior to doing this) the update should be applied.  Once you confirm that this works as intended we can resolve the issue.

ducha-aiki, 2018-08-09:
Hmm...so far npm install is stuck on stage:

> extract:esprima-six-jpike: sill extract esprima-six-jpike@1.1.1-dev-harmony extracted to 

I have removed node_modules, it doesn`t help.

ducha-aiki, 2018-08-10:
I have managed to install, now no wall walking.

closed 2018-08-10 by ducha-aiki

ducha-aiki, 2018-11-02:
@ZhuFengdaaa I just waited for two hours and it was done...