minosworld/minos #83: Generating simple routes through a scene

CaptainTrunky, 2018-08-10:
I would like to generate simple and obvious routes from a starting point to a goal using some simplistic approaches like DFS, A* or Lee algorithm. My assumption is that for doing it I need access to nav_grid, for example, by reading it from harddrive using scene_id or by providing corresponding API on a server side.

I'm facing the following issues at the moment:

- Is there any other recommended way of doing something similar?

- Is there a manual on how to read and manipulate nav_grid?

msavva, 2018-08-12:
To generate routes on the shortest path you would need to write controller logic around the step function which takes the shortest path information (`observation["observation"]["measurements"]["shortest_path_to_goal"]`) at each step and determines an appropriate action to take (turning to stay on the shortest path, and moving forward along the path).

If you want to directly leverage the precomputed occupancy grids in the nav_grid files instead, a description of the format is available at https://github.com/minosworld/minos/wiki/NavGrid-format

ngthanhtin, 2019-06-24:
hi @CaptainTrunky, did you write code to generate simple route and make the agent follow it ?. I have implemented A* algorithm but I dont know how to integrate it in MINOS.

CaptainTrunky, 2019-06-24:
Hi @ngthanhtin . Yes, I did it a while ago. I'm not in position to publish code, but the idea is very simple:

1. Inherit from Simulator by implementing AStarSim
2. Extend it by something similar to:
def get_trajectory(self):
    start_point = random()
    end_point = random()

    observation = self.sim.get_last_observation()
    scene_map = observation.get('map')

    path = self.a_star(map, start_point, goal_point)

3. Reuse gym/demo.py with your simulator class.
4. And you're done!

ngthanhtin, 2019-06-24:
@CaptainTrunky but the positions in MINOS are in floating points, A* is just do with integer, how can I convert it ?

CaptainTrunky, 2019-06-24:
> @CaptainTrunky but the positions in MINOS are in floating points, A* is just do with integer, how can I convert it ?

In both Matterport3d and SUNCG there is metadata describing an occupancy grid for a scene. I'm not familiar with Matterport3D, but for SUNCG you could it them at 'suncg/nav_grids' folder. It's a binary file with simple structure:
- width, height, depth - integers
- matrix 4x4 - affine transformation from world coordinates (agent's position, float) to voxel grid - integers
- 3D occupancy grid, float, size of width x height x depth - 1.0 correspond to 'occupied', 0.0 - 'empty'

So you could read affine transformation matrix to cast your current float position to cells' indices in a map.

ngthanhtin, 2019-06-24:
Thank you, I will give it a try :)

CaptainTrunky, 2019-06-24:
> Thank you, I will give it a try :)

Be aware that by default Minos uses 500 predefined scenes from SUNCG and there is hardcoded list of scenes. If you would like to sample other scenes, this requires some additional tuning.

ngthanhtin, 2019-06-24:
hi @CaptainTrunky I've read the format of grid file as link follow: https://github.com/minosworld/minos/wiki/NavGrid-format but I don't find any 4x4 matrix as you've said before. Did you have to convert any data from grid file ?

CaptainTrunky, 2019-06-24:
> hi @CaptainTrunky I've read the format of grid file as link follow: https://github.com/minosworld/minos/wiki/NavGrid-format but I don't find any 4x4 matrix as you've said before. Did you have to convert any data from grid file ?

I'm talking about SUNCG native .grd files, not about MINOS generated data. Try

> find /path/to/downloaded/suncg -name *grd

ngthanhtin, 2019-06-24:
I don't use MINOS generated data, I use the grid file from Matterport3d data, the file is available when I download the data. Sorry if I don't get what you say :)

CaptainTrunky, 2019-06-24:
> I don't use MINOS generated data, I use the grid file from Matterport3d data, the file is available when I download the data. Sorry if I don't get what you say :)

Oh, I'm not familiar with Matterport3D. Try checking data loader at Simulator for Matterport3D, it should be somewhere at node.js wrapper that simulates scene & agent

ngthanhtin, 2019-06-25:
I've tried but I don't know how to get the map. As you said before, you got the map by scene_map = observation.get('map'), but it will return an image with the format of (w,h,4). 
This is not a map I want to get, I need a 3D map. 
Did you do path finding on scene_map = observation.get('map') or 3D map ?