minosworld/minos #84: Incorrect depth or very "systematic" noise?

ducha-aiki, 2018-08-13:
Hi,

I am observing a bit weird behaviour, when looking near wall. Images of RGB and depth got from the simulator "dev" branch are below.
Agent is standing near the wall at the right and looking at the corner ( around x = 80 vertical line).

[image]
[image]

The question is, why such "steps" pattern is observed in the depth image, is it intended to be like this?

msavva, 2018-08-13:
Hi @ducha-aiki ,  can you give a few more details on the configuration you are running when you get these depth frames (and how you plot the output depth frame above)?  This "staircasing" effect in the depth frames can occur if the depth sensor in `sensors.yml` is set to something other than `encoding: depth`, or if the returned float32 depth values are cast to another data type.  If you run the interactive interface using `python3 -m minos.tools.pygame_client --depth` do you see any staircasing effect in the interactively visualized depth frames?

msavva, 2018-08-15:
@ducha-aiki can you check that you are running the current `dev` branch HEAD with a freshly npm installed server module?  I get the following example outputs from `python3 -m minos.tools.pygame_client --width 512 --height 512 --depth`:
[image]
and from the gym demo through `python3 demo.py --depth --save_observations`:
[image]
[image]
Which do not seem to exhibit any staircasing artifacts.

ducha-aiki, 2018-08-16:
Reinstall takes time, so I would test as soon as I could. But I can answer for sure, that "freshly npm installed server module" doesn`t work for me.  I have definitely edit couple of module versions to be older as adviced here: https://github.com/minosworld/minos/issues/65#issuecomment-406370321

ducha-aiki, 2018-08-21:
@msavva yes, I am still observing this with clean dev version.
python3 -m minos.tools.pygame_client --width 512 --height 512 --depth

[image]

Here I have stepped back a bit, so you can see the place, from where I am observing it.
[image]