minosworld/minos #85: Redwood depth noise

efyphil, 2018-08-16:
I want to try some different noise for depth, instead of simple I found "redwood", but when i tried to use it, I had attributeError 'EasyDict' object has no attribute 'path'. Can you give me information how can I fix it

efyphil, 2018-09-04:
[image]
So, I edit this problem, it's about scale, but now, when I try to have non square frames I have this problem

angelxuanchang, 2018-09-14:
The redwood distortion noise model was designed specifically to model the noise of a Kinect-like sensor with a fixed resolution (hard-coded in the current implementation).  You would have to implement a different noise model that doesn't rely on the hard-coded parameters used by the redwood noise model.  If you need more details about the redwood noise model, you can contact the original authors.