minosworld/minos #96: How to run environment on GPU other than GPU:0 ?

ZhuFengdaaa, 2018-09-10:
How to run environment on GPU other than GPU:0 ? For example, GPU:1.

msavva, 2018-09-14:
Hi @ZhuFengdaaa ,
Currently, setting the `CUDA_VISIBLE_DEVICES` environment variable is the only way to control the assignment to GPUs.  You can make this a bit more programmatic from the python side, by passing it as a parameter similarly to `NODE_BASE_URL` (see the logic at https://github.com/minosworld/minos/blob/master/minos/lib/Simulator.py#L215 ) which is set as an environment variable for the simulation server processes.

ZhuFengdaaa, 2018-09-21:
Update: I found that no matter what `CUDA_VISIBLE_DEVICES` I set, node process will run on GPU:0 for rendering. However, I run one A2c training process on each GPU and hope each process render image on its own GPU rather than GPU:0, otherwise will block each other. (Maybe it is some setting of webgl?)

ZhuFengdaaa, 2018-11-10:
Finally I find out the way by accident. 

Set the environmental variable `DISPLAY=:0.1` or `DISPLAY=:0.2` can assign `node` subprocess on GPU device number 1 or 2.

closed 2018-11-10 by ZhuFengdaaa