minosworld/minos #97: Missing 'stats/v4/suncg.materials.csv'

mhrzica, 2018-09-18:
Hi,
 
I have downloaded SUNCG v2.0 (also tried with v2.1) in `$HOME/work/suncg` and tried to run MINOS with `python3 -m minos.tools.pygame_client` from from the root of the minos repository. After I run that command it takes some time to connect with server, then pygame windows opens and shuts immediately. 

I red simserver.log file and noticed "Error loading scene statistics { [Error: ENOENT: no such file or directory, open '/home/cupec/work//suncg//stats/v4/suncg.materials.csv']". Indeed I cannot find that file if I follow that path, because folder 'stats' does not contain folder v4, just suncg.materials.csv.

Can you please help me resolve this problem?

Here are log files:
[simserver.log](https://github.com/minosworld/minos/files/2391620/simserver.log)
[simulator.log](https://github.com/minosworld/minos/files/2391621/simulator.log)

This is simserver.log:
> 2018-09-18 09:02:43,319 INFO b'Node pro\xc5\xa1lo!'
> 2018-09-18 09:02:43,375 INFO b'Socket.io pro\xc5\xa1lo!'
> 2018-09-18 09:02:45,800 INFO b'Configuration { fileCache: { size: 50 },'
> 2018-09-18 09:02:45,800 INFO b'  imageCache: { size: 200 },'
> 2018-09-18 09:02:45,800 INFO b'  imageQueue: { concurrency: 4 },'
> 2018-09-18 09:02:45,800 INFO b"  base_url: '/home/cupec/work/',"
> 2018-09-18 09:02:45,800 INFO b'  assets_url:'
> 2018-09-18 09:02:45,800 INFO b"   'file:///home/cupec/work/minos/minos/server/node_modules/sstk/ssc/../server/static' }"
> 2018-09-18 09:02:47,391 INFO b'sstk/ssc pro\xc5\xa1lo!'
> 2018-09-18 09:02:47,605 INFO b'Waiting for client connection on port 52216'
> 2018-09-18 09:02:47,676 ERROR b'Cannot register unknown asset 2d3ds'
> 2018-09-18 09:02:47,676 INFO b'register ../server/static/data/matterport/matterport3d.json'
> 2018-09-18 09:02:47,725 INFO b'Registered asset group: mp3d'
> 2018-09-18 09:02:47,728 INFO b'Got 90 assets'
> 2018-09-18 09:02:47,786 INFO b'register ../server/static/data/suncg/suncg.planner5d.models.json'
> 2018-09-18 09:02:47,794 INFO b'Replace asset group p5d'
> 2018-09-18 09:02:47,794 INFO b'Registered asset group: p5d'
> 2018-09-18 09:02:47,862 INFO b'Got 2623 assets'
> 2018-09-18 09:02:48,502 INFO b'Loaded lightSpecs for 202 models.'
> 2018-09-18 09:02:48,502 INFO b'register ../server/static/data/suncg/suncg.planner5d.textures.json'
> 2018-09-18 09:02:48,569 INFO b'Replace asset group p5dTexture'
> 2018-09-18 09:02:48,569 INFO b'Registered asset group: p5dTexture'
> 2018-09-18 09:02:48,769 INFO b'Got 67072 assets'
> 2018-09-18 09:02:48,927 INFO b'Client OJOU-15h8Ec6xQ-yAAAA connected on port 52216'
> 2018-09-18 09:02:49,474 INFO b'THREE.WebGLRenderer 95'
> 2018-09-18 09:02:49,476 ERROR b'THREE.WebGLRenderer: OES_texture_float extension not supported.'
> 2018-09-18 09:02:49,476 ERROR b'THREE.WebGLRenderer: WEBGL_depth_texture extension not supported.'
> 2018-09-18 09:02:49,476 ERROR b'THREE.WebGLRenderer: OES_texture_half_float extension not supported.'
> 2018-09-18 09:02:49,476 ERROR b'THREE.WebGLRenderer: OES_texture_half_float_linear extension not supported.'
> 2018-09-18 09:02:49,476 ERROR b'THREE.WebGLRenderer: OES_standard_derivatives extension not supported.'
> 2018-09-18 09:02:49,476 ERROR b'THREE.WebGLRenderer: OES_element_index_uint extension not supported.'
> 2018-09-18 09:02:49,476 ERROR b'THREE.WebGLRenderer: OES_texture_float_linear extension not supported.'
> 2018-09-18 09:02:49,489 INFO b"Creating Simulator with options { simulator: 'room_simulator',"
> 2018-09-18 09:02:49,489 INFO b'  numSimulators: 1,'
> 2018-09-18 09:02:49,489 INFO b"  modalities: [ 'color', 'measurements' ],"
> 2018-09-18 09:02:49,489 INFO b"  outputs: [ 'color', 'measurements', 'rewards', 'terminals' ],"
> 2018-09-18 09:02:49,489 INFO b'  resolution: [ 84, 84 ],'
> 2018-09-18 09:02:49,489 INFO b'  frameSkip: 1,'
> 2018-09-18 09:02:49,489 INFO b"  host: 'localhost',"
> 2018-09-18 09:02:49,489 INFO b'  logActionTrace: false,'
> 2018-09-18 09:02:49,489 INFO b'  autoStart: true,'
> 2018-09-18 09:02:49,489 INFO b"  collisionDetection: { mode: 'raycast' },"
> 2018-09-18 09:02:49,490 INFO b'  navmap: false,'
> 2018-09-18 09:02:49,490 INFO b"  rewardType: 'dist_time',"
> 2018-09-18 09:02:49,490 INFO b'  observations:'
> 2018-09-18 09:02:49,490 INFO b'   { color: true,'
> 2018-09-18 09:02:49,490 INFO b'     forces: true,'
> 2018-09-18 09:02:49,490 INFO b'     audio: false,'
> 2018-09-18 09:02:49,490 INFO b'     objects: false,'
> 2018-09-18 09:02:49,490 INFO b'     depth: false,'
> 2018-09-18 09:02:49,490 INFO b'     map: false,'
> 2018-09-18 09:02:49,490 INFO b'     rgb: true,'
> 2018-09-18 09:02:49,490 INFO b'     normal: false,'
> 2018-09-18 09:02:49,490 INFO b'     objectType: false,'
> 2018-09-18 09:02:49,490 INFO b'     objectId: false,'
> 2018-09-18 09:02:49,490 INFO b'     roomType: false,'
> 2018-09-18 09:02:49,490 INFO b'     roomId: false },'
[... 299 lines cut ...]
> 2018-09-18 09:02:51,971 INFO b'SimState.sample: 70.172ms'
> 2018-09-18 09:02:51,971 ERROR b'Could not select a goal given spec {"categories":["arch","door"],"select":"random","distFromBbox":true}'
> 2018-09-18 09:02:51,972 INFO b'started... waiting for all images to load'
> 2018-09-18 09:02:52,897 INFO b'Removing subscribers for event drain [ 0 ]'
> 2018-09-18 09:02:52,899 INFO b'Timing waitImages: 926.830ms'
> 2018-09-18 09:02:52,911 INFO b'Timing start: 3367.657ms'
> 2018-09-18 09:02:53,067 ERROR b'THREE.WebGLRenderer: EXT_texture_filter_anisotropic extension not supported.'
> 2018-09-18 09:02:53,814 INFO b'Received close signal. Shutting down simulator server...'
> 2018-09-18 09:02:53,815 INFO b'Closed simulator'
> 2018-09-18 09:02:53,817 INFO Waiting for simserver stdout to finish
> 2018-09-18 09:02:53,844 INFO Finished simserver stdout
> 2018-09-18 09:02:53,844 INFO Finished simserver stderr
> 2018-09-18 09:02:53,845 INFO Waiting for simserver stderr to finish

This is from simulator.log:
> 2018-09-18 09:02:43,261 INFO {'sim_id': 'sim00', 'machine': 'acer', 'sim_git_hash': '12ab97d', 'stk_git_hash': '12ab97d'}
> 2018-09-18 09:02:43,273 INFO sim00:Starting sim server at /home/cupec/work/minos/minos/server/server.js with port 52216
> 2018-09-18 09:02:48,957 INFO sim00:connect
> 2018-09-18 09:02:53,814 INFO sim00:Stopping the simulator
> 2018-09-18 09:02:53,814 INFO Counter({'frames_received': 2})
> 2018-09-18 09:02:53,816 INFO sim00:disconnect
> 2018-09-18 09:02:53,817 INFO sim00:Stopping child servers
> 2018-09-18 09:02:53,817 INFO sim00:Killing sim pid 5117
> 2018-09-18 09:02:53,845 INFO sim00:Stopped child servers
> 2018-09-18 09:02:53,845 INFO sim00:Simulator killed

angelxuanchang, 2018-09-18:
Thanks you for trying out Minos.  We will fix the issue with the path to suncg//stats/v4.  However, it is unlikely to be the main cause of your issues (the simulator should be able to run without these files).

Can you let us know the following: 1) what branch of the Minos repository you are using, 2) what platform you are running minos on, 3) what python/pip version you use and whether you use anaconda 4) what node/npm version you are using, and 5) what your ${MINOS_ROOT}/minos/server/node_modules/sstk/package.json looks like.

mhrzica, 2018-09-18:
1) I am using *master branch of Minos repository (not sure if I checked it correctly, I used `git branch)
2) Platform: Linux - Ubuntu 16.04 LTS
3) Python 3.6.4 |Anaconda custom (64-bit)
4) node version: '10.10.0'
     npm version: '6.4.1',
5) package.json:
> {
>   "name": "sstk",
>   "version": "0.7.0",
>   "description": "SmartScenes ToolKit: a library for viewing, editing and annotating 3D models and scenes",
>   "main": "stk-ssc",
>   "dependencies": {
>     "lerna": "^2.11.0",
>     "shelljs": "^0.7.0"
>   },
>   "scripts": {
>     "postinstall": "node getdata.js && npm run bootstrap",
>     "bootstrap": "lerna bootstrap --hoist",
>     "clean": "lerna clean",
>     "ls": "lerna ls",
>     "test": "lerna run test"
>   },
>   "repository": {
>     "type": "git",
>     "url": "git+https://github.com/smartscenes/sstk.git"
>   },
>   "author": "Manolis Savva <manolis.savva@gmail.com>",
>   "license": "MIT",
>   "bugs": {
>     "url": "https://github.com/smartscenes/sstk/issues"
>   },
>   "homepage": "https://github.com/smartscenes/sstk"
> }

Maybe I should also mention that I have nVida GeForce GTX 1050.

angelxuanchang, 2018-09-18:
Can you also specify the git version you are using?  The sstk version you are getting is newer than the one the master branch of minos depends on.  Please also send us the socketio version that is being used: `conda list | grep socket`